espressif/esp-zigbee-sdk #570: Matching messages sent with responses (transaction numbers)

IgnacioHR, 2025-02-18:
### Question

Hi,

I'm struggling to know if it is possible to register a callback on every command sent by the device in order to be able to match it with the command response. In the function registered using the `esp_zb_core_action_handler_register(my_function)` I can see with messages with a transaction number.

### Additional context.

This is a small part of the log the device is generating.

```
I (3892) MICASA_GAS_METER: In zb_action_handler callback_id=0x1005
I (3892) MICASA_GAS_METER: Command default response:
  From address(0x0) src endpoint(1)
  To address (0x2366) endpoint(1)
    cluster(0x702) profile(0x104) transaction(0x0)
    cmd id(0xb) direction(0x1) is_common(0x1)
I (3906) MICASA_GAS_METER: Default Response: Status: 0x0000, To Command: 0xa
I (10430) MICASA_GAS_METER: In zb_action_handler callback_id=0x1005
I (10430) MICASA_GAS_METER: Command default response:
  From address(0x0) src endpoint(1)
  To address (0x2366) endpoint(1)
    cluster(0x702) profile(0x104) transaction(0x1)
    cmd id(0xb) direction(0x1) is_common(0x1)
I (10445) MICASA_GAS_METER: Default Response: Status: 0x0000, To Command: 0xa
I (33323) MICASA_GAS_METER: Enter deep sleep
```

In other part of the code I'm doing:

```
esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd)
```

So I'm aware the device is pushing the values to be reported using a command `0xa`. I would like to know if it is possible to, somehow, register a callback to know exactly when the transaction was initiated so we can match, in code, when all commands sent out has been fully processed.

xieqinan, 2025-02-19:
Hi,

The `esp_zb_zcl_command_send_status_handler_register()` will be triggered with transaction information when the ZCL command is sent, it may be help you.

IgnacioHR, 2025-02-19:
¿Are there any examples about how to use that function? This is what I'm doing:

```
    esp_err_t ret = esp_zb_device_register(esp_zb_ep_meter_list);
    ESP_RETURN_ON_FALSE(ret == ESP_OK, , TAG, "Failed to register meter endpoint");

    esp_zb_core_action_handler_register(zb_action_handler);
    ESP_LOGI(TAG, "Registering additional callbacks");
    esp_zb_device_cb_id_handler_register(zb_device_cb_id);
    esp_zb_raw_command_handler_register(zb_raw_command_handler);
    esp_zb_zcl_command_send_status_handler_register(zb_command_handler);
    ESP_LOGI(TAG, "Additional callbacks registered");

```

The code compiles OK but the callback functions are not called.

```
I (310) MICASA_GAS_METER: Registering additional callbacks
I (310) MICASA_GAS_METER: Additional callbacks registered
I (321) MICASA_GAS_METER: Initialize Zigbee stack
I (329) MICASA_GAS_METER: Turning on Zigbee radio
I (3176) MICASA_GAS_METER: Device rebooted
I (3177) MICASA_GAS_METER: Zigbee Loop Task...
I (3427) MICASA_GAS_METER: Reporting to client Sum=Yes, Instant=No, Bat=Yes, Status=No, Exten=No
I (3427) MICASA_GAS_METER: CurrentSummationDelivered reported
I (3430) MICASA_GAS_METER: BatteryPercentageRemaining reported
I (4116) MICASA_GAS_METER: In zb_action_handler callback_id=0x1005
I (4117) MICASA_GAS_METER: Command default response:
  From address(0x0) src endpoint(1)
  To address (0x2366) endpoint(1)
    cluster(0x702) profile(0x104) transaction(0)
    cmd id(0xb) direction(0x1) is_common(0x1)
I (4131) MICASA_GAS_METER: Default Response: Status: 0x0000, To Command: 0xa
I (4228) MICASA_GAS_METER: In zb_action_handler callback_id=0x1005
I (4228) MICASA_GAS_METER: Command default response:
  From address(0x0) src endpoint(1)
  To address (0x2366) endpoint(1)
    cluster(0x1) profile(0x104) transaction(1)
    cmd id(0xb) direction(0x1) is_common(0x1)
I (4242) MICASA_GAS_METER: Default Response: Status: 0x0000, To Command: 0xa
I (10585) MICASA_GAS_METER: In zb_action_handler callback_id=0x1005
I (10585) MICASA_GAS_METER: Command default response:
  From address(0x0) src endpoint(1)
  To address (0x2366) endpoint(1)
    cluster(0x702) profile(0x104) transaction(2)
    cmd id(0xb) direction(0x1) is_common(0x1)
I (10599) MICASA_GAS_METER: Default Response: Status: 0x0000, To Command: 0xa
I (10695) MICASA_GAS_METER: In zb_action_handler callback_id=0x1005
I (10695) MICASA_GAS_METER: Command default response:
  From address(0x0) src endpoint(1)
  To address (0x2366) endpoint(1)
    cluster(0x702) profile(0x104) transaction(3)
    cmd id(0xb) direction(0x1) is_common(0x1)
I (10710) MICASA_GAS_METER: Default Response: Status: 0x0000, To Command: 0xa
I (33439) MICASA_GAS_METER: Enter deep sleep
```

The log shows the functions are registered but the only callback that is called is `zb_action_handler` 

I may be doing something wrong.

xieqinan, 2025-04-23:
> The code compiles OK but the callback functions are not called.

Have you send some ZCL commands on your application? 

If the `esp_zb_raw_command_handler_register(zb_raw_command_handler);` is added to the [switch example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch), the `zb_raw_command_handler` will be called once the `BOOT` button is pressed to send ON/OFF command.

xieqinan, 2025-06-03:
Hi,

Do you have any updates?

IgnacioHR, 2025-06-03:
Hello,

I'm sorry for the late response, I moved to other priorities so I can't provide updates on this specific topic.
- In the last month I decided to change the hardware design from ESP32-C6-DevKit to a Seeed Studio XIAO ESP32C6
- I need to read again the specifications just to make sure I'm not misunderstanding something

The device is programmed to report certain values to the host when the values change. I'm forcing to send the values in code.

```
        status = esp_zb_zcl_set_attribute_val(MY_METERING_ENDPOINT,
            ESP_ZB_ZCL_CLUSTER_ID_METERING, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
            ESP_ZB_ZCL_ATTR_METERING_CURRENT_SUMMATION_DELIVERED_ID, &current_summation_delivered, false);
        if (status != ESP_ZB_ZCL_STATUS_SUCCESS) {
            ESP_LOGE(TAG, "Updating value of current summation delivered: %d", status);
            return status;
        }
```

and later
```
void zb_radio_send_values(EventBits_t uxBits)
{
    esp_zb_zcl_report_attr_cmd_t report_attr_cmd = {0};
    report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
    report_attr_cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI;
    report_attr_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_METERING;
    report_attr_cmd.zcl_basic_cmd.src_endpoint = MY_METERING_ENDPOINT;

    if ((uxBits & CURRENT_SUMMATION_DELIVERED_REPORT) != 0) {
        report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_METERING_CURRENT_SUMMATION_DELIVERED_ID;
        report_attr_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_METERING;
        ESP_ERROR_CHECK(esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd));
        ESP_LOGI(TAG, "CurrentSummationDelivered reported");
    }
```

so I'm sure the device is sending values up to the host also because the host is showing the values.
Now, in order to save battery, I would like to put the device in sleep mode when the device is sure the values have arrived the host. At this time all I can do is to set-up a timer for 30 seconds from the last value transmission but this is not efficient. I wanted to capture the surrounding events so I could get rid of the timer and put the device on sleep mode when the last open transaction is completed.

xieqinan, 2026-01-20:
Hi @IgnacioHR,

Do you have any updates? Please let me know your state so that I can provide appropriate assistance.