espressif/esp-zigbee-sdk #571: Failed to initialize Zigbee stack (status: ESP_FAIL) - Only method of recovery a reboot?

OOHehir, 2025-02-21:
### Question

Home to recover from 'Failed to initialize Zigbee stack (status: ESP_FAIL)'

### Additional context.

Hello, I'm using an ESP32-C6 as an end device. My code closely resembles the [ HA_on_off_light](https://github.com/espressif/esp-idf/tree/master/examples/zigbee/light_sample/HA_on_off_light) example. This question has been touched upon before but I'd like to confirm my understanding.

Once the C6 enters the 'Failed to initialize Zigbee stack (status: ESP_FAIL)' state the only method of recovery (i.e. attempt to rejoin the network) is a reboot (i.e. user power cycles the device or esp_restart()) ? There is no method to restart the Zigbee 'rejoin' process without this?

Thanks

xieqinan, 2025-02-24:
Hi,

Please refer to the latest implementation in the SDK, which has fixed this issue: https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c#L67

OOHehir, 2025-02-26:
@xieqinan Many thanks, that change re-establishes the Zigbee connection without a reboot of the ESP32

closed 2025-02-26 by OOHehir