espressif/esp-zigbee-sdk #573: Configuring Zigbee Switch Behavior (Press-and-Hold vs. Press-and-Release) via Coordinator in ESP-Zigbee-SDK

VIinhQuang, 2025-02-26:
### Question

I want to create a Zigbee switch device that can be configured to work as either a press-and-hold or press-and-release action. This configuration should be set by the coordinator.

I’m not sure if this can be implemented using the ESP-Zigbee-SDK. If it is possible, could you guide me on how to do it?

Thank you very much!

xieqinan, 2025-02-26:
Hi,

I believe this feature is related to the switch driver rather than Zigbee. If so, I advise you to first refer to the [GPIO driver documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/gpio.html) of ESP.

VIinhQuang, 2025-02-26:
Hi, @xieqinan 

I understand that whether a switch operates in momentary (press-and-release) mode or toggle (press-and-hold) mode depends on how the device’s firmware is programmed.

However, my question is specifically about whether there is a command or mechanism in the ESP-Zigbee-SDK that allows configuring this mode dynamically through the ESP Coordinator (rather than hardcoding it in the device firmware).

For example, some Aqara Zigbee modules allow users to configure the switch mode via the gateway. Once set to press-and-hold mode, it cannot function as a momentary switch unless reconfigured. This configuration is done through the Aqara gateway.

Can the ESP-Zigbee-SDK achieve similar functionality, where the operation mode of a Zigbee switch can be set and changed via the ESP Coordinator?

xieqinan, 2025-02-27:
This feature seems to be a Zigbee application, but the esp-zigbee-sdk does not provide a direct example for implementation. The [On/Off switch configuration cluster](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_cluster.h#L114) allows users to define a `Toggle`, `Momentary`, or `Multifunction` switch. A reasonable approach would be to control the attributes of this cluster to meet your requirements.

For example:
- On the `switch device` side, you can define both a `Toggle` type switch endpoint and a `Momentary` type switch endpoint.  
- If the `switch device` receives a write `SwitchActions` attribute command for the `Toggle` endpoint, you can switch the actual switch driver to `Toggle` mode. The same applies to `Momentary` mode.

closed 2025-03-01 by VIinhQuang