espressif/esp-zigbee-sdk #576: How to get the transition time of ZigBee level control clusters move actions (IDFGH-14555)

pvginkel, 2025-02-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I would like to control the transition of ZigBee level control clusters myself.

This is currently implemented by the ZigBee stack. I'm seeing log messages like this, a few in sequence as the level changes:

```
ZB_TRACE_LOG[0]: zcl/zcl_continuous_value_change_commands.c:203    steps_number = 5
```

The level of the `ESP_ZB_ZCL_ATTR_LEVEL_CONTROL_CURRENT_LEVEL_ID` attribute is increased in steps to reach the target level. The steps however are quite big so the transition are very noticeable. I would like to control this myself. I would like to just get one notification that the level has changed and the associated transition time, instead of multiple attribute changes of the `ESP_ZB_ZCL_ATTR_LEVEL_CONTROL_CURRENT_LEVEL_ID` attribute.

How can this be achieved?

xieqinan, 2025-03-12:
Hi,

Thank you for your log. However, it contains too much information. You can try disabling debug mode.  

If you send `esp_zb_zcl_level_move_to_level_cmd_req()` with a target value of 254, a transition time of 30, and the original `current_level` set to 0, do you observe that the stack triggers approximately 30 `ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID` callbacks, gradually increasing `current_level` from 0 to 254 in steps of 7 or 8 every 100 ms?  

If so, this is the expected result. If you think the step value too large, you can try dividing it into smaller increments over the 100 ms intervals for finer control.

pvginkel, 2025-03-12:
Thank you for coming back with this. I can confirm that for my use case these steps are too course.

I opted for a different approach. I already had an implementation to more gradually change the light intensity; I just needed to get the underlying parameters. I got this to work by intercepting the `device_handler` on the result of `zb_af_get_endpoint_desc`. I'm manually checking all messages that may have a transition time, storing it and setting it to 0. This works quite well for me.

If someone else hits this issue, have a look at https://github.com/pvginkel/esp-libs/blob/289eff6037a24125deef610a82dbd4554c7876e0/esp-zha-support/src/ZigBeeCommandFilter.cpp#L29 to get a starting point on how to accomplish this.

closed 2025-03-12 by pvginkel