espressif/esp-zigbee-sdk #578: OTA Updates Fail When Multiple Zigbee Devices Connected

Tuominen04, 2025-02-26:
### **Issue Description**
When attempting to perform OTA updates with multiple Zigbee devices connected to the network, the updates consistently fail with corruption errors. It appears that when two or more devices try to update simultaneously, they receive corrupted firmware data.

### **Current Behavior**
Gateway initiates OTA update process
Multiple clients respond and begin requesting update data
Clients receive corrupted data chunks
Updates fail with errors such as "OTA image has invalid magic byte" or "Checksum failed"

### **ExpectedBehavior**
All devices should be able to complete OTA updates successfully, whether updates are performed simultaneously or sequentially.

### **Logs**
### Server log showing update attempts
```
I (284957) OTA: OTA notification send status: ESP_OK
I (285067) OTA: Server received OTA query from client 0xc33e
I (285067) OTA: Client version: 0x1010101
I (285077) OTA: OTA table index: 0x0
I (285107) OTA: OTA notification send status: ESP_OK
I (285107) OTA: OTA transfer initiated for client 0xc33e
I (285137) OTA: Server received OTA query from client 0x8d98
I (285137) OTA: Client version: 0x1010101
I (285137) OTA: OTA table index: 0x0
I (285167) OTA: OTA notification send status: ESP_OK
I (285167) OTA: OTA transfer initiated for client 0x8d98
I (288597) OTA: OTA file opened successfully
I (288607) OTA: Progress: 0.0% (167/523904 bytes) to client 0x0
I (288607) OTA: OTA client address: 0xc33e
I (288607) OTA: OTA version: 0x1010102, image type: 0x1011, server status: 0
I (288857) OTA: Progress: 0.1% (334/523904 bytes) to client 0x0
I (288857) OTA: OTA client address: 0x8d98
I (288857) OTA: OTA version: 0x1010102, image type: 0x1011, server status: 0
I (288997) OTA: Progress: 0.1% (557/523904 bytes) to client 0x0
E (289067) OTA: zb_ota_upgrade_server_status_handler(1018): Failed to parse OTA upgrade server status message
I (289347) OTA: Progress: 0.1% (780/523904 bytes) to client 0x0
...
I (1035607) OTA: Progress: 100.0% (523904/523904 bytes) to client 0x0
I (1035617) OTA: Reached end of OTA file
I (1035927) OTA: OTA file opened successfully
I (1035927) OTA: Progress: 0.0% (133/523904 bytes) to client 0x0
I (1036017) OTA: OTA client address: 0xc33e
I (1036017) OTA: OTA version: 0x1010102, image type: 0x1011, server status: 2
I (1036017) OTA: OTA upgrade time: 0x1234a
```

### Client 1 update failing with checksum error
```
I (3739) ON_OFF_LIGHT: Joined network successfully (Extended PAN ID: 48:31:b7:ff:fe:c3:f2:a2, PAN ID: 0x0e5b, Channel:13, Short Address: 0xc33e)
I (264579) OTA: Queried OTA image from address: 0x0000, endpoint: 6
I (264579) OTA: Image version: 0x1010102, manufacturer code: 0x1001, image size: 523960
I (264589) OTA: Approving OTA image upgrade
I (264589) OTA: -- OTA upgrade start
I (268239) OTA: -- OTA Client receives data: progress [167/523904]
...
I (1015489) OTA: -- OTA Client receives data: progress [523904/523904]
I (1015489) OTA: -- OTA upgrade check status: ESP_OK
I (1015549) OTA: -- OTA upgrade apply
I (1020559) OTA: -- OTA Finish
I (1020559) OTA: -- OTA Information: version: 0x1010102, manufacturer code: 0x1001, image type: 0x1011, total size: 523904 bytes, cost time: 755957 ms,
I (1020559) esp_image: segment 0: paddr=00110020 vaddr=42060020 size=10440h ( 66624) map
I (1020619) esp_image: segment 1: paddr=00120468 vaddr=00000000 size=00000h (     0)
I (1020619) esp_image: segment 2: paddr=00120470 vaddr=00000000 size=00000h (     0)
I (1020629) esp_image: segment 3: paddr=00120478 vaddr=00000000 size=00000h (     0)
I (1020639) esp_image: segment 4: paddr=00120480 vaddr=00000000 size=00000h (     0)
E (1020639) esp_image: Checksum failed. Calculated 0xac read 0x0
E (1020649) OTA: zb_ota_upgrade_status_handler(192): Failed to end OTA partition, status: ESP_ERR_OTA_VALIDATE_FAILED
```

### Client 2 failing with invalid magic byte
```
I (11029) ON_OFF_LIGHT: Joined network successfully (Extended PAN ID: 48:31:b7:ff:fe:c3:f2:a2, PAN ID: 0x0e5b, Channel:13, Short Address: 0x8d98)
I (282309) OTA: Queried OTA image from address: 0x0000, endpoint: 6
I (282309) OTA: Image version: 0x1010102, manufacturer code: 0x1001, image size: 523960
I (282309) OTA: Approving OTA image upgrade
I (282319) OTA: -- OTA upgrade start
I (286079) OTA: -- OTA Client receives data: progress [167/523904]
E (286079) esp_ota_ops: OTA image has invalid magic byte (expected 0xE9, saw 0x00)
E (286089) OTA: zb_ota_upgrade_status_handler(175): Failed to write OTA data to partition, status: ESP_ERR_OTA_VALIDATE_FAILED
I (286099) OTA: OTA status: 4
```

### Enviroments
- SDK: 1.6.4 (last commit 6167fb8004678bc89174f9a2b0fce1aab757f28f)
- idf: 5.3.1
- esp-zboss-lib: 1..6.1
- esp-zigbee-lib: 1.6.1
- Server: esp32-s3
- Clients: esp32-c6

xieqinan, 2025-02-27:
@Tuominen04,

Could you share how you control the `s_ota_image_offset` changes when multiple devices perform OTA simultaneously? Please ensure that multiple OTA client devices have the correct image offset so they receive the correct OTA data.

Tuominen04, 2025-02-27:
The current implementation in the example code uses a single static global variable `s_ota_image_offset`.

```c
static uint32_t s_ota_image_offset = 0;

static esp_err_t zb_ota_next_data_handler_Spiffs(esp_zb_ota_zcl_information_t message, uint16_t index, uint8_t size, uint8_t **data) {
    static FILE *ota_file = NULL;
    static uint8_t file_buffer[2048];
    
    if (!ota_file) {
        ota_file = fopen("/spiffs/led_light.bin", "rb");
         // Validation
    }    
    // Move the file pointer to the correct offset
    if (fseek(ota_file, s_ota_image_offset, SEEK_SET) != 0) {
        // ...
    }

    // Read data into buffer
    size_t read_size = fread(file_buffer, 1, size, ota_file);
    if (read_size > 0) {
        *data = file_buffer;
        s_ota_image_offset += read_size;  // Global offset is incremented here

        // ...
    }
    
    // ...
}

xieqinan, 2025-02-28:
> The current implementation in the example code uses a single static global variable s_ota_image_offset.

I think you should maintain multiple `ota_image_offset` values based on the number of OTA clients in your application. Since there is only one OTA file on the OTA server, each OTA client will have its own OTA progress. You need to ensure that the data offset of the OTA file is correctly managed for each OTA client.  

Could you try optimizing this approach?

chshu, 2025-03-20:
@Tuominen04 please reopen if any follow up questions.

closed 2025-03-20 by chshu