espressif/esp-zigbee-sdk #581: ZDO Device Unavailable (0x3c)

grebenyuk51, 2025-03-01:
### Question

Sometimes I can see in logs in `esp_zb_app_signal_handler()` ZDO Device Unavailable (0x3c)
```
I (286836) ROUTING: err=261, itr=65535
I (286836) Time: 1970-01-06T19:33:39Z
I (288946) Yellow Squash: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
**E (288946) Yellow Squash: ESP_ZB_ZDO_DEVICE_UNAVAILABLE. Short: 0x4ee9, long: 0x00124b0029e8b232**
I (297656) DS18B20: temperature read from DS18B20[0]: 13.50C
I (297656) Network: Network Extended PAN ID: 01:37:0d:52:5e:ca:40:72, PAN ID: 0x07e1, Channel:15)
I (297656) Network: **My addr. Short: 0x4ee9, long: 0x404ccafffe5b2744**
I (297666) ROUTING: err=0, itr=3
short_addr=0xb416
device_type=1
depth=6
rx_on_when_idle=1
relationship=0
lqi=102
rssi=-60
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=296865
I (297686) ROUTING: err=0, itr=37
short_addr=0x8956
device_type=3
depth=0
rx_on_when_idle=1
relationship=3
lqi=25
rssi=-75
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=296885
I (297706) ROUTING: err=0, itr=41
short_addr=0xcd80
device_type=3
depth=1
rx_on_when_idle=1
relationship=3
lqi=10
rssi=-78
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=296904
I (297726) ROUTING: err=0, itr=44
short_addr=0x1828
device_type=3
depth=1
rx_on_when_idle=1
relationship=3
lqi=40
rssi=-72
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=296923
I (297746) ROUTING: err=0, itr=47
short_addr=0x0000
device_type=3
depth=0
rx_on_when_idle=1
relationship=3
lqi=10
rssi=-78
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=296943
I (297766) ROUTING: err=261, itr=65535
I (297766) Time: 1970-01-06T19:33:50Z
**_ZB_TRACE_LOG[0]: nwk/nwk_main.c:3967    unknown or disallowed cmd - drop_**
I (308586) DS18B20: temperature read from DS18B20[0]: 13.44C
I (308586) Network: Network Extended PAN ID: 01:37:0d:52:5e:ca:40:72, PAN ID: 0x07e1, Channel:15)
I (308586) **Network: My addr. Short: 0x4ee9, long: 0x404ccafffe5b2744**
I (308596) ROUTING: err=0, itr=3
short_addr=0xb416
device_type=1
depth=6
rx_on_when_idle=1
relationship=0
lqi=102
rssi=-60
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=307795
I (308616) ROUTING: err=0, itr=23
short_addr=0x0068
device_type=3
depth=1
rx_on_when_idle=1
relationship=3
lqi=45
rssi=-71
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=307814
I (308636) ROUTING: err=0, itr=37
short_addr=0x8956
device_type=3
depth=0
rx_on_when_idle=1
relationship=3
lqi=25
rssi=-75
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=307834
I (308656) ROUTING: err=0, itr=41
short_addr=0xcd80
device_type=3
depth=1
rx_on_when_idle=1
relationship=3
lqi=10
rssi=-78
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=307853
I (308676) ROUTING: err=0, itr=44
short_addr=0x1828
device_type=3
depth=1
rx_on_when_idle=1
relationship=3
lqi=40
rssi=-72
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=307873
I (308696) ROUTING: err=0, itr=47
short_addr=0x0000
device_type=3
depth=0
rx_on_when_idle=1
relationship=3
lqi=10
rssi=-78
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=307892
I (308716) ROUTING: err=261, itr=65535
I (308716) Time: 1970-01-06T19:34:01Z
I (309806) Yellow Squash: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
**E (309806) Yellow Squash: ESP_ZB_ZDO_DEVICE_UNAVAILABLE. Short: 0x4ee9, long: 0x00124b0029e8b232**
W (313616) Yellow Squash: Receive Zigbee action(0x1005) callback
I (313616) Yellow Squash: Default CMD command cluster 1026 (0x402) id 11 (0xb) status 0 from 0x0
I (313626) Yellow Squash: from: 00:12:4b:00:29:e8:b2:32
```

In Wireshark these moments looks like
[image]

What is the proper way of processing such situations?

grebenyuk51, 2025-03-01:
Route request with no Response
[image]

```
I (21405766) ROUTING: err=0, itr=2
short_addr=0xfbe8
device_type=1
depth=3
rx_on_when_idle=1
relationship=0
lqi=178
rssi=-45
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=21404966
I (21405786) ROUTING: err=0, itr=3
short_addr=0xb416
device_type=1
depth=6
rx_on_when_idle=1
relationship=0
lqi=91
rssi=-62
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=21404985
I (21405806) ROUTING: err=0, itr=4
short_addr=0x87f4
device_type=3
depth=1
rx_on_when_idle=1
relationship=3
lqi=15
rssi=-77
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=21405005
I (21405826) ROUTING: err=0, itr=5
short_addr=0x6d1a
device_type=3
depth=0
rx_on_when_idle=1
relationship=3
lqi=20
rssi=-76
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=21405024
I (21405846) ROUTING: err=0, itr=7
short_addr=0x674f
device_type=3
depth=1
rx_on_when_idle=1
relationship=3
lqi=10
rssi=-78
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=21405044
I (21405866) ROUTING: err=0, itr=9
short_addr=0x429d
device_type=3
depth=5
rx_on_when_idle=1
relationship=3
[... 291 lines cut ...]
age=157
device_timeout=10
timeout_counter=21405539
I (21406356) ROUTING: err=0, itr=47
short_addr=0x0000
device_type=3
depth=0
rx_on_when_idle=1
relationship=3
lqi=10
rssi=-78
outgoing_cost=186
age=157
device_timeout=10
timeout_counter=21405559
I (21406376) ROUTING: err=261, itr=65535
I (21406386) Time: 1970-01-07T01:25:38Z
ZB_TRACE_LOG[0]: nwk/nwk_main.c:3967    unknown or disallowed cmd - drop
I (21410346) Yellow Squash: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
E (21410346) Yellow Squash: ESP_ZB_ZDO_DEVICE_UNAVAILABLE. Short: 0x4ee9, long: 0x00124b0029e8b232
I (21417196) DS18B20: temperature read from DS18B20[0]: 12.81C
I (21417196) Network: Network Extended PAN ID: 01:37:0d:52:5e:ca:40:72, PAN ID: 0x07e1, Channel:15)
I (21417196) Network: My addr. Short: 0x4ee9, long: 0x404ccafffe5b2744
I (21417206) ROUTING: err=0, itr=2
```

xieqinan, 2025-03-04:
> Not neighbor
Current device addr is 0x4ee9, and unavailable address is the same
Device is not responsive during these situations. I have to reboot it

This signal is usually generated when a command for a remote device is confirmed as unreachable after multiple APS retransmissions. At this point, the remote device is either offline or does not exist. It would be best to investigate the root cause of the issue from the remote device's side.  

Could you also provide more details about the End Device, such as its logs and relevant functions?

chshu, 2025-03-20:
@grebenyuk51 any update on the issue?

xieqinan, 2025-06-03:
Hi @grebenyuk51 ,

Please see: https://github.com/espressif/esp-zigbee-sdk/issues/581#issuecomment-2696124862.

Please ensure that the destination device is on the same network as the coordinator and is online. If you send a command to an unknown address, the `ESP_ZB_ZDO_DEVICE_UNAVAILABLE` signal will be triggered to indicate that the destination address is unavailable.

Let’s close this issue for now. If it still confuses you, feel free to reopen it.

closed 2025-06-03 by xieqinan

xieqinan, 2025-12-19:
@renson-thijs-dhaenens 

> We see ZDO Device Unavailable (0x3c) when we disable the coordinator, but it does not recover when we reanable the coordinator.

Please ensure the device has re-join to the network when coordinator is enabled again. If they still are on the same network, I believe the communication can be recoverd.