espressif/esp-zigbee-sdk #582: Getting brownout detector resets after starting Zigbee

IgnacioHR, 2025-03-02:
### Question

Hi,

I'm experiencing brownout detector resets when enabling the zigbee radio and don't know how to solve it. If I disable brownout detection the zigbee radio starts just fine but I don't feel comfortable with this situation as the results might be unpredictable.

### Additional context.

Hi,

Let me explain a bit about my setup and the issue I’m facing.

### Prototype Board Overview

Here’s a picture of my prototype board:
[image]

I’m powering the circuit with a **2S2P LiPo battery pack** and a **DC-DC converter**. I’ve added a **1000µF capacitor to the 3.3V power rail**. The ESP32-C6-DEVKIT (version 1.2) has its power pin disconnected, ensuring that **all 3.3V power comes solely from the DC-DC converter**. The board is also connected via USB, but disconnecting it makes no difference in behavior.

### Board operation:

1. The ESP32-C6 is normally in **Deep Sleep mode**.
2. When the **right button** is pressed, the CPU wakes up, increases a counter, and enters deep sleep again after **500 ms**. This works flawlessly.
3. When the **left button** is pressed (or after pressing the right button **10 times**) the Zigbee radio is powered on (this is an **End Device**) and the counter value is reported. **This is where I experience brownout issues, preventing the radio from starting properly**.

### Troubleshooting Steps Taken
1. **No issue when powered via USB** but this solution is not possible for production.
2. I had to add an **external reset capacitor and resistor**, as removing the JP5 also disconnects the reset. However, I'm confident this is **a brownout issue** as `esp_reset_reason()` consistently reports a brownout reset when the zigbee radio is turned on.
3. I've tried **reducing the transmission power** by calling:``` 
esp_zb_set_tx_power(IEEE802154_TXPOWER_VALUE_MIN);
``` before `esp_zb_start(false)` but **power consumption remains unchanged** according to ammeter readings.
4. Adding **more capacitors** delays the brownout, but **1000uF should be enough**, so I suspect I might be missing something else.

### Questions:

- Could there be any additional power spikes when initializing Zigbee that I’m not accounting for?
- Is there a recommended way to confirm whether esp_zb_set_tx_power() is actually applying the intended power reduction?
- Any suggestions on additional power stability improvements?

Thanks in advance for your help!

Best regards,
Ignacio

chshu, 2025-03-04:
@IgnacioHR It appears to be a power supply issue. Please refer to Section 5.5 of the [ESP32-H2 Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-h2_datasheet_en.pdf) for detailed current consumption values and ensure that the power supply meets the hardware requirements.

The function `esp_zb_set_tx_power()` configures the transmission power, you can measure the power consumption during transmission. But note that it can't change the current consumption during CPU active and RF receiving.

IgnacioHR, 2025-03-04:
@chshu Thank you for your response.

After reviewing the [PCB Layout](https://dl.espressif.com/dl/schematics/esp32-c6-devkitc-1-pcb-layout_v1.2.pdf) I noticed the RST pin shall work even if the JP5 is disconnected. Then I decided to replace the 1000uF Capacitor with a smaller 22uF capacitor and **the board seems to restart ok**.

Regarding consumption there might be something in the API but I can't do anything but report it to you **so you can take a close look at it** and repeat my experiments. The point is that changes in the `esp_zb_set_tx_power()` does not change current consumption and this is a bit unexpected. 

This is an screen capture of my tests. the peaks corresponds to consumption during packet transmission the values goes up over 450mah. Everything matches the values indicated in [Section 5.6 of the the ESP32-c6 data sheet](https://www.espressif.com/sites/default/files/documentation/esp32-c6_datasheet_en.pdf)

[image]

closed 2025-03-04 by IgnacioHR