espressif/esp-zigbee-sdk #585: Road_Map

Usama050, 2025-03-04:
### Question

Hello! I need to make a project using ESP32-H2-Devki-M1. I want to connect Zigbee Sensor with ESP32-H2 how can I do that. Can anyone suggest me a road map Thank you in advance!

Usama050, 2025-03-11:
I am using ESP_IDF extension in VS code and I am trying to running the example of HA but it did not work can you suggest me how can I run the example in VS-Code ESP-IDF extension and I am using window 11

Usama050, 2025-03-11:
Somehow I run the example but I need your help can I use this example to connect my sensor with esp32h2 

[image]

xieqinan, 2025-03-12:
> Somehow I run the example but I need your help can I use this example to connect my sensor with esp32h2

If your sensor is a standard Zigbee device, it should be able to connect to the network created by the [HA_on_off_switch](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch). Please note that the On/Off switch functions as a Home Automation (HA) coordinator. You need to determine the type of your sensor to ensure seamless functional integration.

Usama050, 2025-03-12:
Thank You Sir I got your point in that example "HA_on_off_switch" they used two esp32 one as a coordinator and other is as end device. In my case end device will be my sensors. But do I need to add specific libraries or just I have to turn on my sensor and they will connect with esp32h2.  and one more thing is there any need to do any changing in " Espressif IoT Development Framework Configuration " menu. Thank you Sir, I appreciate you effort. 

[image]

Usama050, 2025-03-13:
Hello! Sir, How are you doing? I hope you are doing well. I am waiting for you reply!! Thank you

xieqinan, 2025-03-14:
Hi,

> my sensor is not connected should I need to change the code the sensor is a zigbee button "KOJIMA-SPB-ZGB-W

I suggest that you first read the illustration of the `KOJIMA-SPB-ZGB-W` to understand the role of the device, the type of network it can join (distributed or centralized), and the supported network joining methods (such as install code, steering based on a private link key, or `touchlink`) and so on.

> Sir can we meet on zoom, MS teams or Google me just for 30 minutes to 1 hour if you can give me some time I will be grateful for this act of kindness.

I believe you can reach out to [technical-inquiries](https://www.espressif.com/en/contact-us/technical-inquiries) for assistance.

Usama050, 2025-03-19:
Hello! Sir, how are you doing? I hope you are doing well. I connected my ZigBee door Sensor ZG-102ZL with esp32h2 but I can not read it's value on serial monitor can you help me how can I do that this is just my last step Thank you in advance. I will be grateful for all your help. Thank you 

[image]

xieqinan, 2025-03-20:
Hi,
This [example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L184) provide a method to read the attribute from remote device.

chshu, 2025-03-20:
@Usama050 please create separate issue for new question.

closed 2025-03-20 by chshu