espressif/esp-zigbee-sdk #586: About the maximum rate of zigbee transmission.

Mushan-bin, 2025-03-04:
How much data can a zigbee terminal transmit to the zigbee Coordinator at one time and how often?
If I want to transfer a large amount of data to a zigbee coordinator through a zigbee terminal, and do it in a very short time, is it feasible?

lpy4105, 2025-03-05:
Would like to share more details about you requirement, i.e. the interval and data length of each transmission? The maximum rate is limited by 802.15.4 PHY and is further reduced by MAC/NWK/APS/ZCL layer overhead.

Mushan-bin, 2025-03-05:
How many bytes of data can be transferred by zigbee1 second

Mushan-bin, 2025-03-07:
I want to send data to the coordinator through the terminal custom cluster, want to reach 8KB/s, how to set the coordinator to receive stable and no packet loss. @lpy4105

lpy4105, 2025-03-07:
I think there is nothing needed on receiver side. On the sender side, if you are sending short packets (< 80 Bytes), the transmission interval is better to be more than tens of milliseconds; if your are sending long packets, the transmission interval is better to be more than hundreds of milliseconds.

You can register `esp_zb_zcl_command_send_status_callback_t` via `esp_zb_zcl_command_send_status_handler_register()` to monitor the transmission status of each ZCL packet. You would get "SUCCESS" only if the packet was acknowledged by peer APS layer.

To better understand your requirement, would you mind share so details about your use case?

Mushan-bin, 2025-03-07:
The zigbee terminal sends data (240 bytes) to the zigbee Coordinator through a custom cluster, with a delay of 500ms, and the total amount of data sent is 250,000 bytes. It takes about 9 minutes, but the data will cause packet loss due to network state guidance. I want zigbee's transmission to be faster and without packet loss. How should it be operated? It can reach 250kbps. @lpy4105

lpy4105, 2025-03-07:
You could try the following parameters if possible: sending 640 bytes per 120 ms.

For the packet loss, please check from `esp_zb_zcl_command_send_status_callback_t` to ensure there is no transmission failure. If there is still data loss, it is better to provide more detailed information, e.g. pcap file, device log and example code.

Mushan-bin, 2025-03-08:
Can the custom cluster send 640 bytes at a time? Why can't the coordinator receive me when I'm larger than 240 bytes?

> 如果可能，您可以尝试以下参数：每 120 毫秒发送 640 字节。
> 
> 对于丢包情况，请检查 以确保没有传输失败。如果仍然有数据丢失，最好提供更详细的信息，例如 pcap 文件、设备日志和示例代码。`esp_zb_zcl_command_send_status_callback_t`

Mushan-bin, 2025-03-10:
> Can the custom cluster send 640 bytes at a time? Why can't the coordinator receive me when I'm larger than 240 bytes?
> 
> > 如果可能，您可以尝试以下参数：每 120 毫秒发送 640 字节。
> > 对于丢包情况，请检查 以确保没有传输失败。如果仍然有数据丢失，最好提供更详细的信息，例如 pcap 文件、设备日志和示例代码。`esp_zb_zcl_command_send_status_callback_t`

@lpy4105

lpy4105, 2025-03-11:
> Can the custom cluster send 640 bytes at a time?

Yes, it can, the APS layer could handle the fragmentation transmission.

> Why can't the coordinator receive me when I'm larger than 240 bytes

It might depend on the coordinator's receiving ability. What device do you use as a coordinator?

Mushan-bin, 2025-03-11:
The ESP32-H2 acts as a coordinator to receive device information.

[image]
The data received by the coordinator from the terminal device is processed by the callback function. @lpy4105

lpy4105, 2025-03-11:
What `data type` did you use for sending the request? Would you mind share the code of sending the message?

Mushan-bin, 2025-03-12:
[image]
The first thing I send is the length of the data. @lpy4105

Mushan-bin, 2025-03-12:
> [image]我发送的第一件事是数据的长度。

When the size of cmd_data exceeds 240 bytes, the message->data.size received by the coordinator is 1

lpy4105, 2025-03-12:
To send "CHAR_STRING", the first byte of the "value" should contain the length of the followed data, please refer to the ZCL spec.

To send arbitrary length of data, please try "ESP_ZB_ZCL_ATTR_TYPE_SET". But please keep in mind that this is just a workaround, and the behaviour might be changed in the future release.

chshu, 2025-03-20:
@Mushan-bin please reopen if any follow up questions.

closed 2025-03-20 by chshu