espressif/esp-zigbee-sdk #590: Time sync from time cluster not working

mundevx, 2025-03-07:
### Question

Hi,

I am following these steps.

1. I am subscribing on core actions:
esp_zb_core_action_handler_register(zb_action_handler);

2. Handling read attribute callback
static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void *message)
{
    ESP_LOGW(TAG, "Receive Zigbee action(0x%x) callback", callback_id);
    esp_err_t ret = ESP_OK;
    switch (callback_id) {
    case ESP_ZB_CORE_CMD_READ_ATTR_RESP_CB_ID:
        ret = zb_read_attr_resp_handler((esp_zb_zcl_cmd_read_attr_resp_message_t *)message);
        break;
    default:
        ESP_LOGE(TAG, "No handler for action(0x%x)", callback_id);
        break;
    }
    return ret;
}

3. write needed behavior
static esp_err_t zb_read_attr_resp_handler(const esp_zb_zcl_cmd_read_attr_resp_message_t *message)
{
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    ESP_RETURN_ON_FALSE(message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS, ESP_ERR_INVALID_ARG, TAG, "Received message: error status(%d)", message->info.status);

    esp_zb_zcl_read_attr_resp_variable_t *variable = message->variables;
    ESP_LOGI(TAG, "Read attribute response: status(%d), cluster(0x%x), attribute(0x%x), type(0x%x), value(%d)", 
            message->info.status,
            message->info.cluster,
            variable->attribute.id,
            variable->attribute.data.type,
            variable->attribute.data.value ? *(uint8_t *)variable->attribute.data.value : 0);

    if (message->info.dst_endpoint == ENDPOINT)
    {
        switch (message->info.cluster) 
        {
        case ESP_ZB_ZCL_CLUSTER_ID_TIME:
            switch(variable->attribute.id)
            {
                case ESP_ZB_ZCL_ATTR_TIME_LOCAL_TIME_ID:
                    time_t t = *(uint32_t*) variable->attribute.data.value + 946684800;
                    //TODO
                    break;
                default:
                    ESP_LOGE(TAG, "Message data: cluster(0x%x), attribute(0x%x)  ", message->info.cluster, variable->attribute.id);
            }
            break;
        default:
            ESP_LOGE(TAG, "Message data: cluster(0x%x), attribute(0x%x)  ", message->info.cluster, variable->attribute.id);
        }
    }
    return ESP_OK;
}

4. I am Calling this method after network steering on Router device

void read_server_time()
{
    uint16_t attributes[] = 
    {
        ESP_ZB_ZCL_ATTR_TIME_LOCAL_TIME_ID
    };

    esp_zb_zcl_read_attr_cmd_t read_req;
    read_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    read_req.attr_field = attributes;
    read_req.attr_number = sizeof(attributes) / sizeof(uint16_t);
    read_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_TIME;
    read_req.zcl_basic_cmd.dst_endpoint = ENDPOINT;
    read_req.zcl_basic_cmd.src_endpoint = ENDPOINT;
    read_req.zcl_basic_cmd.dst_addr_u.addr_short = 0x0000;  //coordinator
    ESP_LOGI(TIME_TAG, "Request read time attr");
    esp_zb_zcl_read_attr_cmd_req(&read_req);
}

But i am getting "(uint32_t*) variable->attribute.data.value" as NULL? What I am doing wrong?

xieqinan, 2025-03-12:
I believe the `esp_zb_zcl_read_attr_cmd_req()` command is correct. However, I need more context to analyze the issue further. Could you please provide details about your application? Specifically:  

- What is the background of your application?
- Who is acting as the coordinator?  
- How does the coordinator synchronize time with the Time Server cluster attribute?  
- What is the content of `esp_zb_zcl_cmd_read_attr_resp_message_t *message`?  

These details will help in identifying potential issues and providing a more accurate solution.

xieqinan, 2025-04-23:
Hi @mundevx ,

Do you have any updates to share?

mundevx, 2025-04-23:
Hi @xieqinan ,

Coordinator is Tuya.

xieqinan, 2025-06-13:
@mundevx ,

Please double-check whether the TuYa Coordinator supports the Timer Cluster as a **server**. We can confirm that `esp_zb_zcl_read_attr_cmd_req()` functions according to the specification.

Could you please provide more feedback on this comment: [https://github.com/espressif/esp-zigbee-sdk/issues/590#issuecomment-2716268780](https://github.com/espressif/esp-zigbee-sdk/issues/590#issuecomment-2716268780)? It may help resolve your issue.

xieqinan, 2025-08-12:
Hi @mundevx ,

Changing `esp_zb_zcl_read_attr_cmd_t read_req;` to `esp_zb_zcl_read_attr_cmd_t read_req = {0}` may be helpful, please give the change a test.

mundevx, 2025-08-12:
Yes, close this issue.