espressif/esp-zigbee-sdk #591: Is the reporting method appropriate

PASCAL-BeomGeun, 2025-03-08:
### Question

Configured End Device with ESP32-C6.

It consists of 2 temperatures, 1 humidity, battery voltage, and remaining battery capacity, and reports data as soon as it is joined to the network.

I wrote it like the code below, is there a way I can optimize it more than this?
Whether the value changes or not, the data will be sent unconditionally.

``
#include "esp_log.h"
#include "esp_check.h"
#include "esp_sleep.h"
#include "ha/esp_zigbee_ha_standard.h"

#include "app_main.h"
#include "app_zigbee.h"

static const char *TAG = "APP_ZIGBEE";

uint8_t network_initalize_fail_count = 0;
uint8_t network_steering_fail_count = 0;
extern wakeup_reason_t reason;

esp_zb_zcl_report_attr_cmd_t report_attr_cmd[] = {
    {
        .address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT,
        .attributeID = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID,
        .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI,
        .clusterID = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT,
        .zcl_basic_cmd.src_endpoint = MAIN_ENDPOINT
    },
    {
        .address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT,
        .attributeID = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID,
        .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI,
        .clusterID = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT,
        .zcl_basic_cmd.src_endpoint = ONE_ENDPOINT
    },
    {
        .address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT,
        .attributeID = ESP_ZB_ZCL_ATTR_REL_HUMIDITY_MEASUREMENT_VALUE_ID,
        .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI,
        .clusterID = ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT,
        .zcl_basic_cmd.src_endpoint = MAIN_ENDPOINT
    },
    {
        .address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT,
        .attributeID = ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_VOLTAGE_ID,
        .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI,
        .clusterID = ESP_ZB_ZCL_CLUSTER_ID_POWER_CONFIG,
        .zcl_basic_cmd.src_endpoint = MAIN_ENDPOINT
    },
    {
        .address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT,
        .attributeID = ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_PERCENTAGE_REMAINING_ID,
        .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI,
        .clusterID = ESP_ZB_ZCL_CLUSTER_ID_POWER_CONFIG,
        .zcl_basic_cmd.src_endpoint = MAIN_ENDPOINT
    }
};
#define REPORTING_ATTR_COUNT (sizeof(report_attr_cmd) / sizeof(report_attr_cmd[0]))

void update_sensor_value(int16_t temp1, int16_t temp2, uint16_t humi, uint8_t volt, uint8_t soc)
{
    esp_zb_lock_acquire(portMAX_DELAY);

    esp_zb_zcl_set_attribute_val(MAIN_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
            ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, &temp1, false);
    esp_zb_zcl_set_attribute_val(MAIN_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
[... 202 lines cut ...]

    /* Set primary channel set */
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    /* Start Zigbee stack */
    ESP_ERROR_CHECK(esp_zb_start(false));
    
    /* Main loop */
    esp_zb_stack_main_loop();
}

void zigbee_init(void)
{
    // Zigbee 플랫폼 설정
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };

    // Zigbee 플랫폼 설정 적용
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));

    xTaskCreate(esp_zb_task, "zigbee_task", 4096, NULL, 5, NULL);
}
""

xieqinan, 2025-03-10:
Hi @PASCAL-BeomGeun ,

> I wrote it like the code below, is there a way I can optimize it more than this?
Whether the value changes or not, the data will be sent unconditionally.

You can call `esp_zb_zcl_config_report_cmd_req()` to config the interval of reporting, please refer to [switch example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L112) or [temperature sensor example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor/main/esp_zb_temperature_sensor.c#L197).

PASCAL-BeomGeun, 2025-03-10:
I posted the wrong question.
Correct the question.

I update the attribute values using ``esp_zb_zcl_set_attribute_val`` and report the values for each attribute using ``esp_zb_zcl_report_attr_cmd_req``, is there any other way to report all the attribute values at once without using a repeat statement?

xieqinan, 2025-03-10:
> is there any other way to report all the attribute values at once without using a repeat statement?

Sorry, this feature is not supported. The current SDK (v1.6.4) only allows ZCL to report attributes separately, so you need to iterate through the reported attributes in application as you are currently doing.

PASCAL-BeomGeun, 2025-03-10:
Thank you.
It helped me a lot.

closed 2025-03-10 by PASCAL-BeomGeun