espressif/esp-zigbee-sdk #592: Question: Implementing coordinator function of binding devices connected to it

Shatodor, 2025-03-10:
I have coordinator and two devices connected to it, for example switch and light.  There are examples of binding inititated from switch or light. How do I implement binding between switch and light initiated from coordinator?

lpy4105, 2025-03-11:
You could send binding requests to these two devices from coordinator via `esp_zb_zdo_device_bind_req()`. Please be aware of the differences between `dst_address_u` and `req_dst_addr` in the request parameters.

Shatodor, 2025-03-11:
> You could send binding requests to these two devices from coordinator via `esp_zb_zdo_device_bind_req()`. Please be aware of the differences between `dst_address_u` and `req_dst_addr` in the request parameters.

Can you please explain difference between them?
Description in docs seems pretty the same to me: "The destination address for the binding entry" and "Destination address of the request send to". How can this adresses can be different?

Actually it looks like in next expample req_dst_addr set to adress of device sending bind requiest, but there is already src_address for that.
[examples/esp_zigbee_HA_sample/HA_color_dimmable_switch/main/esp_zb_switch.c](url)

Thanks for your help in advance.

lpy4105, 2025-03-12:
> Can you please explain difference between them?

Sure. 
`req_dst_addr` indicates the address the request will be sent to. In your case, should be the address of `light` or `switch` depending on which device you want to create binding on. 
`dst_address_u` is the "destination address" parameter of the binding. 

In other words, you could send request to `<req_dst_addr>` and request it to bind `<dst_address_u>`.

For example, you want to request `light` to bind `switch` from the coordinator, you could set `req_dst_addr = <light_address>` and `dst_address_u = <switch_address>`.

Shatodor, 2025-03-12:
Tried some adresses combinations with no success. To filnaly cliryfy: 
I have light with `<light_addr_short>`,` <light_addr_long>.` `<light_endp>` and switch with  `<switch_addr_short>`,` <switch_addr_long>`,  `<switch_endp>` . How I set parameters of bind_req to initiate in from coordinaor?

```
esp_zb_zdo_bind_req_param_t bind_req;

bind_req.src_address = ???;
bind_req.src_endp =  ???;

bind_req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;

bind_req.dst_address_u.addr_long = switch_addr_long;
        
bind_req.dst_endp = light_endp;
bind_req.req_dst_addr = light_addr_shortt;

esp_zb_zdo_device_bind_req(&bind_req, bind_cb, (void *) & (info_ctx));
```
`memcpy` deliberately omitted.

`bind_req.dst_address_u.addr_short ` should be set? In expamples it isn't.

lpy4105, 2025-03-13:
To request `light` bind `switch`:
```
esp_zb_zdo_bind_req_param_t bind_req;

bind_req.src_address = <light_addr_long>; /* Binding is from light */
bind_req.src_endp =  <light_endp>;        /* Binding is from light */

bind_req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;

bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
bind_req.dst_address_u.addr_long = <switch_addr_long>; /* Binding is to switch */
bind_req.dst_endp = <switch_endp>;                     /* Binding is to switch */

bind_req.req_dst_addr = <light_addr_short>; /* The request is to light */

esp_zb_zdo_device_bind_req(&bind_req, bind_cb, (void *) & (info_ctx));
```

BTW, you could use `esp_zb_zdo_binding_table_req()` to query the binding table from remote device, this would help you better understand the binding parameters.

closed 2025-03-20 by chshu