espressif/esp-zigbee-sdk #593: esp_zb_secur_ic_only_enable() - Co-ordinator only?

OOHehir, 2025-03-10:
### Question

Hello, my code is based on the HA_on_off_light example. I'm trying to get the 'install code' method of pairing working & have made the changes that you have previously provided as a patch, namely:
`
#if INSTALLCODE_POLICY_ENABLE == true

  ESP_ERROR_CHECK(esp_zb_set_long_address(zigbee_long_address));
  ESP_ERROR_CHECK(esp_zb_secur_ic_set(ESP_ZB_IC_TYPE_64, install_code));
  ESP_ERROR_CHECK(esp_zb_secur_ic_only_enable(true)); // results in error: undefined reference to esp_zb_secur_ic_only_enable

#endif
`
I've been trying to force the use of install codes when pairing but using the esp_zb_secur_ic_only_enable(true) results in the build error undefined reference to 'esp_zb_secur_ic_only_enable'. Is this function for co-ordinators only? Is there any method to know whether has joined using the install code? I'm using a Conbee as a co-ordinator so that maybe an issue.

Thanks,

xieqinan, 2025-03-11:
> 'esp_zb_secur_ic_only_enable'. Is this function for co-ordinators only? 

Yes, it is.

>  Is there any method to know whether has joined using the install code? I'm using a Conbee as a co-ordinator so that maybe an issue.

I don't think there is a very intuitive way to determine the device's joining method just from the printed logs. However, you can observe the sniffer: if the device uses the preconfigured key generated by the install code, the frames will be encrypted and cannot be parsed using the default `Alliance09` key.

OOHehir, 2025-03-12:
@xieqinan Thanks for the quick reply & suggestion. Could I perhaps suggest that this information would be visible if [debug](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#enable-debug-mode-and-trace-logging) of the Zigbee library was enabled?

closed 2025-03-12 by OOHehir

xieqinan, 2025-03-13:
> Could I perhaps suggest that this information would be visible if [debug](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#enable-debug-mode-and-trace-logging) of the Zigbee library was enabled?

There doesn't seem to be a visible log indicating which key is used to parse the `Transport Key` command for non-coordinator devices. As we know, the install code is used to generate a standard APS link key, which is essentially the same as the default `Alliance09` key.