espressif/esp-zigbee-sdk #595: Zigbee Deepsleep Rejoin duration

SUHAN447, 2025-03-11:
### Question

I am using an ESP32-C6 as a gateway and an ESP32-H2 as an end device. These end devices send data over APS and then enter deep sleep. However, when they wake up, the reconnection time is very long (>4 seconds). This has become a major problem for me. Is there a way to shorten this reconnection time?

xieqinan, 2025-03-11:
Deep sleep will cause all peripherals, including NVRAM, to go offline. This means that when the device wakes up, it will be factory new and require a complete joining procedure. You can set `esp_zb_set_primary_network_channel_set()` to a single channel instead of `ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK` to reduce the channel scan duration.

xieqinan, 2025-03-12:
Sorry, my mistake. The previous comment [here](https://github.com/espressif/esp-zigbee-sdk/issues/595#issuecomment-2713698241) contains some incorrect conclusions. I have corrected it. Please review it again.

SUHAN447, 2025-03-12:
This is absolutely correct; I am achieving a rejoin time of less than 500 ms. Additionally, my total time has been reduced to 1 second. Thank you very much!

closed 2025-03-12 by SUHAN447