espressif/esp-zigbee-sdk #596: Question: How to intercept "Off with effect" command to "On/Off" cluster

wejn, 2025-03-15:
### Question

By default the stack handles `Off with effect` (0x40) for `On/Off` cluster (0x6) by triggering `ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID` (set attribute value callback).

Unfortunately the supplemental info sent with the `Off with effect`:

[image]

namely the `Effect identifier` and `Effect variant` are thus lost.

Is there an easy way to surface (selectively) this info?

I have found some info in https://github.com/espressif/esp-zigbee-sdk/issues/75#issuecomment-1687387855 that the `esp_zb_zcl_add_privilege_command()` could be used.

Is that the right approach? Or is there a better one?

xieqinan, 2025-03-17:
> Is that the right approach? Or is there a better one?

If you intend to obtain the `Effect identifier` and `Effect variant`, I believe `esp_zb_zcl_add_privilege_command()` can be used. However, please pay attention to its return value. If it returns `true`, it indicates that the command has been handled by you, and the stack will skip processing the command.

wejn, 2025-03-17:
Would it be possible for you to share the default handler for OffWithEffect?

I’m torn between two approaches:
- return true, but then also handle default scene, onoff value, (and possibly other things?)
- return false, record the additional values, and then try taking them into consideration in the set attributes callback

Seeing what I’m up against in the real handler would be nice…

xieqinan, 2025-03-18:
> Would it be possible for you to share the default handler for OffWithEffect?

The default handler is based on the following description:

[image]

wejn, 2025-03-18:
Re description: yeah, not what I had in mind. I can read the ZCL spec. Anyway, I'll find some way without default impl, probably the second approach outlined in the comment above yours.

---

In any case, I checked the `esp_zb_zcl_add_privilege_command`, and I don't believe that's applicable here.

Because return from that `ESP_ZB_CORE_CMD_PRIVILEGE_COMMAND_REQ_CB_ID` callback is of type `esp_err_t`.

However, in #597 you recommended `esp_zb_raw_command_handler_register` which might fit the bill (signature-wise for sure).

wejn, 2025-03-18:
Anyway, I just confirmed it, raw command handler does the trick:

``` c
// [...]

typedef ZB_PACKED_PRE struct my_off_with_effect_cmd_req_s
{
  zb_uint8_t effect_id; // Effect identifier
  zb_uint8_t effect_variant; // Effect variant
} ZB_PACKED_STRUCT
my_off_with_effect_cmd_req_t;

bool zb_raw_command_handler(uint8_t bufid) {
    uint8_t buf[zb_buf_len(bufid)];
    zb_zcl_parsed_hdr_t *cmd_info = ZB_BUF_GET_PARAM(bufid, zb_zcl_parsed_hdr_t);
    memcpy(buf, zb_buf_begin(bufid), sizeof(buf));

    if (cmd_info->addr_data.common_data.dst_endpoint == MY_LIGHT_ENDPOINT) {
        // [...]

        if (cmd_info->cluster_id == ESP_ZB_ZCL_CLUSTER_ID_ON_OFF &&
                cmd_info->cmd_id == ESP_ZB_ZCL_CMD_ON_OFF_OFF_WITH_EFFECT_ID) {
               my_off_with_effect_cmd_req_t *req = (my_off_with_effect_cmd_req_t *)buf;
               ESP_LOGI(TAG, "O.w.E. detected: effect: 0x%02x, variant: 0x%02x", req->effect_id, req->effect_variant);
               // FIXME: Make use of this
        }
    }

    return false;
}

// [...]

static void esp_zb_task(void *pvParameters)
{   
    // initialize Zigbee stack (just for context)
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZR_CONFIG();
    esp_zb_init(&zb_nwk_cfg); 

   // [...]

    esp_zb_device_register(light_ep); // just for context
    esp_zb_core_action_handler_register(zb_action_handler); // just for context
    esp_zb_raw_command_handler_register(zb_raw_command_handler); // IMPORTANT
   
    // [...] 
    
    ESP_ERROR_CHECK(esp_zb_start(false)); // just for context...
    esp_zb_stack_main_loop(); 
}
```

From my log:

``` text
// Sending OffWithEffect
I (22814) MAIN: O.w.E. detected: effect: 0x00, variant: 0x02
I (22814) SCENES: Store scene 0 for group 0
+-------+----------+----------+-----------------+-------------------------------------+
| Index | Group ID | Scene ID | Transition Time |   [Cluster ID | Extension Field]   |
+-------+----------+----------+-----------------+-------------------------------------+
|   00  |  0x0000  |   0x00   |      0000       | [ 0x0300 | 0x6e  0x1 ] | [ 0x0008 | 0xfe ] | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
I (22854) MAIN: Light turns off
```

closed 2025-03-18 by wejn