espressif/esp-zigbee-sdk #597: Question: How to intercept which nodes are sending "Read attributes" commands to our ED?

wejn, 2025-03-15:
### Question

I'm trying to figure out if there's a way to intercept (gather) data (or aggregate statistics) as to what network nodes are sending my end device `ZCL: Read attributes` commands?

xieqinan, 2025-03-17:
> I'm trying to figure out if there's a way to intercept (gather) data (or aggregate statistics) as to what network nodes are sending my end device ZCL: Read attributes commands?

The `esp_zb_raw_command_handler_register()` may be help you, the usage can refer to https://github.com/espressif/esp-zigbee-sdk/issues/526#issuecomment-2572588627.

wejn, 2025-03-18:
Thank you, that's very useful.

closed 2025-03-18 by wejn