espressif/esp-zigbee-sdk #598: Multiple On Off Light Endpoints not discovered by Alexa

rvanil-yahoo, 2025-03-16:
### Question

Hello Folks,

I use the Zigbee_On_Off_Light.ino example that comes with esp32 board manager to add second light in the hopes to get one end device to have multiple endpoints. I initialized the new light like shown below:

ZigbeeLight zbLight = ZigbeeLight(ZIGBEE_LIGHT_ENDPOINT);
ZigbeeLight zbLight2 = ZigbeeLight(ZIGBEE_LIGHT_ENDPOINT + 1);

I made more changes like shown below.

  zbLight.setManufacturerAndModel("Espressif", "ZBLightBulb");
  zbLight2.setManufacturerAndModel("Espressif2", "ZBLightBulb2");

  // Set callback function for light change
  zbLight.onLightChange(setLED);
  zbLight2.onLightChange(setLED);

  //Add endpoint to Zigbee Core
  Serial.println("Adding ZigbeeLight endpoint to Zigbee Core");
  Zigbee.addEndpoint(&zbLight);
  Zigbee.addEndpoint(&zbLight2);

Rest of the code is same as in the example.

When I use Alexa Echo Plus 1st generation (really old one). I successfully discovers the first one, but not the second light bulb. Not sure what I am doing wrong here to get multiple endpoints discovered by Alexa natively. Please help.

Thanks
Anil

xieqinan, 2025-03-17:
Hi,

I reviewed the above code, and it seems fine. Could you please provide more information, such as Alexa logs, ESP logs, and sniffer captures, to help debug this issue?

rvanil-yahoo, 2025-03-25:
Thank you for responding, I am using arduino ide to figure out what is happening, unfortunately I dont have alexa logs. Is there any documentation anywhere regarding sniffer captures? I can try to set that up, meanwhile I am trying to compile and flash using end device debug mode.

rvanil-yahoo, 2025-03-25:
Here are some logs that I got from end device debug mode.

18:56:41.170 -> ZBOSS: mac/mac.c:1039   Oops - error beacon payload is too small buf_sz 15 - drop it
18:56:41.864 -> ZBOSS: mac/mac.c:1039   Oops - error beacon payload is too small buf_sz 15 - drop it
18:56:41.864 -> ZBOSS: mac/mac.c:1039   Oops - error beacon payload is too small buf_sz 15 - drop it
18:56:41.864 -> ZBOSS: mac/mac.c:1039   Oops - error beacon payload is too small buf_sz 15 - drop it
18:56:42.091 -> ZBOSS: nwk/nwk_join.c:186   low lqi 0
18:56:42.130 -> ZBOSS: nwk/nwk_join.c:186   low lqi 0
18:56:42.130 -> ZBOSS: nwk/nwk_join.c:186   low lqi 0
18:56:42.130 -> ZBOSS: nwk/nwk_join.c:186   low lqi 0
18:56:42.130 -> ZBOSS: nwk/nwk_join.c:186   low lqi 0
18:56:42.130 -> ZBOSS: nwk/nwk_join.c:293   Will assoc to pan 0xcfb2 on channel 11, dev 0x0, iface_id 0
18:56:42.650 -> ZBOSS: mac/mac_associate.c:448   received association response
18:56:42.650 -> ZBOSS: zdo/zdo_app_join.c:771   CONGRATULATIONS! joined status 0, addr 0xca5e, iface_id 0, ch 11, rejoin 0
18:56:42.650 -> ZBOSS: zdo/zdo_app_join.c:773   xpanid <mac address>
18:56:42.650 -> ZBOSS: secur/zdo_secur.c:211   NWK key <long key>
18:56:42.683 -> ZBOSS: nwk/nwk_main.c:3775   Put 21 into input queue (blocked by 20)
18:56:42.683 -> ZBOSS: nwk/nwk_main.c:2360   set input_blocked_by and process 21 from in queue
18:56:42.751 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 0
18:57:54.359 -> ZBOSS: nwk/nwk_main.c:2339   Drop packet with improper protocol version: 0x5

rvanil-yahoo, 2025-03-25:
19:02:03.054 -> ZBOSS: nwk/nwk_join.c:186   low lqi 0
19:02:03.054 -> ZBOSS: nwk/nwk_join.c:186   low lqi 0
19:02:03.054 -> ZBOSS: nwk/nwk_join.c:186   low lqi 0
19:02:03.054 -> ZBOSS: nwk/nwk_join.c:186   low lqi 0
19:02:03.054 -> ZBOSS: nwk/nwk_join.c:186   low lqi 0
19:02:03.054 -> ZBOSS: nwk/nwk_join.c:186   low lqi 0
19:02:03.054 -> ZBOSS: nwk/nwk_join.c:293   Will assoc to pan 0xcfb2 on channel 11, dev 0x0, iface_id 0
19:02:03.553 -> ZBOSS: mac/mac_associate.c:448   received association response
19:02:03.553 -> ZBOSS: zdo/zdo_app_join.c:771   CONGRATULATIONS! joined status 0, addr 0xca5e, iface_id 0, ch 11, rejoin 0
19:02:03.586 -> ZBOSS: zdo/zdo_app_join.c:773   xpanid [mac_address]
19:02:03.586 -> ZBOSS: secur/zdo_secur.c:211   NWK key [key]
19:02:03.619 -> ZBOSS: nwk/nwk_main.c:3775   Put 25 into input queue (blocked by 3)
19:02:03.619 -> ZBOSS: nwk/nwk_main.c:2360   set input_blocked_by and process 25 from in queue
19:02:03.619 -> ZBOSS: nwk/nwk_main.c:3775   Put 6 into input queue (blocked by 5)
19:02:03.619 -> ZBOSS: nwk/nwk_main.c:2360   set input_blocked_by and process 6 from in queue
19:02:03.619 -> ZBOSS: nwk/nwk_main.c:3775   Put 8 into input queue (blocked by 7)
19:02:03.654 -> ZBOSS: nwk/nwk_main.c:2360   set input_blocked_by and process 8 from in queue
19:02:03.686 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 0

rvanil-yahoo, 2025-03-25:
ZBOSS is not advertising 2 lights. Here is what I see in the serial monitor after i restart the ESP32-C6

19:10:11.538 -> ZBOSS: common/zb_init_default.c:181   ED build
19:10:11.538 -> ZBOSS: common/zb_init_default.c:187   sizes: g_zb 7576 sched 224 bpool 16 nwk 912 aps 3040 addr 284 zdo 840
19:10:11.538 -> ZBOSS: common/zb_init_default.c:196   sec 2
19:10:11.538 -> ZBOSS: common/zb_init_default.c:198   zcl 768
19:10:11.538 -> ZBOSS: common/zb_init_default.c:201   zll 960
19:10:11.538 -> ZBOSS: common/zb_init_default.c:207   nvram 140
19:10:11.538 -> ZBOSS: common/zb_init_default.c:210   buttons 200
19:10:11.578 -> ZBOSS: common/zb_init_default.c:212   err_ind 4
19:10:11.578 -> ZBOSS: common/zb_init_default.c:218   scheduler q size 80
19:10:11.578 -> ZBOSS: common/zb_init_default.c:221   g_mac 816 g_imac 28
19:10:11.578 -> ZBOSS: common/zb_init_default.c:225   Configurable mem build, use ZBOSS lib defaults = 0
19:10:11.578 -> ZBOSS: common/zb_init_default.c:231   ZB_IOBUF_POOL_SIZE 80 ZB_NWK_IN_Q_SIZE 40 ZB_MAC_PENDING_QUEUE_SIZE 20 ZB_APS_DST_BINDING_TABLE_SIZE 16 ZB_APS_BIND_TRANS_TABLE_SIZE 20 ZB_N_APS_RETRANS_ENTRIES 27
19:10:11.611 -> ZBOSS: common/zb_init_default.c:261   ZB_N_APS_KEY_PAIR_ARR_MAX_SIZE 8 ZB_IEEE_ADDR_TABLE_SIZE 64 ZB_NEIGHBOR_TABLE_SIZE 64 ZB_NWK_ROUTING_TABLE_SIZE 0 ZB_APS_DUPS_TABLE_SIZE 32
19:10:11.611 -> ZBOSS: zdo/af_descriptor.c:357   ep_count 1
19:10:11.611 -> ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
19:10:11.651 -> ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
19:10:11.651 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
19:10:11.651 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 5 comm status 0
19:10:11.651 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1

19:10:14.429 -> ZBOSS: nwk/nwk_join.c:251   No dev for join
19:10:14.429 -> ZBOSS: zdo/zdo_commissioning.c:210   Can't find PAN to join to! param 0
19:10:14.429 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 3
19:10:15.432 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1

rvanil-yahoo, 2025-03-25:
Is it even possible to have two lights on single end device with esp32 c6?

xieqinan, 2025-03-25:
> Is there any documentation anywhere regarding sniffer captures?

Yes, you can refer to [docs](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark).

xieqinan, 2025-03-25:
> Is it even possible to have two lights on single end device with esp32 c6?

Yes, it MUST be possible.

xieqinan, 2025-03-25:
> ZBOSS: zdo/af_descriptor.c:357 ep_count 1

This log indicates that only a light endpoint is registered by your application.  

Could you iterate through `_zb_ep_list` and print the data model [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/Zigbee/src/ZigbeeCore.cpp#L148C34-L148C45)? I think that might help identify the issue.

rvanil-yahoo, 2025-03-25:
Let me do that and get back to to you.

rvanil-yahoo, 2025-03-25:
I flashed with 2 endpoints code and I see this in the log. I think alexa is not detecting the second one as it stop discovery after discovering first device.

16:51:26.325 -> 16 g_imac 28
16:51:26.325 -> ZBOSS: common/zb_init_default.c:225   Configurable mem build, use ZBOSS lib defaults = 0
16:51:26.325 -> ZBOSS: common/zb_init_default.c:231   ZB_IOBUF_POOL_SIZE 80 ZB_NWK_IN_Q_SIZE 40 ZB_MAC_PENDING_QUEUE_SIZE 20 ZB_APS_DST_BINDING_TABLE_SIZE 16 ZB_APS_BIND_TRANS_TABLE_SIZE 20 ZB_N_APS_RETRANS_ENTRIES 27
16:51:26.325 -> ZBOSS: common/zb_init_default.c:261   ZB_N_APS_KEY_PAIR_ARR_MAX_SIZE 8 ZB_IEEE_ADDR_TABLE_SIZE 64 ZB_NEIGHBOR_TABLE_SIZE 64 ZB_NWK_ROUTING_TABLE_SIZE 0 ZB_APS_DUPS_TABLE_SIZE 32
**16:51:26.325 -> ZBOSS: zdo/af_descriptor.c:357   ep_count 2**
16:51:26.362 -> ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
16:51:26.362 -> ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
16:51:26.362 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
16:51:26.362 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 5 comm status 0
16:51:26.401 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1

16:51:32.289 -> ZBOSS: nwk/nwk_join.c:293   Will assoc to pan 0xcfb2 on channel 11, dev 0x0, iface_id 0
16:51:32.806 -> ZBOSS: mac/mac_associate.c:448   received association response
16:51:32.806 -> ZBOSS: zdo/zdo_app_join.c:771   CONGRATULATIONS! joined status 0, addr 0xd328, iface_id 0, ch 11, rejoin 0
16:51:32.806 -> ZBOSS: zdo/zdo_app_join.c:773   xpanid [mac]
16:51:32.806 -> ZBOSS: secur/zdo_secur.c:211   NWK key [somekey]
16:51:32.989 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 0
16:51:36.513 -> ZBOSS: nwk/nwk_main.c:2339   Drop packet with improper protocol version: 0x5
16:51:41.410 -> ZBOSS: zdo/af_rx.c:463   APS pkt 2 for ep 1 is not recognized - drop

when I delete the device in light switch, i see the following logs.

16:55:33.057 -> ZBOSS: nwk/nwk_main.c:4505   zb_nwk_do_leave param 9 rejoin 0
16:55:33.057 -> ZBOSS: common/zb_address.c:1138   zb_address_delete ref 8
16:55:33.057 -> ZBOSS: common/zb_address.c:1138   zb_address_delete ref 8
16:55:33.097 -> ZBOSS: common/zb_address.c:1146   zb_address_delete error: entry is not used
16:55:33.097 -> ZBOSS: common/zb_address.c:1138   zb_address_delete ref 0
16:55:33.161 -> ESP-ROM:esp32c6-20220919
16:55:33.161 -> Build:Sep 19 2022
16:55:33.161 -> rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
16:55:33.161 -> Saved PC:0x4001975a
16:55:33.161 -> SPIWP:0xee
16:55:33.161 -> mode:DIO, clock div:2
16:55:33.161 -> load:0x40875720,len:0x1228
16:55:33.161 -> load:0x4086c110,len:0xd9c
16:55:33.161 -> load:0x4086e610,len:0x2f74
16:55:33.161 -> entry 0x4086c110
16:55:33.571 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x20 attr_value 0x4087e18f
16:55:33.603 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 1
16:55:33.603 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x30 attr_value 0x4087e18e
16:55:33.603 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 1
16:55:33.603 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x21 attr_value 0x4087e18e
16:55:33.603 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 2
16:55:33.640 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x18 attr_value 0x4087e18f
16:55:33.640 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 1
16:55:33.640 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x20 attr_value 0x4087e18f
16:55:33.640 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 1
16:55:33.640 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x20 attr_value 0x4087e18e
16:55:33.675 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 1
16:55:33.675 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x21 attr_value 0x4087e18c
16:55:33.675 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 2
16:55:33.675 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x10 attr_value 0x4087e18b
16:55:33.713 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 1
16:55:33.713 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x18 attr_value 0x4087e18a
16:55:33.713 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 1
16:55:33.713 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x10 attr_value 0x4087e18f
16:55:33.713 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 1
16:55:33.713 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x20 attr_value 0x4087e19f
16:55:33.748 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 1
16:55:33.748 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x30 attr_value 0x4087e19e
16:55:33.748 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 1
16:55:33.748 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x21 attr_value 0x4087e19e
[... cut ...]
16:55:34.092 -> ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
16:55:34.092 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
16:55:34.125 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 5 comm status 0
16:55:34.125 -> ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1

xieqinan, 2025-03-26:
> I flashed with 2 endpoints code and I see this in the log. I think alexa is not detecting the second one as it stop discovery after discovering first device.

Okay, it would be best to use a sniffer to capture the communication process between Alexa and the ESP device to confirm the above assumption.  

Could you please share what changes you made that allowed the device to register two endpoints compared to before?

rvanil-yahoo, 2025-03-26:
Here is the entire code.

#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

#include "Zigbee.h"

/* Zigbee light bulb configuration */
#define ZIGBEE_LIGHT_ENDPOINT 10
#define ZIGBEE_LIGHT_ENDPOINT2 11

uint8_t led = RGB_BUILTIN;
uint8_t button = BOOT_PIN;

ZigbeeLight zbLight = ZigbeeLight(ZIGBEE_LIGHT_ENDPOINT);
ZigbeeLight zbLight2 = ZigbeeLight(ZIGBEE_LIGHT_ENDPOINT2);

/********************* RGB LED functions **************************/
void setLED(bool value) {
  digitalWrite(led, value);
}

/********************* Arduino functions **************************/
void setup() {
  Serial.begin(115200);

  // Init LED and turn it OFF (if LED_PIN == RGB_BUILTIN, the rgbLedWrite() will be used under the hood)
  pinMode(led, OUTPUT);
  digitalWrite(led, LOW);

  // Init button for factory reset
  pinMode(button, INPUT_PULLUP);

  //Optional: set Zigbee device name and model
  zbLight.setManufacturerAndModel("Espressif", "ZBLightBulb");
  zbLight2.setManufacturerAndModel("Espressif", "ZBLightBulb");

  // Set callback function for light change
  zbLight.onLightChange(setLED);
  zbLight2.onLightChange(setLED);

  //Add endpoint to Zigbee Core
  Serial.println("Adding ZigbeeLight endpoint to Zigbee Core");
  Zigbee.addEndpoint(&zbLight);
  Zigbee.addEndpoint(&zbLight2);

  // When all EPs are registered, start Zigbee. By default acts as ZIGBEE_END_DEVICE
  if (!Zigbee.begin()) {
    Serial.println("Zigbee failed to start!");
    Serial.println("Rebooting...");
    ESP.restart();
  }
  Serial.println("Connecting to network");
  while (!Zigbee.connected()) {
    Serial.print(".");
    delay(100);
  }
  Serial.println();
}

void loop() {
  // Checking button for factory reset
  if (digitalRead(button) == LOW) {  // Push button pressed
    // Key debounce handling
    delay(100);
    int startTime = millis();
    while (digitalRead(button) == LOW) {
      delay(50);
      if ((millis() - startTime) > 3000) {
        // If key pressed for more than 3secs, factory reset Zigbee and reboot
        Serial.println("Resetting Zigbee to factory and rebooting in 1s.");
        delay(1000);
        Zigbee.factoryReset();
      }
    }
    // Toggle light by pressing the button
    zbLight.setLight(!zbLight.getLightState());
  }
  delay(100);
}

xieqinan, 2025-03-27:
The above implementation seems similar to the original issue: https://github.com/espressif/esp-zigbee-sdk/issues/598#issue-2922757826  

Anyway, do you have any other issues related to this ticket?

xieqinan, 2025-06-03:
Hi @rvanil-yahoo,

> I flashed with 2 endpoints code and I see this in the log. I think alexa is not detecting the second one as it stop discovery after discovering first device.

Please double-check this behavior of Alexa, if the issue is from the Alexa, please consider closing this issue.

spacejumper, 2025-06-07:
Hey, I ran into the same issue on the ESP32-H2 with Zigbee2MQTT on Home Assistant. I tried the code above, but it still doesn't work.