espressif/esp-zigbee-sdk #600: ZDO signal fluctuates

Mushan-bin, 2025-03-17:
When I set the transmitting power of the terminal device to 20, when sending a large packet of data less than 5 meters away from the terminal and the coordinator, sometimes the terminal will give a ZDO signal: NLME Status Indication (0x32), status: ESP_OK, and then packet loss will occur on the coordinator side. ZDO signal: NLME Status Indication (0x32), status: ESP_OK How to avoid this phenomenon.

xieqinan, 2025-03-18:
Please follow the [comment](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zdo/esp_zigbee_zdo_common.h#L478) on the NLME Status Indication to debug the cause of the log first.

xieqinan, 2025-04-23:
Hi @Mushan-bin ，

Are there any remaining related questions you'd like to discuss?  If not, kindly consider closing this issue.

closed 2025-05-21 by Mushan-bin