espressif/esp-zigbee-sdk #601: Zigbee tx signal power

diazmanuel, 2025-03-18:
### Question

I'm trying to adjust the power parameters of the ZigBee transmitter, and I'm having some questions. I understand there's also a lack of documentation on this topic.
Below are the following questions on this topic:

1. What are the possible values ​​that the esp_zb_set_tx_power function takes? Both maximum and minimum values ​​and steps between valid values.
2. What unit is used by the esp_zb_set_tx_power function? According to the documentation, it says DB, but this is a unit of gain, not power. It would make more sense for it to be dBm.
3. In case the unit of the power setting function is in DB (gain), how much output power does the maximum gain set in this function represent?
4. In my projects, I use several wireless connection protocols simultaneously, such as Zigbee, BLE, and Wi-Fi.
Does working with multiple protocols reduce the maximum output power of the Zigbee signal? If so, would reducing the power of other protocols, such as BLE and Wi-Fi, achieve any benefits?

xieqinan, 2025-03-19:
> What are the possible values ​​that the esp_zb_set_tx_power function takes? Both maximum and minimum values ​​and steps between valid values.

Please refer to [this comment](https://gitlab.espressif.cn:6688/thread_zigbee/esp-zboss/-/blob/master/components/esp_zb_sdk/include/esp_zigbee_core.h?ref_type=heads#L554). The `IEEE802154_TXPOWER_VALUE_MIN` and `IEEE802154_TXPOWER_VALUE_MAX` values vary depending on the chip model. You can find these values in the ESP-IDF `hal` component, such as in [ieee802154_ll.h](https://github.com/espressif/esp-idf/blob/master/components/hal/esp32h2/include/hal/ieee802154_ll.h#L18).

> What unit is used by the esp_zb_set_tx_power function? According to the documentation, it says DB, but this is a unit of gain, not power. It would make more sense for it to be dBm.

This is an incorrect comment; the correct unit is dBm.

> In case the unit of the power setting function is in DB (gain), how much output power does the maximum gain set in this function represent?

The  same above.

> In my projects, I use several wireless connection protocols simultaneously, such as Zigbee, BLE, and Wi-Fi.
Does working with multiple protocols reduce the maximum output power of the Zigbee signal? If so, would reducing the power of other protocols, such as BLE and Wi-Fi, achieve any benefits?

If multiple protocols run on the same ESP chip, each protocol will maintain its own TX power and transmit frames using a time-division multiplexing method on the same antenna. As a result, the TX power of each protocol will not affect the others. For more details, refer to [ESP-IDF Coexistence Guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32h2/api-guides/coexist.html).

xieqinan, 2025-04-23:
Hi @diazmanuel ,

Did the above answers help resolve your question?   Please feel free to discuss any Zigbee-related issues here.  
If the issue has been resolved, kindly consider closing it.

diazmanuel, 2025-04-23:
yes, thanks you very much

closed 2025-04-23 by diazmanuel

xieqinan, 2025-05-07:
@josef2600 ,

>  i would say if possible, please add the <hal/ieee802154_ll.h> to <esp_zigbee_core.h>.

The `esp_zigbee_core.h` header only provides Zigbee-related API support. Since `IEEE802154_TXPOWER_VALUE_MAX` belongs to the IEEE802154 component, I believe it's better to include `<hal/ieee802154_ll.h>` at the application layer based on the requirement of user.

josef2600, 2025-05-07:
@xieqinan 
thank you. yes you are correct, i didn't thought about that one! but then again, how the programmers are to know what to add to where in the code? as i said, you are correct and it is logical step, but, we can add a `ifdef` to the `esp_zigbee_core.h` to check if it is esp32-c6 or esp32-h2 or others. there are so much things in the include files that i get confused all the time.
but, in this situation, you know the best how to act and what to do. i am not the handler for IDF, and it scares me! but i say what i think so maybe it can be useful. thank you for your answer.
best regards.