espressif/esp-zigbee-sdk #602: About zigbee 2.4G signal modulation

Mushan-bin, 2025-03-20:
How zigbee terminals continuously send a 2.4Ghz signal so that the spectrometer can continuously receive it to test the strength of the terminal's hardware RF circuits. Can you provide an example or solution?

Mushan-bin, 2025-03-21:
I want it to always produce 2.4GHZ signal, can always appear in the spectrometer, how to operate @chshu

Jayliu0712, 2025-03-25:
> [@Mushan-bin](https://github.com/Mushan-bin) are you looking for the continuously carrier signal, or transmitting Zigbee packets continuously (some idle time between each packet)?
> 
> For the second, you can use the [all_device_types_app](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_all_device_types_app) example and run the [iperf](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-console/README.md#iperf) test.

Hi，good afternoon！I’m looking for the continuously carrier signal，that can help me modulate the RF circuit signal through a spectrum analyzer to achieve optimal performance，can zigbee do this or not？If the answer is yes，could you please provide an example or solution？Thank you！

Jayliu0712, 2025-03-26:
> [@Mushan-bin](https://github.com/Mushan-bin) are you looking for the continuously carrier signal, or transmitting Zigbee packets continuously (some idle time between each packet)?
>
> For the second, [...]

@chshu I think my requirement is the former. So, how can I achieve it?

chshu, 2025-03-26:
@Mushan-bin @Jayliu0712, please refer to the RF test docs here: https://docs.espressif.com/projects/esp-test-tools/en/latest/esp32h2/development_stage/rf_test_items/zigbee_non_signaling_test.html#non-signaling-test

Jayliu0712, 2025-03-26:
@chshu Okay, I've solved this problem. Thank you！

closed 2025-03-26 by chshu