espressif/esp-zigbee-sdk #603: Windows covering type does not work properly

widapro, 2025-03-21:
### Question

When using the **ESP_ZB_ZCL_ATTR_WINDOW_COVERING_TYPE_TILT_BLIND_TILT_ONLY** type, the device is defined by the coordinator with LIFT attributes.

Coordinator: ZHA Home-Assistant
esp32-h2
ESP-IDF: v5.3.2
esp-zboss-lib: 1.6.4
esp-zigbee-lib: 1.6.4

How can I register my device as TILT only?

```
static void esp_zb_task(void *pvParameters)
{
    /* Initialize Zigbee stack with Zigbee end-device config */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG(); // Zigbee End Device config
    //esp_zb_sleep_enable(true);  // enable Zigbee light sleep
    esp_zb_init(&zb_nwk_cfg);
    //esp_zb_sleep_set_threshold(2000); // 2000 milliseconds

    uint8_t test_attr;
    test_attr = 0;

    /* basic cluster */
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, ESP_MANUFACTURER_NAME);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, ESP_MODEL_IDENTIFIER);

    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &test_attr);
    esp_zb_cluster_update_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
    
    /* identify cluster */
    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
    esp_zb_identify_cluster_add_attr(esp_zb_identify_cluster, ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID, &test_attr);

    /* window cover cluser */
    esp_zb_attribute_list_t *window_attr_list = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING);
    esp_zb_window_covering_cluster_cfg_t wc_config = {
            .covering_type   = ESP_ZB_ZCL_ATTR_WINDOW_COVERING_TYPE_TILT_BLIND_TILT_ONLY,
            .covering_status = ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CONFIG_TILT_CONTROL_IS_CLOSED_LOOP,
            .covering_mode   = ESP_ZB_ZCL_ATTR_WINDOW_COVERING_TYPE_MOTOR_IS_RUNNING_IN_MAINTENANCE_MODE,
        };
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_WINDOW_COVERING_TYPE_ID,
        ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &wc_config.covering_type);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CONFIG_STATUS_ID,
        ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &wc_config.covering_status);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_MODE_ID,
        ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &wc_config.covering_mode);
    
    uint8_t installed_open_limit_tilt = 100;
    uint8_t installed_closed_limit_tilt = 0;
    uint8_t current_position_tilt = 10;
    uint8_t phy_tilt_limit = 50;
    uint8_t intermediate_setpoints = 10;

    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_INSTALLED_OPEN_LIMIT_TILT_ID,
        ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &installed_open_limit_tilt);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_INSTALLED_CLOSED_LIMIT_TILT_ID,
        ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &installed_closed_limit_tilt);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_PHY_CLOSED_LIMIT_TILT_ID,
        ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &phy_tilt_limit);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_INTERMEDIATE_SETPOINTS_TILT_ID,
        ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &intermediate_setpoints);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_ID,
        ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &current_position_tilt);    

    /* create cluster lists for this endpoint */
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
[... 105 lines cut ...]
                "attribute": "ZCLAttributeDef(id=0x0014, name='velocity_lift', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read|Write: 3>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0017": {
                "attribute": "ZCLAttributeDef(id=0x0017, name='window_covering_mode', type=<flag 'WindowCoveringMode'>, zcl_type=<DataTypeId.map8: 24>, access=<ZCLAttributeAccess.Read|Write: 3>, mandatory=True, is_manufacturer_specific=False)",
                "value": 1
              },
              "0x0000": {
                "attribute": "ZCLAttributeDef(id=0x0000, name='window_covering_type', type=<enum 'WindowCoveringType'>, zcl_type=<DataTypeId.enum8: 48>, access=<ZCLAttributeAccess.Read: 1>, mandatory=True, is_manufacturer_specific=False)",
                "value": 7
              }
            },
            "unsupported_attributes": [
              8,
              9,
              16,
              17,
              "current_position_lift_percentage",
              "current_position_tilt_percentage",
              "installed_closed_limit_lift",
              "installed_open_limit_lift"
            ]
          }
}
```

xieqinan, 2025-03-21:
Please double-check the code below against the ZCL specification; there are some errors present.
```
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
                            ESP_ZB_ZCL_ATTR_WINDOW_COVERING_WINDOW_COVERING_TYPE_ID, ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM,
                            ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &wc_config.covering_type);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
                            ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CONFIG_STATUS_ID, ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM,
                            ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &wc_config.covering_status);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
                            ESP_ZB_ZCL_ATTR_WINDOW_COVERING_MODE_ID, ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM,
                            ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &wc_config.covering_mode);

    uint8_t installed_open_limit_tilt = 100;
    uint8_t installed_closed_limit_tilt = 0;
    uint8_t current_position_tilt = 10;
    uint8_t phy_tilt_limit = 50;
    uint8_t intermediate_setpoints = 10;

    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
                            ESP_ZB_ZCL_ATTR_WINDOW_COVERING_INSTALLED_OPEN_LIMIT_TILT_ID,
                            ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,
                            &installed_open_limit_tilt);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
                            ESP_ZB_ZCL_ATTR_WINDOW_COVERING_INSTALLED_CLOSED_LIMIT_TILT_ID,
                            ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,
                            &installed_closed_limit_tilt);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
                            ESP_ZB_ZCL_ATTR_WINDOW_COVERING_PHY_CLOSED_LIMIT_TILT_ID, ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM,
                            ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &phy_tilt_limit);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
                            ESP_ZB_ZCL_ATTR_WINDOW_COVERING_INTERMEDIATE_SETPOINTS_TILT_ID,
                            ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &intermediate_setpoints);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
                            ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_ID, ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM,
                            ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &current_position_tilt);
```

[image]

Such as:
```
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
                            ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_ID, ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM,
                            ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &current_position_tilt);
```
Correct it
```
    uint16_t current_position_tilt = 10;
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
                            ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_ID, ESP_ZB_ZCL_ATTR_TYPE_U16,
                            ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &current_position_tilt);
```
```

widapro, 2025-04-11:
Thank you @xieqinan for your response and for pointing me to the correct place in the spec that I missed. I have corrected the types according to the specification, but the device is still detected as a LIFT in Home Assistant and only provides position control, not tilt.

However, via Manage Zigbee Device in Home Assistant I can successfully send the command **"go_to_tilt_percentage"** and the device successfully receives and processes it.

[image]

[image]

Diagnostic info from Home-Assistant:
```
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 1429,
    "manufacturer": "ESPRESSIF",
    "model": "esp32h2",
    "name": "ESPRESSIF esp32h2",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4660,
    "power_source": "Mains",
    "lqi": 200,
    "rssi": -50,
    "last_seen": "2025-04-10T23:04:37",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": {
        "logical_type": 2,
        "complex_descriptor_available": 0,
        "user_descriptor_available": 0,
        "reserved": 0,
        "aps_flags": 0,
        "frequency_band": 8,
        "mac_capability_flags": 140,
        "manufacturer_code": 4660,
        "maximum_buffer_size": 108,
        "maximum_incoming_transfer_size": 1613,
        "server_mask": 11264,
        "maximum_outgoing_transfer_size": 1613,
        "descriptor_capability_field": 0
      },
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0202",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0102"
          ],
          "output_clusters": []
        }
      },
      "manufacturer": "ESPRESSIF",
      "model": "esp32h2"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "button.espressif_esp32h2_identify",
        "name": "ESPRESSIF esp32h2"
      },
      {
        "entity_id": "cover.espressif_esp32h2",
        "name": "ESPRESSIF esp32h2"
      },
      {
        "entity_id": "sensor.espressif_esp32h2_window_covering_type",
        "name": "ESPRESSIF esp32h2"
[... 288 lines cut ...]
    

    /* create cluster lists for this endpoint */
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_window_covering_cluster(esp_zb_cluster_list, window_attr_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = HA_WINDOW_COVERING_ENDPOINT,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = ESP_ZB_HA_WINDOW_COVERING_DEVICE_ID,
        .app_device_version = 0
    };

    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, endpoint_config);
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    ESP_ERROR_CHECK(esp_zb_start(true));
    esp_zb_stack_main_loop();
}
```

xieqinan, 2025-04-11:
>  but the device is still detected as a LIFT in Home Assistant and only provides position control, not tilt.

I'm not very familiar with the graphical composition of HA, so I'm not sure why the tilt is not displayed. Since you're able to control the tilt percentage, I believe the `ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_PERCENTAGE_ID` attribute has been added successfully.

If you could share how this graphical component is represented in Home Assistant, I'd be glad to continue investigating this issue with you.

xieqinan, 2025-05-16:
Hi @widapro ,

Do you have any updates?

xieqinan, 2025-06-03:
Hi @widapro ,

I believe that the issue has resolved by the https://github.com/espressif/esp-zigbee-sdk/issues/603#issuecomment-2742175983. If the problem still confuses you, feel free to reopen it.

closed 2025-06-03 by xieqinan