espressif/esp-zigbee-sdk #604: ESP32H2 and ZIGBEE sensors

Usama050, 2025-03-22:
### Question

Hello to all, I hope you guys are doing well. I have a ESP32-H2 Devkit-M1 and two ZigBee sensors one is door sensor(ZG-102ZL/ZG-102Z) and second one is a button(KOJIMA-SPB-ZGB-W). I am using ESP-IDF. I used this example "[https://github.com/espressif/esp-zigbee ... off_switch](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch)" and I connected my door sensor with ESP32H2 but I am not sure how can read it's output on serial monitor. I am using esp32h2 as a coordinator but I am not sure it is working well or not. if anyone know how to do it please let me know or share some video or blog link, if any one expert in this field please help. I think first of all I need to make esp32h2 work as a ZigBee coordinator and then sensors will automatically be connected with esp32h2. if anyone know how to do this task please help me. Thank you in advance.

Usama050, 2025-03-26:
@victpork, @pedrominatel , @kedars , @mahavirj any thing suggest me sir kindly!
I want to make an ESP32H2 a zigbee coordinator so some zigbee end devices can connect with this Coordinator and then I can read the data from sensor. Can anyone give me some example related to this task, because I am new with ZigBee and documentation is very hard for me to understand. Thank you

Usama050, 2025-03-26:
@[Daniel-Craciun](https://github.com/Daniel-Craciun) I guess bro you resolve similar problem. Kindly Help me

xieqinan, 2025-03-27:
Hi @Usama050 ,

You need to first determine the output value of your door sensor. I recommend using [Home Assistant](https://www.home-assistant.io/) to pair with your door sensor, as it provides an easier way to understand the Zigbee mechanism.

Usama050, 2025-04-03:
Hi @xieqinan my door sensor output is 0 or 1. So I can use IAS or BINARY cluster. do I need to make custom cluster like you guys did in [example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_thermostat/main/esp_zb_thermostat.c#L113) or not. just help me to make this cluster and end point.

mmackh, 2025-04-07:
The name of the door/contact sensor is Parasoll from Ikea - however it implements IAS. Is there an example somewhere of a contact switch implementing the steps to registering reporting like this: https://github.com/espressif/esp-zigbee-sdk/issues/462#issuecomment-2455337059?

Usama050, 2025-04-09:
I got the output from my door sensor using IAS Cluster now I need to refine my code 
I (129089) ESP_ZB_THERMOSTAT: Send 'read attributes' command
I (131049) ESP_ZB_THERMOSTAT: Read attribute response: from address(0xd170) src endpoint(1) to dst endpoint(1) cluster(0x500)
I (131049) ESP_ZB_THERMOSTAT: Read attribute response: status(0), cluster(0x500), attribute(0x2), type(0x19), value(1)
I (131059) ESP_ZB_THERMOSTAT: Read attribute response: status(0), cluster(0x500), attribute(0x2), type(0x19), value(1)
I (131069) ESP_ZB_THERMOSTAT: Read attribute response: status(134), cluster(0x500), attribute(0x3), type(0x0), value(0)
I (133129) ESP_ZB_THERMOSTAT: Send 'configure reporting' command
I (133129) ESP_ZB_THERMOSTAT: Send 'read attributes' command
I (134269) ESP_ZB_THERMOSTAT: Read attribute response: from address(0xd170) src endpoint(1) to dst endpoint(1) cluster(0x500)
I (134269) ESP_ZB_THERMOSTAT: Read attribute response: status(0), cluster(0x500), attribute(0x2), type(0x19), value(0)
I (134279) ESP_ZB_THERMOSTAT: Read attribute response: status(0), cluster(0x500), attribute(0x2), type(0x19), value(0)
I (134289) ESP_ZB_THERMOSTAT: Read attribute response: status(134), cluster(0x500), attribute(0x3), type(0x0), value(0)

Usama050, 2025-04-11:
@xieqinan Hello! Sir, how are you doing? I got the output of door sensor. Now I have to connect other sensors with my ESP32H2 ZigBee coordinator and my question is that, do I have to make new cluster function for each sensor or I can add parameters in same function for other sensors. 

[image]

and one more thing the data which I can get from my sensors how can I upload then into a server via URAT or other communication protocol.

xieqinan, 2025-04-11:
Hi,

> do I have to make new cluster function for each sensor or I can add parameters in same function for other sensors.

You can call `esp_zb_ep_list_add_gateway_ep()` to register all client clusters supported in the esp-zigbee-sdk, instead of creating a new one for other client clusters.

>  one more thing the data which I can get from my sensors how can I upload then into a server via URAT or other communication protocol.

Yes, you can upload your data via Wi-Fi if you're using Wi-Fi-enabled chips such as the [ESP32-C6](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitm-1/user_guide.html) or the [ESP-Zigbee-Gateway board](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html). Please refer to the [gateway example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway) in the esp-zigbee-sdk.

If you plan to upload data via UART instead, simply add the UART driver. You can refer to the [UART example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart) in the esp-idf.

Usama050, 2025-04-11:
@xieqinan Hi, so this command "esp_zb_ep_list_add_gateway_ep()" can register all the client cluster and make their endpoint. Thanks I can give a try.

Usama050, 2025-04-13:
@xieqinan Hi, so this command "esp_zb_ep_list_add_gateway_ep()" can register all the client cluster and make their endpoint. Thanks waiting for your reply Sir, Thank you!

mmackh, 2025-04-14:
@Usama050 could you share some sample code please?

Usama050, 2025-04-14:
@mmackh I used this [example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_thermostat) in this example I changed the attribute according to my door sensor, cluster and endpoint function 

[image].

Usama050, 2025-04-14:
@xieqinan Hi, so this command "esp_zb_ep_list_add_gateway_ep()" can register all the client cluster from esp-sdk, but for reading the values from I have to add Measurement cluster attributes for every sensor I guess? Need your help. Thank you!

xieqinan, 2025-04-15:
> but for reading the values from sensors I have to add Measurement cluster attributes for every sensor I guess?

No, it does not need.

Usama050, 2025-04-16:
@xieqinan Hello! Sir I call the "esp_zb_ep_list_add_gateway_ep()" and code is working.

[image]

[image]
now I am waiting for sensor, I ordered them, but my question is that I do not need to add anything in esp_app_zb_attribute_handler to handle different cluster types.

[image]

xieqinan, 2025-04-17:
> now I am waiting for sensor, I ordered them, but my question is that I do not need to add anything in esp_app_zb_attribute_handler to handle different cluster types.

My mistake—I misunderstood your point. The `esp_zb_ep_list_add_gateway_ep()` indeed only contributes to creating the Zigbee data model. So you’re right: those handlers still need to be registered manually by the user.

Usama050, 2025-04-18:
@xieqinan Hello! Sir, waiting for you reply. Do I have to make cluster and end point for each sensor or this command "esp_zb_ep_list_add_gateway_ep()" can do everything? Thank you

Usama050, 2025-04-20:
@xieqinan Hello! Sir, waiting for reply!!!!

xieqinan, 2025-04-21:
> . Do I have to make cluster and end point for each sensor or this command "esp_zb_ep_list_add_gateway_ep()" can do everything? 

Sorry for the delay. I believe I already addressed this question in my previous reply. The `esp_zb_ep_list_add_gateway_ep()` function registers all cluster clients supported by the SDK on a single endpoint, so you don’t need to manually add the cluster data models for new sensors to control them. **However, the corresponding handlers are still required for different clusters.**

If the explanation above doesn’t fully resolve your issue, please provide more details about your requirements. Also, could you please give it a try and let us know the result? Let's discuss this issue based on the actual result.

https://github.com/espressif/esp-zigbee-sdk/issues/604#issuecomment-2796614081

https://github.com/espressif/esp-zigbee-sdk/issues/604#issuecomment-2811556571

Usama050, 2025-04-21:
@xieqinan Hello! Sir, I gave a try the first off all I made a struct which have information about sensor, then make global instances for all sensors, then do some changing in app button handler function and then do some modification in function bind_cb and user_find_cb here is the code
 /* Structure to store remote sensor device parameters */
 typedef struct temp_sensor_device_params_s {
     esp_zb_ieee_addr_t ieee_addr;      // 64-bit IEEE address
     uint8_t  endpoint;                 // Endpoint number
     uint16_t short_addr;               // 16-bit network address
     uint16_t cluster_id;               // Cluster ID (e.g., TEMP_MEASUREMENT, IAS_ZONE)
 } temp_sensor_device_params_t;
 
 // Global instance storing bound sensor information 
 static temp_sensor_device_params_t temp_sensor;

 /* Generic structure to store ANY Zigbee sensor device */
typedef struct zigbee_sensor_params_s {
    esp_zb_ieee_addr_t ieee_addr;  // 64-bit IEEE address
    uint8_t  endpoint;             // Endpoint number
    uint16_t short_addr;           // 16-bit network address
    uint16_t cluster_id;           // Cluster ID (e.g., TEMP_MEASUREMENT, IAS_ZONE)
    const char *sensor_type;       // Human-readable type ("Temperature", "Door", etc.)
} zigbee_sensor_params_t;

// Global instances for different sensors
static zigbee_sensor_params_t *current_binding_sensor = NULL;
static zigbee_sensor_params_t tempe_sensor = {
    .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT,
    .sensor_type = "Temperature"
};

static zigbee_sensor_params_t door_sensor = {
    .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE,
    .sensor_type = "Door Contact"
};

static zigbee_sensor_params_t smoke_sensor = {
    .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE,
    .sensor_type = "Smoke Detector"
};

// Global instances for different sensors
static zigbee_sensor_params_t gas_sensor = {
    .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_CARBON_DIOXIDE_MEASUREMENT,
    .sensor_type = "Gas Sensor"
};

static zigbee_sensor_params_t humidity_sensor = {
    .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT,
    .sensor_type = "Humidity Sesnor"
};

static zigbee_sensor_params_t motion_sensor = {
    .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_OCCUPANCY_SENSING,
    .sensor_type = "Motion Detector"
};

// Global instances for different sensors
static zigbee_sensor_params_t vibration_sensor = {
    .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE,
    .sensor_type = "Vibration sensor"
};

static zigbee_sensor_params_t light_sensor = {
    .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ILLUMINANCE_MEASUREMENT,
    .sensor_type = "Light Sensor"
};

static zigbee_sensor_params_t air_quality_sensor = {
    .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_PM2_5_MEASUREMENT,
    .sensor_type = "Air_quality_measure"
};

[... 291 lines cut ...]
 /* Find matching temperature sensors in the network in our case the door sensor*/
 static void find_temperature_sensor(esp_zb_zdo_match_desc_req_param_t *param, esp_zb_zdo_match_desc_callback_t user_cb, void *user_ctx)
 {
     uint16_t cluster_list[] = {ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE,           // Door sensor
                                ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT,    // Temperature
                                ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT, // Humidity
                                ESP_ZB_ZCL_CLUSTER_ID_ON_OFF,              // Button
                                ESP_ZB_ZCL_CLUSTER_ID_PM2_5_MEASUREMENT,   // PM2.5
                                ESP_ZB_ZCL_CLUSTER_ID_CARBON_DIOXIDE_MEASUREMENT, // CO2
                                ESP_ZB_ZCL_CLUSTER_ID_OCCUPANCY_SENSING,   // Occupancy
                                ESP_ZB_ZCL_CLUSTER_ID_ILLUMINANCE_MEASUREMENT, // Light
                                ESP_ZB_ZCL_CLUSTER_ID_FLOW_MEASUREMENT,        // flow measurement
                                ESP_ZB_ZCL_CLUSTER_ID_LEVEL_CONTROL,            // dimmer or smart switch
                                ESP_ZB_ZCL_CLUSTER_ID_METERING                  // measure the engergy, water and gas
        };
     param->profile_id = ESP_ZB_AF_HA_PROFILE_ID;
     param->num_in_clusters = ARRAY_LENTH(cluster_list);
     param->num_out_clusters = 0;
     param->cluster_list = cluster_list;
     esp_zb_zdo_match_cluster(param, user_cb, (void *)&temp_sensor);
 }
 
there is no error while running the code but when I flashed it and try to connect the sensor it gave me that error

[image]

Usama050, 2025-04-22:
@xieqinan is this the right way to add multiple cluster in **"bind_req->cluster_id"**
` if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
         ESP_LOGI(TAG, "Found the sensor");
         /* Store the information of the remote device */
         zigbee_sensor_params_t *sensor = (zigbee_sensor_params_t *)user_ctx;
         //temp_sensor_device_params_t *sensor = (temp_sensor_device_params_t *)user_ctx;
         sensor->endpoint = peer_endpoint;
         sensor->short_addr = peer_addr;
         esp_zb_ieee_address_by_short(sensor->short_addr, sensor->ieee_addr);
 
         /* 1. Send binding request to the sensor */
         esp_zb_zdo_bind_req_param_t *bind_req = (esp_zb_zdo_bind_req_param_t *)calloc(sizeof(esp_zb_zdo_bind_req_param_t), 1);
         bind_req->req_dst_addr = peer_addr;
 
         /* populate the src information of the binding */
         memcpy(bind_req->src_address, sensor->ieee_addr, sizeof(esp_zb_ieee_addr_t));
         bind_req->src_endp = peer_endpoint;
         bind_req->cluster_id = sensor->cluster_id;
 
         /* populate the dst information of the binding */
         bind_req->dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
         esp_zb_get_long_address(bind_req->dst_address_u.addr_long);
         bind_req->dst_endp = HA_THERMOSTAT_ENDPOINT;`

Usama050, 2025-05-12:
@xieqinan Hello! How are you doing? I successfully connected my sensor with ESP32H2 and I got the output of them as well. 

[image]

[image]

I only get output when I click the boot button on controller I guess it is because in [thermostat example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_thermostat/main/esp_zb_thermostat.c#L26) we have app_button_handler function I guess I have to remove this one and the command you told me which can add all client cluster did not work for me If you can suggest me example where you guys used "esp_zb_ep_list_add_gateway_ep()". It will be very helpful for me. Thank you! Have a good day.

Usama050, 2025-05-13:
@xieqinan Hello! How are you doing? I have a smart button "Tuya ERS-10TZBVK-AA" which have two mode of operations **COMMAND MODE** and **EVENT MODE**. I am wondering which cluster can I use for this smart button "Tuya ERS-10TZBVK-AA" .

[image]

[image]

Usama050, 2025-05-15:
@xieqinan  Hello! waiting for reply Sir!!!!!

Usama050, 2025-05-19:
@xieqinan Hello! How are you? How can I get output from sensor without pressing boot button on my ESP32H2 controller? waiting for your reply. Thank you in advance!

Usama050, 2025-05-20:
@xieqinan Hello! ????

xieqinan, 2025-05-21:
>  I am wondering which cluster can I use for this smart button "Tuya ERS-10TZBVK-AA" .

It looks like the clusters involved are `level_control` and `color_control`, but I'm not entirely sure. Please contact Tuya for more detailed support on this.

By the way, I often have trouble understanding your exact point. Could you please provide more detailed information about your application and try to maintain consistent context within the discussion?

xieqinan, 2025-06-13:
@Usama050 ,

Do you have any updates? If the issue has been resolved, please consider closing it.