espressif/esp-zigbee-sdk #605: Thermostat, crash when try to report attributes.

dzungpv, 2025-03-23:
I am developing a thermostat router device, the basic is working but now I try to update state of below attribute to coordinator:
- ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPIED_HEATING_SETPOINT_ID
- ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPIED_COOLING_SETPOINT_ID
- ESP_ZB_ZCL_ATTR_THERMOSTAT_SYSTEM_MODE_ID
- ESP_ZB_ZCL_ATTR_THERMOSTAT_THERMOSTAT_RUNNING_STATE_ID

But the application crash with logs:
```
Zigbee stack assertion failed zcl/zcl_general_commands.c:612

abort() was called at PC 0x42019d51 on core 0
--- 0x42019d51: zb_assert at ??:?

--- Stack dump detected
Core  0 register dump:
MEPC    : 0x408008d0  RA      : 0x40807a4a  SP      : 0x40821290  GP      : 0x4080e534  
--- 0x408008d0: panic_abort at /Users/vietdzung/esp/esp-idf-v5.3.x/components/esp_system/panic.c:463
0x40807a4a: __ubsan_include at /Users/vietdzung/esp/esp-idf-v5.3.x/components/esp_system/ubsan.c:311

TP      : 0x40821410  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x408212f4  A0      : 0x408212bc  A1      : 0x408212f2  
A2      : 0x00000000  A3      : 0x408212e9  A4      : 0x00000001  A5      : 0x40815000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000002  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x40820720  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
--- 0x40800001: _vector_table at /Users/vietdzung/esp/esp-idf-v5.3.x/components/riscv/vectors_intc.S:54

MHARTID : 0x00000000  

--- Warning: Failed to run gdb_panic_server.py script: Command '['riscv32-esp-elf-gdb', '--batch', '-n', '/Users/vietdzung/dev/projects/esp32/thermostat_zigbee/build/color_light_bulb.elf', '-ex', 'target remote | "/Users/vietdzung/.espressif/python_env/idf5.3_py3.12_env/bin/python" -m esp_idf_panic_decoder --target esp32c6 "/var/folders/xk/7_gfb97d3_98j05bt3k0wdlm0000gn/T/tmpuwpw4sy7"', '-ex', 'bt']' died with <Signals.SIGABRT: 6>.
b"dyld[7533]: symbol '__ZTINSt3__13pmr15memory_resourceE' missing from root that overrides /usr/lib/libc++.1.dylib. Use of that symbol in /System/Library/PrivateFrameworks/caulk.framework/Versions/A/caulk is being set to 0xBAD4007.\ndyld[7533]: missing symbol called\n\n\nFatal signal: Abort trap: 6\n----- Backtrace -----\n---------------------\nA fatal error internal to GDB has been detected, further\ndebugging is not possible.  GDB will now terminate.\n\nThis is a bug, please report it.  For instructions, see:\n<https://www.gnu.org/software/gdb/bugs/>.\n\n"

Stack memory:
40821290: 0x00000000 0x00000000 0x408212f0 0x4080caba 0x00000000 0x40820720 0x00000000 0x4080f258
408212b0: 0x408212f4 0x4080f274 0x408212f0 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
408212d0: 0x43502074 0x34783020 0x39313032 0x20313564 0x63206e6f 0x2065726f 0x00000030 0x42010000
408212f0: 0x40820030 0x31303234 0x31356439 0x00000000 0x0000002f 0x0000ffff 0x4207dfc9 0x42019d54
40821310: 0x408213be 0x408213be 0x00000c1c 0x00000264 0x408213be 0x00000012 0x00000012 0x42036d6c
40821330: 0x00000001 0x408213be 0x00000002 0x00000201 0x00000000 0x4080e1a0 0x00000000 0x4200f7a4
40821350: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0x00000000
40821370: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40821390: 0x42085000 0x00000834 0x00000000 0x4200c8b4 0x00000000 0x00000000 0x00000000 0x00010400
408213b0: 0x00000000 0x00000000 0x00000000 0x08340000 0x00000000 0x00000000 0x00000000 0x4200caa4
408213d0: 0x00000000 0x00000000 0x00000000 0x408086a2 0x00000000 0x00000000 0x00000000 0x00000000
408213f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40821410: 0xa5a5a5a5 0xa5a5a5a5 0x0005b1f1 0x40814ec4 0x40814ec4 0xaded62fe 0xd989c4f8 0xe2c2ec2a
40821430: 0xb9056a63 0x5e9a2f30 0xdd1e5d5d 0xfb2b588f 0xf35deedf 0xa6dfc6ba 0x85629568 0x9725dc7a
40821450: 0xb9ac94ad 0x23157d98 0xdde6a177 0x74b8dafb 0x5f351aaa 0x9f55152e 0x842b1f4a 0xfa9fcac9
40821470: 0x3c10fba7 0xa5718f00 0x9573cd5b 0xb8c6ecfd 0x431fd97d 0xd3bb14b7 0x4b6e9eb1 0x7939f973
40821490: 0x8d1f4636 0x04897d35 0x88d6eff9 0x4e3d5425 0x5367bf77 0x7e67e90a 0xda29d86e 0x8e4af09e
408214b0: 0x69d93184 0x80009a1b 0x7ff62742 0x51385362 0x29128475 0xd96f4da1 0x650401bb 0x5ca3e298
408214d0: 0x22209a47 0x0905a58a 0xfb50ca66 0xefef67b6 0x6b56b7f1 0xd77f1399 0x00ea4508 0xd2840f01
408214f0: 0x0852de06 0xc9065f10 0xc75bfc2b 0x74e12550 0x6a962fa7 0x1edf1fe8 0x17e4625d 0x5a30078a
40821510: 0x4d924336 0x9619b98e 0xd66df27c 0xdbfbc784 0x36fd57ad 0xb751b3fd 0xee1185e4 0xd5162992
40821530: 0x915e1ec7 0x99a6a85e 0x3ffbaf36 0x5de239ae 0x5c24adfc 0x897f05ae 0xa5aa6938 0xde4d5072
40821550: 0x8da9425c 0x833daded 0xdde17064 0x71adbd1b 0xb5d26090 0x89f55ee8 0x485f280b 0x17b76754
40821570: 0x93149120 0x731bd1ad 0x8e361764 0xb2f746f6 0x16b5bc99 0xfc223ac0 0xe5505607 0xbbe25d74
40821590: 0x17d48998 0x2e6cca41 0xcfaf9774 0xb64fed1f 0x05f20fe0 0xfefbecdd 0x50220fe4 0xc4dd6f28
408215b0: 0xff8242cc 0x28d5877c 0xafe82fcb 0xc9f503fb 0xfcb59623 0x7174d1ad 0x51312292 0x1e3050c7
408215d0: 0x06dc03f1 0x3015a484 0x9fe58fe5 0x680c3994 0x575166e7 0xbb29c272 0x6941eb01 0x896f804f
408215f0: 0xc359bc78 0xadcd61c2 0xb7a67da1 0xd1734edc 0x2bd3398e 0xb9ec81e3 0x14a0b410 0x24858f37
40821610: 0x290153dc 0xe7cd1644 0xf52c1ff8 0xece3a7f0 0x85b52191 0x1bd0d6c1 0x89c52252 0xe3dd4105
40821630: 0x81590218 0xe12b0954 0xf6ce8872 0xb4e352b5 0xf16cc7bb 0x3dbec15d 0x14a1782a 0x93600ad9
40821650: 0x03c921a7 0x84646515 0xecad7096 0xdd725927 0x0ab56fae 0xb76b06eb 0xb8cb5ad1 0x66337be1
40821670: 0xe6c78c10 0xf14cc608 0x515bbc4d 0x76bc03e5 0xed9b4de9 0x9d069f96 0x5b5ae745 0xa881d3ea
```

The code cause the crash is:

```
static esp_zb_zcl_report_attr_cmd_t heating_setpoint_cmd_req = {
    .zcl_basic_cmd = {
        .dst_addr_u = {
            .addr_short = 0x0000,
        },
        .dst_endpoint = HA_THERMOSTAT_ENDPOINT,
        .src_endpoint = HA_THERMOSTAT_ENDPOINT,
    },
    .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
    .clusterID = ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT,
    .attributeID = ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPIED_HEATING_SETPOINT_ID
};

esp_zb_zcl_status_t status = ESP_ZB_ZCL_STATUS_SUCCESS;

    /* Update Occupied Heating Setpoint */
    status = esp_zb_zcl_set_attribute_val(HA_THERMOSTAT_ENDPOINT,
                                          ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
                                          ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPIED_HEATING_SETPOINT_ID,
                                          &heating_setpoint_val, false);
    if (status != ESP_ZB_ZCL_STATUS_SUCCESS) {
        ESP_LOGE(TAG, "Set attribute heating setpoint failed: 0x%x", status);
    }

    esp_zb_zcl_attr_t *value_r = esp_zb_zcl_get_attribute(HA_THERMOSTAT_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPIED_HEATING_SETPOINT_ID);
    if (value_r == NULL)
    {
        ESP_LOGE(TAG, "Attribute  heating setpoint not found: 0x%x, attribute: 0x%x", ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT, ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPIED_HEATING_SETPOINT_ID);
        return;
    }

    status = esp_zb_zcl_report_attr_cmd_req(&heating_setpoint_cmd_req);
    if (status != ESP_ZB_ZCL_STATUS_SUCCESS) {
        ESP_LOGE(TAG, "Sending heating setpoint report command failed! (0x%x)", status);
    }
```
Device has been joined a coordinator running zigbee2mqtt on zstack 3.0. And it can receive ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID esp_zb_zcl_set_attr_value_message_t

I remove the code
`status = esp_zb_zcl_report_attr_cmd_req(&heating_setpoint_cmd_req);` it will not crash anymore but the thermostate not update back to the coordinator. I have read a similar issue here https://github.com/espressif/esp-zigbee-sdk/issues/203 but my MIN and MAX HEATING_SETPOINT has been set to the thermostat cluster.

I also have problem update the ESP_ZB_ZCL_ATTR_THERMOSTAT_THERMOSTAT_RUNNING_STATE_ID, there is no docs mention this value data, only this https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/zcl/esp_zigbee_zcl_thermostat.html#_CPPv4N28esp_zb_zcl_thermostat_attr_t54ESP_ZB_ZCL_ATTR_THERMOSTAT_THERMOSTAT_RUNNING_STATE_IDE

xieqinan, 2025-03-24:
> I remove the code
status = esp_zb_zcl_report_attr_cmd_req(&heating_setpoint_cmd_req); it will not crash anymore but the thermostate not update back to the coordinator. I have read a similar issue here https://github.com/espressif/esp-zigbee-sdk/issues/203 but my MIN and MAX HEATING_SETPOINT has been set to the thermostat cluster.

Based on ZCL spec, The `ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPIED_HEATING_SETPOINT_ID` attribute is not reportable.

> I also have problem update the ESP_ZB_ZCL_ATTR_THERMOSTAT_THERMOSTAT_RUNNING_STATE_ID, there is no docs mention this value data

The ZCL of esp-zigbee-sdk is based on the [Zigbee Cluster Library](https://zigbeealliance.org/wp-content/uploads/2021/10/07-5123-08-Zigbee-Cluster-Library.pdf). If you find any missing information or errors in the documentation, please refer to it. If possible, you can also submit a [PR](https://github.com/espressif/esp-zigbee-sdk/pulls) for this SDK.

dzungpv, 2025-03-24:
> Based on ZCL spec, The `ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPIED_HEATING_SETPOINT_ID` attribute is not reportable.
>
>
> The ZCL of esp-zigbee-sdk is based on the [Zigbee Cluster [...]

I know that standard is optional, you can see the picture, but many commercial product can report those parameter, relate problem discuss here about ZBOSS https://devzone.nordicsemi.com/f/nordic-q-a/115861/zboss-reporting-of-attributes-that-are-not-required-to-be-reportable

[image]

xieqinan, 2025-03-25:
As mentioned, *"Other attributes MAY optionally be reported."* The `esp-zigbee-sdk` does not restrict users from setting attribute access for a cluster. A standard thermostat device that follows the ZCL specification and Home Automation profile can be created using the `esp_zb_thermostat_ep_create()` API. Additionally, you can use `esp_zb_cluster_add_attr()` or `esp_zb_cluster_add_manufacturer_attr()` to add custom attributes to the related cluster.

Does this answer your question?

dzungpv, 2025-03-25:
> `esp_zb_cluster_add_attr`

OK, I will try if it can work and let you know later.

xieqinan, 2025-04-23:
@dzungpv，

> OK, I will try if it can work and let you know later.

Do you have any updates?

xieqinan, 2025-05-16:
Hi @dzungpv 

Adding the `heating_setpoint_cmd_req.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI` can help fix this crashing when calling `esp_zb_zcl_report_attr_cmd_req()` in your application.

If you expect to add the `report` access to the attribute of cluster, you can follow this comment: https://github.com/espressif/esp-zigbee-sdk/issues/605#issuecomment-2749890439.

So I believe the issue has been resolved, please feel free to reopen if it is still presists.

closed 2025-05-16 by xieqinan