espressif/esp-zigbee-sdk #606: Low power consumption when using zigbee features

Mushan-bin, 2025-03-24:
I want to use zigbee terminal power consumption to reach the microamperage level, when zigbee terminal and zigbee coordinator network connection is successful, whether the power consumption can reach single digit microamperage, only when zigbee terminal sends data to zigbee coordinator, mA level will appear, want this effect, How to configure the menuconfig in ESP-IDF, or there are other ways to achieve such a condition.

xieqinan, 2025-03-24:
Please refer to the [light sleep README](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/light_sleep#example-output) and [deep sleep README](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/deep_sleep) first. You can select the appropriate mode based on your application.

Mushan-bin, 2025-03-25:
When I use the ESP32-H2 device, 15mA instant current will be generated during power-on, and then the power consumption will reach 10mA before entering the zigbee-light_sleep state. How can I reduce this 15mA instant current and the power consumption can reach microamps before entering zigbee-light_sleep state? How to power the device to low power. @xieqinan

xieqinan, 2025-03-26:
The device will transmit frames to search for an available network upon powering on, resulting in high power consumption. You may use `esp_zb_set_tx_power()` to reduce the tx power to reduce consumption, but it cannot reach microamp levels due to hardware limitations. Please refer to the `6.4 Current Consumption Characteristics` section of [esp32h2 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-h2-mini-1_mini-1u_datasheet_en.pdf).

Mushan-bin, 2025-03-27:
@xieqinan Can the instantaneous current of the ESP32-H2 device be reduced when it is powered on and how?

xieqinan, 2025-03-27:
> Can the instantaneous current of the ESP32-H2 device be reduced when it is powered on and how?

I don't think it can, but you can get a more authoritative answer regarding low-power issues, as this is a system-related matter, by submitting an issue in [esp-idf](https://github.com/espressif/esp-idf/issues).

xieqinan, 2025-06-03:
Hi @Mushan-bin ,

We’ve added the [light-sleep FAQs](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/faq.html#a1-4-1-how-to-configure-the-zigbee-end-device-to-enter-light-sleep-mode) to explain the low-power feature.

That said, low-power behavior is a system-level matter, and Zigbee itself does not affect the hardware’s power management.

Let’s close this issue for now. For further low-power-related questions, please consider submitting them to the [esp-idf repository](https://github.com/espressif/esp-idf/issues) for a more authoritative response.

closed 2025-06-03 by xieqinan