espressif/esp-zigbee-sdk #607: assert failed: isr_handle_tx_ack_phase_rx_abort esp_ieee802154_dev.c:552

cheldonec, 2025-03-24:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4.0

### esp-zigbee-lib version.

1.6.4

### esp-zboss-lib version.

1.6.4

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

At one point, the board started restarting randomly with this error (about once every 20 to 40 minutes). The frequency is not the same. The zigbee channel was used on 13. I assumed it was interference and changed the channel to 26. The failure happened again, but twice a night, much less often. Is there any way to ignore this malfunction and double-check, for my project, restarting the board is unacceptable.

WI-FI is disabled, coex is disabled.

I saw a similar problem (https://github.com/espressif/esp-zigbee-sdk/issues/326 ?ysclid=m8mn3ji6ku291871759) But it doesn't seem to have been solved.

### What is the actual behavior?

Periodic restart with the assert failed error: isr_handle_tx_ack_phase_rx_abort esp_ieee802154_dev.c:552 (s_ieee802154_state == IEEE802154_STATE_TX_ACK || s_ieee802154_state == IEEE802154_STATE_TX_ENH_ACK)

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### More Information.

W (862) ZB_MAIN_MODULE: 3.Network(0xbd47) closed, devices joining not allowed.
I (872) ZB_MAIN_MODULE: Device started up in non factory-reset mode
I (872) ZB_MAIN_MODULE: Device rebooted

assert failed: isr_handle_tx_ack_phase_rx_abort esp_ieee802154_dev.c:552 (s_ieee802154_state == IEEE802154_STATE_TX_ACK || s_ieee802154_state == IEEE802154_STATE_TX_ENH_ACK)
Core  0 register dump:
MEPC    : 0x40800962  RA      : 0x4080978a  SP      : 0x40813080  GP      : 0x40810b14
--- 0x40800962: panic_abort at C:/Espressif/esp/v5.4/esp-idf/components/esp_system/panic.c:468
0x4080978a: __ubsan_include at C:/Espressif/esp/v5.4/esp-idf/components/esp_system/ubsan.c:311

TP      : 0x40823410  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x0000001a  S1      : 0x00000001  A0      : 0x408130bc  A1      : 0x4081158d
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x40820000
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x4081323d
S4      : 0x4081158c  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000
--- 0x40800001: _vector_table at C:/Espressif/esp/v5.4/esp-idf/components/riscv/vectors_intc.S:54

MHARTID : 0x00000000

Stack memory:
40813080: 0x00000000 0x00000000 0x420b1550 0x4080f1da 0x4081176c 0x420b1550 0x4081177c 0x420a949f
--- 0x4080f1da: newlib_include_assert_impl at C:/Espressif/esp/v5.4/esp-idf/components/newlib/assert.c:92

408130a0: 0x40811780 0x408130b4 0x40811784 0x420a95c8 0x4081158c 0x00323535 0x00000000 0x65737361
[... 83 lines cut ...]
I (383) esp_image: segment 4: paddr=000c615c vaddr=40810310 size=024c8h (  9416) load
I (397) boot: Loaded app from partition at offset 0x10000
I (397) boot: Disabling RNG early entropy source...
I (408) cpu_start: Unicore app
I (417) cpu_start: Pro cpu start user code
I (417) cpu_start: cpu freq: 160000000 Hz
I (417) app_init: Application information:
I (417) app_init: Project name:     smarthomecoordinator
I (422) app_init: App version:      1
I (425) app_init: Compile time:     Mar 23 2025 10:18:58
I (430) app_init: ELF file SHA256:  e071c496e...
I (435) app_init: ESP-IDF:          v5.4-dirty
I (439) efuse_init: Min chip rev:     v0.0
I (443) efuse_init: Max chip rev:     v0.99 
I (447) efuse_init: Chip rev:         v0.0
I (451) heap_init: Initializing. RAM available for dynamic allocation:
I (458) heap_init: At 4081FC90 len 0005C980 (370 KiB): RAM
I (462) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (467) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (473) spi_flash: detected chip: generic
I (476) spi_flash: flash io: dio
I (480) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (485) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (492) main_task: Started on CPU0
I (492) main_task: Calling app_main()

zwx1995esp, 2025-03-24:
Hi @cheldonec  Thanks for providing such detailed information! Could you enable some debugging configurations and share more details when an assertion occurs? Let's enable the following configurations:
```
CONFIG_IEEE802154_DEBUG=y
CONFIG_IEEE802154_ASSERT=y
CONFIG_IEEE802154_RECORD=y
CONFIG_IEEE802154_RECORD_EVENT=y
CONFIG_IEEE802154_RECORD_EVENT_SIZE=30
CONFIG_IEEE802154_RECORD_STATE=y
CONFIG_IEEE802154_RECORD_STATE_SIZE=10
CONFIG_IEEE802154_RECORD_CMD=y
CONFIG_IEEE802154_RECORD_CMD_SIZE=10
CONFIG_IEEE802154_RECORD_ABORT=y
CONFIG_IEEE802154_RECORD_ABORT_SIZE=10

CONFIG_ESP_INT_WDT=n

CONFIG_FREERTOS_ISR_STACKSIZE=8192
```

cheldonec, 2025-03-25:
Hi @zwx1995esp It looks like it was a one-time glitch for 2 days. I'm going to ask you not to close the subject. I'm trying to repeat mine, it will be important for everyone, because I thought that my code needed to be redone, although I took this very seriously.

zwx1995esp, 2025-03-26:
> Hi [@zwx1995esp](https://github.com/zwx1995esp) It looks like it was a one-time glitch for 2 days. I'm going to ask you not to close the subject. I'm trying to repeat mine, it will be important for everyone, because I thought that my code needed to be redone, although I took this very seriously.

Ok, thanks for your such patient debugging, wait for your good news.

cheldonec, 2025-03-28:
Hi @zwx1995esp 
There was an opportunity to test in another location geographically and everything was fine. The error was repeated in the same place where it first appeared. It's definitely some kind of hindrance that can't be found. However, maybe there is some solution to ignore the error?), although I can't imagine how)
This Test with WI-FI connected and coex enable. (zb channel 11)

I (17737) ZB_MAIN_MODULE: Production configuration is ready
W (17737) ZB_MAIN_MODULE: Production configuration is not present
I (17737) ZB_MAIN_MODULE: Initialize Zigbee stack
I (17737) ZB_MAIN_MODULE: Device started up in factory-reset mode
I (17747) ZB_MAIN_MODULE: Start network formation
I (17897) ZB_MAIN_MODULE: 3.ESP_ZB_NWK_SIGNAL_PERMIT_JOIN_STATUS:
W (17897) ZB_MAIN_MODULE: 3.Network(0x8e6c) closed, devices joining not allowed.
I (17897) ZB_MAIN_MODULE: Formed network successfully (Extended PAN ID: 54:32:04:ff:fe:08:6d:24, PAN ID: 0x8e6c, Channel:11, Short Address: 0x0000)
I (18507) ZB_MAIN_MODULE: 3.ESP_ZB_NWK_SIGNAL_PERMIT_JOIN_STATUS:
I (18507) ZB_MAIN_MODULE: 3.Network(0x8e6c) is open for 180 seconds
I (18507) ZB_MAIN_MODULE: Network steering started
W (35367) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x100f4d3, TALO:0xbee360d4, (ssn:5, win:64, cur_ssn:5), CONF:0xc0000005
W (44687) ieee802154: Print the record event, current event index: 0
W (44687) ieee802154: index  0: event: 0x   2,         RX_DONE, state:        RX, timestamp: 43067951 
W (44687) ieee802154: index  1: event: 0x   4,     ACK_TX_DONE, state:    TX_ACK, timestamp: 43068512 
W (44687) ieee802154: index  2: event: 0x 800,     TX_SFD_DONE, state:    TX_CCA, timestamp: 43070354 
W (44687) ieee802154: index  3: event: 0x   1,         TX_DONE, state:    TX_CCA, timestamp: 43071935 
W (44687) ieee802154: index  4: event: 0x1000,     RX_SFD_DONE, state:    RX_ACK, timestamp: 43072246 
W (44687) ieee802154: index  5: event: 0x   8,     ACK_RX_DONE, state:    RX_ACK, timestamp: 43072436 
W (44687) ieee802154: index  6: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43231804 
W (44687) ieee802154: index  7: event: 0x   2,         RX_DONE, state:        RX, timestamp: 43232220 
W (44687) ieee802154: index  8: event: 0x   4,     ACK_TX_DONE, state:    TX_ACK, timestamp: 43232781 
W (44687) ieee802154: index  9: event: 0x 800,     TX_SFD_DONE, state:    TX_CCA, timestamp: 43234636 
W (44687) ieee802154: index 10: event: 0x   1,         TX_DONE, state:    TX_CCA, timestamp: 43236217 
W (44687) ieee802154: index 11: event: 0x1000,     RX_SFD_DONE, state:    RX_ACK, timestamp: 43236539 
W (44687) ieee802154: index 12: event: 0x   8,     ACK_RX_DONE, state:    RX_ACK, timestamp: 43236730 
W (44687) ieee802154: index 13: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43396884 
W (44687) ieee802154: index 14: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43398733 
W (44687) ieee802154: index 15: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43401243 
W (44687) ieee802154: index 16: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43403093 
W (44687) ieee802154: index 17: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43405655 
W (44687) ieee802154: index 18: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43407505 
W (44687) ieee802154: index 19: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43573593 
W (44687) ieee802154: index 20: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43574319 
W (44687) ieee802154: index 21: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43737887 
W (44687) ieee802154: index 22: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43738615 
W (44687) ieee802154: index 23: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43739713 
W (44687) ieee802154: index 24: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43741559 
W (44687) ieee802154: index 25: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43902254 
W (44687) ieee802154: index 26: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 43902982 
W (44687) ieee802154: index 27: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 44242204 
W (44687) ieee802154: index 28: event: 0x   2,         RX_DONE, state:        RX, timestamp: 44242937 
W (44687) ieee802154: index 29: event: 0x  10,        RX_ABORT, state:        RX, timestamp: 44244664 rx abort reason:   10,       TX_ACK_TIMEOUT
W (44687) ieee802154: Print the record event done.
W (44687) ieee802154: Print the record state, current state index: 5
W (44687) ieee802154: index  0: line:  366, state:        RX, timestamp: 43232824
W (44687) ieee802154: index  1: line:  867, state:    TX_CCA, timestamp: 43234170
W (44687) ieee802154: index  2: line:  422, state:    RX_ACK, timestamp: 43236230
W (44687) ieee802154: index  3: line:  366, state:        RX, timestamp: 43236778
W (44687) ieee802154: index  4: line:  366, state:        RX, timestamp: 44243019
W (44687) ieee802154: index  5: line:  366, state:        RX, timestamp: 43068559
W (44687) ieee802154: index  6: line:  867, state:    TX_CCA, timestamp: 43069888
W (44687) ieee802154: index  7: line:  422, state:    RX_ACK, timestamp: 43071948
W (44687) ieee802154: index  8: line:  366, state:        RX, timestamp: 43072482
W (44687) ieee802154: index  9: line:  445, state:    TX_ACK, timestamp: 43232240
W (44687) ieee802154: Print the record state done.
[... cut ...]
40824550: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40824570: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 244f1e9c9

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
--- 0x4001975a: software_reset_cpu in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe84
load:0x4086e610,len:0x303c
entry 0x4086c11a
I (26) boot: ESP-IDF v5.4-dirty 2nd stage bootloader

zwx1995esp, 2025-03-31:
Hi @cheldonec Thanks for your logs. 

> The error was repeated in the same place where it first appeared.

Question: are there lots of 15.4 frames in the issue-reproduced place? 

I added more logs to show some more information for the further debugging. [debug_154_crash.zip](https://github.com/user-attachments/files/19532806/debug_154_crash.zip) Could you please apply this patch to your IDF branch and set these two configurations, then try to reproduce this issue in the issue-reproduced place again? 
You can apply this patch via these commands:
```
$ cd <your-local-IDF-path>
git apply <path-of-the-patch>/debug_154_crash.patch
```

And you can enable the frame recording feature by setting these configurations:
```
CONFIG_IEEE802154_RECORD_TXRX_FRAME=y
CONFIG_IEEE802154_RECORD_TXRX_FRAME_SIZE=15
```

zwx1995esp, 2025-04-18:
Hi @cheldonec Do we have any chance of reproducing the issue with the debugging info enabled? May luck be with you :)

cheldonec, 2025-04-18:
Hi @zwx1995esp I run the test on weekends, since last time it was on weekends that there was a failure. But so far without success ( Maybe the interference is gone. Let's see if it works out. If you need to close the topic, then that's probably what you should do for now, if I catch a glitch, I'll contact you.

cheldonec, 2025-04-23:
Hi @zwx1995esp Hooray!!! I caught a glitch!!!

W (31531501) ieee802154: Print the record event, current event index: 28
W (31531501) ieee802154: index  0: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31462075026 
W (31531501) ieee802154: index  1: event: 0x 800,     TX_SFD_DONE, state:    TX_CCA, timestamp: 31462779680 
W (31531501) ieee802154: index  2: event: 0x   1,         TX_DONE, state:    TX_CCA, timestamp: 31462781263 
W (31531501) ieee802154: index  3: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31464539309 
W (31531501) ieee802154: index  4: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31474007009 
W (31531501) ieee802154: index  5: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31476334069 
W (31531501) ieee802154: index  6: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31476693739 
W (31531501) ieee802154: index  7: event: 0x 800,     TX_SFD_DONE, state:    TX_CCA, timestamp: 31477839681 
W (31531501) ieee802154: index  8: event: 0x   1,         TX_DONE, state:    TX_CCA, timestamp: 31477841264 
W (31531501) ieee802154: index  9: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31479471341 
W (31531501) ieee802154: index 10: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31488991050 
W (31531501) ieee802154: index 11: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31492379505 
W (31531501) ieee802154: index 12: event: 0x 800,     TX_SFD_DONE, state:    TX_CCA, timestamp: 31492889650 
W (31531501) ieee802154: index 13: event: 0x   1,         TX_DONE, state:    TX_CCA, timestamp: 31492891233 
W (31531501) ieee802154: index 14: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31494437386 
W (31531501) ieee802154: index 15: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31504011082 
W (31531501) ieee802154: index 16: event: 0x 800,     TX_SFD_DONE, state:    TX_CCA, timestamp: 31507929667 
W (31531501) ieee802154: index 17: event: 0x   1,         TX_DONE, state:    TX_CCA, timestamp: 31507931250 
W (31531501) ieee802154: index 18: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31509481431 
W (31531501) ieee802154: index 19: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31518923073 
W (31531501) ieee802154: index 20: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31518992041 
W (31531501) ieee802154: index 21: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31522541521 
W (31531501) ieee802154: index 22: event: 0x 800,     TX_SFD_DONE, state:    TX_CCA, timestamp: 31522999663 
W (31531501) ieee802154: index 23: event: 0x   1,         TX_DONE, state:    TX_CCA, timestamp: 31523001246 
W (31531501) ieee802154: index 24: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31524492481 
W (31531501) ieee802154: index 25: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31531170402 
W (31531501) ieee802154: index 26: event: 0x   2,         RX_DONE, state:        RX, timestamp: 31531171007 
W (31531501) ieee802154: index 27: event: 0x  10,        RX_ABORT, state:        RX, timestamp: 31531172735 rx abort reason:   10,       TX_ACK_TIMEOUT
W (31531501) ieee802154: index 28: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31458925976 
W (31531501) ieee802154: index 29: event: 0x1000,     RX_SFD_DONE, state:        RX, timestamp: 31459312586 
W (31531501) ieee802154: Print the record event done.
W (31531501) ieee802154: Print the record state, current state index: 8
W (31531501) ieee802154: index  0: line:  370, state:        RX, timestamp: 31477841316
W (31531501) ieee802154: index  1: line:  877, state:    TX_CCA, timestamp: 31492889178
W (31531501) ieee802154: index  2: line:  370, state:        RX, timestamp: 31492891286
W (31531501) ieee802154: index  3: line:  877, state:    TX_CCA, timestamp: 31507929196
W (31531501) ieee802154: index  4: line:  370, state:        RX, timestamp: 31507931302
W (31531501) ieee802154: index  5: line:  877, state:    TX_CCA, timestamp: 31522999192
W (31531501) ieee802154: index  6: line:  370, state:        RX, timestamp: 31523001299
W (31531501) ieee802154: index  7: line:  370, state:        RX, timestamp: 31531171069
W (31531501) ieee802154: index  8: line:  370, state:        RX, timestamp: 31462781316
W (31531501) ieee802154: index  9: line:  877, state:    TX_CCA, timestamp: 31477839210
W (31531501) ieee802154: Print the record state done.
W (31531501) ieee802154: Print the record cmd, current cmd index: 2
W (31531501) ieee802154: index  0: line:  368, cmd:        rx, timestamp: 31523001288
W (31531501) ieee802154: index  1: line:  368, cmd:        rx, timestamp: 31531171060
W (31531501) ieee802154: index  2: line:  205, cmd:      stop, timestamp: 31492889145
W (31531501) ieee802154: index  3: line:  876, cmd:    tx cca, timestamp: 31492889167
W (31531501) ieee802154: index  4: line:  368, cmd:        rx, timestamp: 31492891275
W (31531501) ieee802154: index  5: line:  205, cmd:      stop, timestamp: 31507929161
W (31531501) ieee802154: index  6: line:  876, cmd:    tx cca, timestamp: 31507929185
W (31531501) ieee802154: index  7: line:  368, cmd:        rx, timestamp: 31507931291
W (31531501) ieee802154: index  8: line:  205, cmd:      stop, timestamp: 31522999159
W (31531501) ieee802154: index  9: line:  876, cmd:    tx cca, timestamp: 31522999181
W (31531501) ieee802154: Print the record cmd done.
[... cut ...]
408188c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
408188e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 7b9a6247b

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
--- 0x4001975a: software_reset_cpu in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe84
load:0x4086e610,len:0x3048
entry 0x4086c11a
I (22) boot: ESP-IDF v5.4.1-dirty 2nd stage bootloader

zwx1995esp, 2025-04-23:
Hi, @cheldonec  Kudos to you on such patient debugging!

Per your logs, I see there is a packet:
```
W (31531501) ieee802154: index 14: rx frame, length: 18, buffer index: 19(Valid buffer), timestamp: 31531171023, mac conf: 1208004b, packet:
W (31531501) ieee802154: frag(018:001008): 75 e7 0f 6e b9 c6 95 c9
W (31531501) ieee802154: frag(018:009016): 6f f3 76 bc 0b d3 ce fd
W (31531501) ieee802154: frag(018:017~024): ae 0a 12 94 15 3b 38 32
```
Only one question: Do you have any idea about which device will send this packet? This packet is a `Multipopurse frame` defined by IEEE802.15.4 spec. Seems like Zigbee devices do not use this type of frame.

The crash you encountered is a known issue which we also encountered and fixed these days. The 15.4 Mac driver codes have some bugs when handling the multipurpose frame. And we have already fixed it. The internal MR is under reviewing process and ready for merging. I will leave a message here if the related fix commit is avaliable on Github. And then you can have a try.

cheldonec, 2025-04-23:
Hi @zwx1995esp Unfortunately, I have no assumptions about this device. Thanks for the detailed answer. I hope to try the fixes sooner. Have a nice day!!!

xieqinan, 2025-06-13:
@cheldonec ,

Based on the test, the following comment appears to have resolved your issue:[https://github.com/espressif/esp-zigbee-sdk/issues/607#issuecomment-2921699444](https://github.com/espressif/esp-zigbee-sdk/issues/607#issuecomment-2921699444).

Please run a quick verification.
I'll go ahead and close this issue. Feel free to reopen it if you’re still experiencing confusion or need further clarification.

closed 2025-06-13 by xieqinan