espressif/esp-zigbee-sdk #608: Inquiry about checking zigbee end device availability

iamiha, 2025-03-24:
### Question

I am currently working with the ESP-Zigbee-SDK and would like to know the recommended approach for checking if a Zigbee end device is available or not from the Zigbee coordinator.

Are there specific APIs or mechanisms within the SDK that can help determine if an end device is online, offline, or unreachable? 

Any guidance or example references would be greatly appreciated.

xieqinan, 2025-04-23:
Hi,

Did the above solutions help resolve your question?   Please feel free to discuss any Zigbee-related issues here.  If the issue has been resolved, kindly consider closing it.

xieqinan, 2025-05-21:
Hi,

A [FAQ document](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/faq.html#a1-3-how-to-check-whether-a-zigbee-device-is-on-the-network) has been added to address this issue. If it does not resolve your problem, please feel free to reopen the issue.

closed 2025-05-21 by xieqinan