espressif/esp-zigbee-sdk #609: Binary input : multiple endpoints

jrossouw, 2025-03-24:
### Question

I am trying to implement a sensor with dual binary inputs. The binary input on endpoint 1 works, but the one on endpoint 2 is never reported to my coordinator. Just checking is this feature is supported.

xieqinan, 2025-04-25:
Sorry for the delay. If you intend to report an attribute of a cluster to a remote device, you need to bind this cluster to the remote device and configure reporting for the attribute. Please double-check whether the second Binary Input cluster has been bound and configured for reporting, just like the first one.

jrossouw, 2025-04-26:
Thank you for the response. I only intend to report attributes to the coordinator at this point - does that required configuring reporting and binding? The sample as it is currently works correctly against my Homey Pro.

xieqinan, 2025-04-27:
> I only intend to report attributes to the coordinator at this point - does that required configuring reporting and binding?

Yes, the configuring reporting and binding is necessary for reporting attribute.

> The sample as it is currently works correctly against my Homey Pro.

Do you mean that the attribute of the binary-input cluster on a single endpoint can be reported correctly (without configuring and binding explicitly), but it fails on multiple endpoints?  If so, I suspect that `Homey Pro` only configures and binds one endpoint.   If possible, please check and provide the sniffer packet for verification.

jrossouw, 2025-06-25:
The example I posted above works correctly with my Home Pro coordinator.