espressif/esp-zigbee-sdk #611: esp_zb_device_cb_id_handler_register is never called for smart blinds

marrog86, 2025-03-26:
Hello, 
I took the esp_zb_light example as starting point. 
I bound the device successfully to my ZB network. I use Zb2Mqtt and it recognizes successfully my device as smart blind and offers state/position/tilt buttons to me. 
Open/Close and Stop command works as expected. I get the commands. 

So I tried to use esp_zb_device_cb_id_handler_register as in the example, but it never gets called. 
It was mentioned in [this issue](https://github.com/espressif/esp-zigbee-sdk/issues/345#issuecomment-2156449315)
What is the error, that the callback function never gets executed?

This is my Code: 
`

#include "esp_zb_light.h"
#include "esp_check.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"
#include "zboss_api.h"
#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile light (End Device) source code.
#endif
#define HA_ESP_BLINDS_ENDPOINT 1
static const char *TAG = "ESP_ZB_ON_OFF_LIGHT";
/********************* Define functions **************************/
static esp_err_t deferred_driver_init(void)
{
    static bool is_inited = false;
    if (!is_inited) {
        //light_driver_init(LIGHT_DEFAULT_OFF);
        is_inited = true;
    }
    return is_inited ? ESP_OK : ESP_FAIL;
}
static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_RETURN_ON_FALSE(esp_zb_bdb_start_top_level_commissioning(mode_mask) == ESP_OK, , TAG, "Failed to start Zigbee commissioning");
}
void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        ESP_LOGI(TAG, "Initialize Zigbee stack");
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Deferred driver initialization %s", deferred_driver_init() ? "failed" : "successful");
            ESP_LOGI(TAG, "Device started up in%s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : " non");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network steering");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            } else {
                ESP_LOGI(TAG, "Device rebooted");
            }
        } else {
            ESP_LOGW(TAG, "%s failed with status: %s, retrying", esp_zb_zdo_signal_to_string(sig_type),
                     esp_err_to_name(err_status));
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb,
                                   ESP_ZB_BDB_MODE_INITIALIZATION, 1000);
        }
        break;
    case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status == ESP_OK) {
            esp_zb_ieee_addr_t extended_pan_id;
[... 133 lines cut ...]
    };
    esp_zcl_utility_add_ep_basic_manufacturer_info(esp_zb_covering_ep, HA_ESP_BLINDS_ENDPOINT, &info);
    esp_zb_device_register(esp_zb_covering_ep);
    esp_zb_device_cb_id_handler_register(device_callback); //funktioniert nicht!
    esp_zb_raw_command_handler_register(raw_command_handler);
    esp_zb_core_action_handler_register(zb_action_handler);
    
    
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    esp_zb_set_primary_network_channel_set(ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_stack_main_loop();
}
void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}

`

xieqinan, 2025-04-10:
Hi,

What do you expect to obtain from `esp_zb_device_cb_id_handler_register()`? This callback will only be triggered when an event is not included in `esp_zb_core_action_callback_id_t`, but is still triggered by the stack. In this case, `esp_zb_device_cb_id_handler_register()` provides a way for the user to handle such unknown events.

marrog86, 2025-04-17:
Hello Thanks for your response.
Intention was to implement the example as it has been written in the other issue. (See first post)

for me it is now obsolete, since I found the correct way to get the payload data, and all works as expected there. 

Is the behavior of "only triggered when an event not included" mentioned in the documentation? Perhaps pointing this out would avoid similar questions in the future. 

Thanks

closed 2025-04-17 by marrog86