espressif/esp-zigbee-sdk #613: Error  ESP_ZIGBEE_ZCL_CUSTOM_CLUSTER: Received TO_CLI custom command, but cannot find the custom client cluster on endpoint

Shatodor, 2025-03-27:
I have switch that supports generic ZigBee and Tuya. When switch changes state I getting report to standart cluster (0x0006) and custom(0xef00) with error `E () ESP_ZIGBEE_ZCL_CUSTOM_CLUSTER: Received TO_CLI custom command, but cannot find the custom client cluster (0xef00) on endpoint (1)`.
So how can I make ingore this error or process this custom command?

xieqinan, 2025-03-28:
You need to register a custom (0xEF00) client cluster to endpoint 1 to handle the reports from the custom (0xEF00) cluster.

Shatodor, 2025-03-28:
Thank you, that worked. But is there more general solution for this task? Or I have to add custom cluster for each new one I encounter?

xieqinan, 2025-03-31:
> But is there more general solution for this task? Or I have to add custom cluster for each new one I encounter?

Yes, you have to add custom cluster for each new one. Due to each "to client" or "to server" custom command requires the corresponding custom cluster to be registered on the receiving device.

closed 2025-03-31 by Shatodor