espressif/esp-zigbee-sdk #614: ESP32 C6 esp_zb_sleep() function causes 45 mA current consumption.

Teodor12, 2025-03-28:
Hello everyone!

I'm using an ESP32 C6 Devkit C for implementing a low-power device. I'm using the esp_zb_sleep_now() the following way:

```
case ESP_ZB_COMMON_SIGNAL_CAN_SLEEP: {
            if (zbCore.getConnectionStatus()) {
                ESP_LOGD(LOG_TAG, "Zigbee is sleeping...");
                ESP_ERROR_CHECK(zbCore.setSleepTreshold(2750));
                zbCore.sleep();
            }
            break;
```

The most important configuration options:
# CONFIG_PM_ENABLE is not set
# CONFIG_FREERTOS_USE_TICKLESS_IDLE is not set

I must have disabled these options, due to the boards undefined behavior, although in the official light_sleep example the were enabled. 

Still don't know why but the current consumption while sleeping is ~45mA. Did I miss something or the esp32c6 really has some issues with the esp-idf-zigbee-sdk in terms of sleeping?

Thank you in advance!

xieqinan, 2025-03-31:
Hi,

The `CONFIG_PM_ENABLE` and `CONFIG_FREERTOS_USE_TICKLESS_IDLE` options must be enabled for the ESP Zigbee sleep device; otherwise, the Zigbee task will not enter sleep mode.

> I must have disabled these options due to the board's undefined behavior.

Please compare your configuration with the official light sleep example to identify any differences that prevent `CONFIG_PM_ENABLE` and `CONFIG_FREERTOS_USE_TICKLESS_IDLE` from being enabled. If possible, please attach the `sdkconfig` file for further analysis.

Teodor12, 2025-03-31:
Hello! 

I attached the sdkconfig files.

[sdkconfig.defaults.txt](https://github.com/user-attachments/files/19531147/sdkconfig.defaults.txt)
[sdkconfig.txt](https://github.com/user-attachments/files/19531148/sdkconfig.txt)

xieqinan, 2025-04-01:
I used your `sdkconfig` in the official `light_sleep` example, and it allowed me to set the `CONFIG_PM_ENABLE` and `CONFIG_FREERTOS_USE_TICKLESS_IDLE` options in `idf.py menuconfig`. Have you enabled them manually? If the issue persists and the behavior remains undefined, please share a screenshot of this menuconfig page.

[image]

Teodor12, 2025-04-01:
I'm using C++ with the Zigbee SDK. Can it have any unexpected impact on the library's functionality or behaviour?

xieqinan, 2025-04-02:
> I'm using C++ with the Zigbee SDK. Can it have any unexpected impact on the library's functionality or behaviour?

No, I think it is the same.

Ricsi1231, 2025-04-04:
Hello xieqinan! What was your current consumption in the mentioned example and what was the method used for the measurement?

Thank you!

xieqinan, 2025-04-07:
> What was your current consumption in the mentioned example and what was the method used for the measurement?

You can refer to the [README](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/light_sleep#example-output) of light sleep.

xieqinan, 2025-04-27:
Hi, @Ricsi1231 ,

Do you have any updates? I have double-check this problem, it may be triggered by the missing RTC configuration of ESP32C6 in the light sleep example. But it has been fixed, please refer to commit: https://github.com/espressif/esp-zigbee-sdk/commit/28b5393ec4bc563388cd0ab13495544ccd320223. Please test again with this commit.

xieqinan, 2025-05-16:
Hi @Ricsi1231 ,

I believe you problem  can be fixed by comment https://github.com/espressif/esp-zigbee-sdk/issues/614#issuecomment-2832919673. If it is still presists, please feel free to reopen this issue.

closed 2025-05-16 by xieqinan