espressif/esp-zigbee-sdk #616: Define and report custom cluster attribute

florian-bailly, 2025-03-29:
### Question

Hi,

I'm trying to define and report a custom cluster attribute. I have a radar sensor that can report more than the occupancy value.

I defined a class that extends the ZigbeeOccupancySensor and added the custom cluster to the existing cluster list.

I set my custom attribute to a test value and then report it.

I tried a lot of combinations (`esp_zb_custom_cluster_add_custom_attr`, `esp_zb_cluster_add_manufacturer_attr`, `esp_zb_cluster_add_attr`) and with or without the `ESP_ZB_ZCL_ATTR_MANUF_SPEC` attribute.

I also tried to set the attribute value with `esp_zb_zcl_set_attribute_val` or `esp_zb_zcl_set_manufacturer_attribute_val`.

And finally I tried reporting with the manuf_code set to ESP_ZB_ZCL_ATTR_NON_MANUFACTURER_SPECIFIC, a custom value or not set.

There must be something wrong I'm not seeing.

Here's a sample code:
```cpp
#define ZB_ZCL_CLUSTER_ID_MY 0xff01

typedef struct esp_zb_my_cluster_cfg_s {
  uint8_t attr0;
} esp_zb_my_cluster_cfg_t;

typedef enum {
  ZB_ZCL_ATTR_MY_ATTR0  = 0,
} esp_zb_my_cluster_attr_t;

ZigbeeMySensor::ZigbeeMySensor(uint8_t endpoint) : ZigbeeOccupancySensor(endpoint) {
  esp_zb_my_cluster_cfg_t cluster_cfg = {0};
  esp_zb_attribute_list_t *esp_zb_custom_cluster_attr_list = esp_zb_zcl_attr_list_create(ZB_ZCL_CLUSTER_ID_MY);

  ESP_ERROR_CHECK(esp_zb_custom_cluster_add_custom_attr(esp_zb_custom_cluster_attr_list, ZB_ZCL_ATTR_MY_ATTR0, ESP_ZB_ZCL_ATTR_TYPE_8BIT, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &cluster_cfg.attr0));

  ESP_ERROR_CHECK(esp_zb_cluster_list_add_custom_cluster(_cluster_list, esp_zb_custom_cluster_attr_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
}

void ZigbeeMySensor::setAttr0(uint8_t value) {
  esp_zb_lock_acquire(portMAX_DELAY);
  ESP_ERROR_CHECK(esp_zb_zcl_set_attribute_val(
    _endpoint, ZB_ZCL_CLUSTER_ID_MY, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ZB_ZCL_ATTR_MY_ATTR0, &value, false
  ));
  esp_zb_lock_release();
}

bool ZigbeeMySensor::reportAttribute(uint16_t attributeID) {
  esp_zb_zcl_report_attr_cmd_t report_attr_cmd;
  report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
  report_attr_cmd.attributeID = attributeID;
  report_attr_cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI;
  report_attr_cmd.clusterID = ZB_ZCL_CLUSTER_ID_MY;
  report_attr_cmd.zcl_basic_cmd.src_endpoint = _endpoint;
  // report_attr_cmd.manuf_code = ESP_ZB_ZCL_ATTR_NON_MANUFACTURER_SPECIFIC;

  esp_zb_lock_acquire(portMAX_DELAY);
  esp_err_t ret = esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd);
  esp_zb_lock_release();

  if (ret != ESP_OK) {
    log_e("Failed to send report: 0x%x: %s", ret, esp_err_to_name(ret));
    return false;
  }
  log_v("Report sent");
  return true;
}
```

I'm running ESP IDF 5.4.1

Any help appreciated, thank you!

xieqinan, 2025-03-31:
Could you please share the complete process of creating the Zigbee data model? I suspect that the `_cluster_list` has not been registered with the Zigbee device.

florian-bailly, 2025-03-31:
The main code is this one:
```cpp
#define ZIGBEE_ENDPOINT 10

ZigbeeMySensor zbDevice = ZigbeeMySensor(ZIGBEE_ENDPOINT);

zbDevice.setManufacturerAndModel("testManu", "testModel");
Zigbee.addEndpoint(&zbDevice);
```

the basic and occupancy cluster is done through the `ZigbeeOccupancySensor` class provided by the [Zigbee ESP Arduino lib](https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee/src).

so when I call the `Zigbee.addEndpoint`, it does:
```cpp
esp_zb_ep_list_add_ep(_zb_ep_list, ep->_cluster_list, ep->_ep_config);
```

florian-bailly, 2025-04-13:
@xieqinan do you think it's related to the zigbee-sdk or the Arduino ESP32 implementation?

xieqinan, 2025-04-14:
Based on your implementation, it seems the custom cluster should have already been added to the Zigbee stack. How did you determine that the custom cluster was not found?

If you're using `esp_zb_zcl_report_attr_cmd_req()` to report a custom attribute in `ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT` mode, please ensure that the custom cluster has been added to the binding table.

To verify which clusters have been added to the current device, you can call the `esp_zb_zdo_simple_desc_req()` function from the remote device. This will allow you to retrieve the list of input and output clusters that the target device currently supports.

florian-bailly, 2025-05-17:
thanks @xieqinan for bumping this topic!

I started from scratch again and got something working by adding a "custom" attribute to an existing cluster.

In the end it's the same as a custom cluster for my need.

The issue was also on the herdsman converter side, but as I didn't have any debug info it was hard to know where to dig.

Here's the part of the code to add the attribute:
```c++
#define ZB_ZCL_ATTR_CUSTOM_1 = 0xff00

esp_zb_attribute_list_t *cluster = esp_zb_cluster_list_get_cluster(_cluster_list, ESP_ZB_ZCL_CLUSTER_ID_OCCUPANCY_SENSING, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

uint8_t attr_custom_1 = 0;
ESP_ERROR_CHECK(esp_zb_cluster_add_attr(cluster, ESP_ZB_ZCL_CLUSTER_ID_OCCUPANCY_SENSING, ZB_ZCL_ATTR_CUSTOM_1, ESP_ZB_ZCL_ATTR_TYPE_8BIT, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY|ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &attr_custom_1));
```

and the herdsman converter:
```js
const {occupancy} = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');

const occupancy_custom_attr = {
    cluster: 'msOccupancySensing',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        const value = msg.data[65280]; // 0xff00 in decimal

        if (typeof value !== 'undefined') {
            return {customAttr1: value === 1};
        }
    },
};

const definition = {
    zigbeeModel: ['ZigbeeOccupancyPIRSensor'],
    model: 'ZigbeeOccupancyPIRSensor',
    vendor: 'Espressif',
    description: 'Occupancy sensor',
    extend: [occupancy()],
    fromZigbee: [occupancy_custom_attr],
    exposes: [
        exposes.binary('customAttr1', exposes.access.STATE, true, false)
            .withDescription('Custom Attribute 1'),
    ],
    meta: {},
};

module.exports = definition;
```

It was also helpful to enable verbose flags in my configuration (Arduino):
```ini
  -DLOG_LOCAL_LEVEL=ESP_LOG_VERBOSE
  -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
```

I hope it'll help others.

I might try with a custom cluster when the project is finished in order to provide a working example.
In the meantime I close this topic, feel free to reopen it.

closed 2025-05-17 by florian-bailly