espressif/esp-zigbee-sdk #617: The degree of low power consumption

Mushan-bin, 2025-04-02:
In addition to generating 30ma current when sending data on zigbee terminal, I wonder whether it can be achieved at single-digit microamps at other times. I've used the light_sleep[light sleep README](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/light_sleep#example-output) and deep_sleep examples[Deep sleep README](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/deep_sleep), but I still haven't reached this point.

Mushan-bin, 2025-04-02:
At the same time, when using the ESP32-H2 development board to burn light_sleep and deep_sleep programs, the sleep power consumption is 1mA, is there any other operation needed to achieve the single-digit microamp level reflected in the picture?

xieqinan, 2025-04-07:
> At the same time, when using the ESP32-H2 development board to burn light_sleep and deep_sleep programs, the sleep power consumption is 1mA, is there any other operation needed to achieve the single-digit microamp level reflected in the picture?

Please share your test environment and the steps you followed for measuring the current of the sleep device. If you followed the above [README](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/light_sleep#example-output) but obtained different results, kindly provide the log and a screenshot of your current measurement for further analysis.

tomasz90, 2025-04-14:
@Mushan-bin test your board first without zigbee, just test light sleep and deep sleep modes alone. This will tell you what is a "floor".
Sometimes boards don't have efficient voltage regulators.

xieqinan, 2025-05-16:
@Mushan-bin 

Do you have any updates?

xieqinan, 2025-06-13:
@Mushan-bin ,

The [FAQs](https://github.com/espressif/esp-zigbee-sdk/blob/main/docs/en/faq.rst#a14-zigbee-light-sleep-mode) can be used as a reference to address your question.

For issues related to light-sleep or deep-sleep, please refer to the [esp-idf light sleep example](https://github.com/espressif/esp-idf/tree/master/examples/system/light_sleep), which provides more detailed steps for testing and achieving low power consumption.

Since there has been no feedback for some time, I’ll go ahead and close this issue. Feel free to reopen it if you're still experiencing confusion or need further clarification.

closed 2025-06-13 by xieqinan