espressif/esp-zigbee-sdk #620: Trying to create simple Zigbee device with electrical measurement cluster

InnerBushman, 2025-04-09:
### Question

I'm trying to create simple Zigbee device using  electrical measurement cluster.
I modified existing example for temperature sensor by replacing definitions for temperature cluster with definitions for electrical measurement. I'm at a point the sketch compiles but when I run it I'm encountering the error: ```E (259) ESP_ZIGBEE_CLUSTER: The requested cluster ID (0x b04) is not found```.

My question is, is the electrical measurement cluster already implemented in the library (3.2.0)?
My code is just cobbled together cause I'm not a real coder. I don't even understand how the whole Zigbee thing works. I'm just doing my best to follow examples.
If the cluster is already implemented, can someone help me create the most basic example using it?

### Additional context.

[Zigbee_test02_elec.zip](https://github.com/user-attachments/files/19674894/Zigbee_test02_elec.zip)

lpy4105, 2025-04-10:
Hi @InnerBushman 
Going through the provided example, I believe the problem should be your were not creating the correct cluster:
```c
ZigbeeElectricMeas::ZigbeeElectricMeas(uint8_t endpoint) : ZigbeeEP(endpoint) {
  _device_id = ESP_ZB_HA_TEMPERATURE_SENSOR_DEVICE_ID;
  _humidity_sensor = false;

  esp_zb_temperature_sensor_cfg_t temp_sensor_cfg = ESP_ZB_DEFAULT_TEMPERATURE_SENSOR_CONFIG();
  _cluster_list = esp_zb_temperature_sensor_clusters_create(&temp_sensor_cfg);

  _ep_config = {
    .endpoint = _endpoint, .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID, .app_device_id = ESP_ZB_HA_TEMPERATURE_SENSOR_DEVICE_ID, .app_device_version = 0
  };
}
```

You need to change the `_device_id` to correct value and initiate `_cluster_list` with correct functions. In your case, 
- use `esp_zb_zcl_cluster_list_create()` to create an empty cluster list;
- use `esp_zb_electrical_meas_cluster_create()` to create the cluster;
- use `esp_zb_cluster_list_add_electrical_meas_cluster()` to add the cluster.

We have guide may help you to understand the data model and get started: https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#developing-your-product

BTW, for Arduino stuff, you'd better to raise a issue in https://github.com/espressif/arduino-esp32

InnerBushman, 2025-04-10:
Thank you for your reply.
> * use `esp_zb_electrical_meas_cluster_create()` to create the cluster;

I'm trying to create the cluster like you said but I don't know what to provide as a parameter. 
In the original example for temperature sensor it is done like this:

```
 esp_zb_temperature_sensor_cfg_t temp_sensor_cfg = ESP_ZB_DEFAULT_TEMPERATURE_SENSOR_CONFIG();
_cluster_list = esp_zb_temperature_sensor_clusters_create(&temp_sensor_cfg);
```

where `ESP_ZB_DEFAULT_TEMPERATURE_SENSOR_CONFIG` is defined in `ha/esp_zigbee_ha_standard.h`.
But when I try to do the same with:

```
  esp_zb_electrical_meas_cluster_cfg_t elec_sensor_cfg = ESP_ZB_DEFAULT_ELECTRICAL_MEASUREMENT_CONFIG();
  _cluster_list = esp_zb_electrical_meas_cluster_create(&elec_sensor_cfg);
```
the `ESP_ZB_DEFAULT_ELECTRICAL_MEASUREMENT_CONFIG` (or anything remotely similar) is not defined anywhere. 
I do not know how to define it. Following example of other already defined clusters didn't really work out for me. Anything I tried so far only causes all sorts of errors during compilation, only proving the point I have no idea what I'm really doing.
The information provided [in the docs](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/esp_zigbee_cluster.html#_CPPv437esp_zb_electrical_meas_cluster_createP36esp_zb_electrical_meas_cluster_cfg_t) didn't really tell me anything either. I'm assuming I'm suppose to provide `uint32_t measured_type` [as indicated here](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/esp_zigbee_type.html#_CPPv436esp_zb_electrical_meas_cluster_cfg_s), and the type can be taken (I presume) from `esp_zigbee_zcl_electrical_meas.h`. But how do I do that correctly?

lpy4105, 2025-04-11:
> I'm assuming I'm suppose to provide uint32_t measured_type [as indicated [...]

Yes, that's correct, the valid value for "measured_type" could be found in the corresponding cluster header as well as in our [API doc](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/zcl/esp_zigbee_zcl_electrical_meas.html#_CPPv452esp_zb_zcl_electrical_measurement_measurement_type_t).

InnerBushman, 2025-04-11:
Assuming I want to create a cluster with just `ESP_ZB_ZCL_ELECTRICAL_MEASUREMENT_DC_MEASUREMENT  = 0x00000040,     /*!< DC Measurement bit */` how should I pass it to `_cluster_list = esp_zb_temperature_sensor_clusters_create(>>>HERE<<<);`
Can you give a one liner example of how to correctly do it? No matter how I try to pass it I keep getting conversion errors like:
```
error: cannot convert 'esp_zb_attribute_list_t*' {aka 'esp_zb_attribute_list_s*'} to 'esp_zb_cluster_list_t*' {aka 'esp_zb_cluster_list_s*'} in assignment
   14 |  _cluster_list = esp_zb_electrical_meas_cluster_create(&elec_sensor_cfg);
```

lpy4105, 2025-04-11:
Try the followings:
```c
esp_zb_electrical_meas_cluster_cfg_t elec_sensor_cfg = {.measured_type = ESP_ZB_ZCL_ELECTRICAL_MEASUREMENT_DC_MEASUREMENT};
esp_zb_attribute_list_t *elec_meas_cluster = esp_zb_electrical_meas_cluster_create(&elec_sensor_cfg);
_cluster_list = esp_zb_zcl_cluster_list_create();
esp_zb_cluster_list_add_electrical_meas_cluster(_cluster_list, elec_meas_cluster, ZB_ZCL_CLUSTER_SERVER_ROLE);
```

InnerBushman, 2025-04-21:
Thank you for your help. Sorry for the late reply as I was sick in bed. I've tried your example and I was able to make the code compile, connect and expose voltage/current/power in my home assistant. 
I still have some errors during start up in the serial monitor and the reported values are all N/A, but it's something. I struggle to find some definitions related to electrical measurement cluster but I'm slowly chipping away on it. For example I'm not sure what I should put in place of `ESP_ZB_HA_LIGHT_SENSOR_DEVICE_ID` so for now I'm putting `ESP_ZB_HA_TEST_DEVICE_ID` in. There's no electrical measurement `esp_zb_ha_standard_devices_t` list in `esp_zigbee_zcl_common.h`. Does that mean it's not considered "home automation" device? Or is it just not defined yet?
I'll try to upload something resembling a code when I run into more roadblocks (and I'm sure there will be some cause I can already see them coming) but until then I'll try to clean it up because right now it's a mess.

lpy4105, 2025-04-21:
> Does that mean it's not considered "home automation" device? Or is it just not defined yet?

There is no HA device that only contains a single electrical measurement cluster, but it is a optiional common cluster for all HA devices.

InnerBushman, 2025-04-26:
I followed your advice and asked on the arduino side of ESP32 repo. Although they could not help me currently, the electrical measurement cluster example has been added to to do list. I will close this issue for now and wait for the example. Thanks @lpy4105 .

InnerBushman, 2025-04-26:
Will be awaiting example from https://github.com/espressif/arduino-esp32 repository.

closed 2025-04-26 by InnerBushman