espressif/esp-zigbee-sdk #621: Zigbee stack initialization block，watchdog got triggered.

Jayliu0712, 2025-04-10:
### Question

When I create a Zigbee terminal task，the function esp_zb_init() was blocked，and watchdog was triggered.My code is as follows(Irrelevant parts have been omitted). I did some debugging. When I commented out the SPI screen initialization program, the esp_zb_task could be created and run normally. Could you please give me some advice？

### Additional context.

void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack with Zigbee coordinator config */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_sleep_enable(true);
    /*zigbee stack init*/
    esp_zb_init(&zb_nwk_cfg);

    ESP_ERROR_CHECK(zb_endpoint_cluster_attribute());

    esp_zb_core_action_handler_register(zb_action_handler);

    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    static uint8_t long_addr[] = {0xaa, 0xaa, 0xaa, 0xaa, 0xbb, 0xbb, 0xbb, 0xb5};
    static uint8_t install_code[] = {0x96, 0x6b, 0x9f, 0x3e, 0xf9, 0x8a, 0xe6, 0x05, 0x97, 0x08}; // IC + CRC
    esp_zb_set_long_address(long_addr);
    esp_zb_secur_ic_set(ESP_ZB_IC_TYPE_64, install_code);
    ESP_ERROR_CHECK(esp_zb_start(false));
    // esp_zb_main_loop_iteration();
    esp_zb_stack_main_loop();
}

void esp_send_task(void *pvParameters)
{
    .
    .
    .
    while (1)
    {
        .
        .
        .
                ulTaskNotifyTake(pdTRUE, portMAX_DELAY); //triggered by button interrupt
                xTaskCreate(esp_zb_task, "esp_zb_task", 1024 * 8, NULL, 5, &zbTaskHandle);//run only once
        .
        .
        .
    }
}

void app_main(void)
{
    /*esp light sleep initialization*/
    esp_err_t rc = ESP_OK;
#ifdef CONFIG_PM_ENABLE
    int cur_cpu_freq_mhz = CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ;
    esp_pm_config_t pm_config = {
        .max_freq_mhz = cur_cpu_freq_mhz,
        .min_freq_mhz = cur_cpu_freq_mhz,
#if CONFIG_FREERTOS_USE_TICKLESS_IDLE
        .light_sleep_enable = true
#endif
    };
    rc = esp_pm_configure(&pm_config);
#endif
    return rc;
    .
    .
    .
    .
    .
    .
    .
    /*SPI SCREEN INIT*/
       spi_bus_config_t bus_config = {
        .miso_io_num = SCREEN_MISO,
        .mosi_io_num = SCREEN_MOSI,
        .sclk_io_num = SCREEN_CLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        // .flags = SPICOMMON_BUSFLAG_MASTER,
        // .max_transfer_sz = 240 * 320 * sizeof(uint16_t)
    };
    spi_bus_initialize(SPI2_HOST, &bus_config, SPI_DMA_CH_AUTO);

    spi_device_interface_config_t dev_config = {
        // .clock_source = SPI_CLK_SRC_DEFAULT,
        .clock_source = SOC_MOD_CLK_XTAL,
        .command_bits = 0,
        .address_bits = 0,
        .dummy_bits = 0,
        .clock_speed_hz = 400 * 1000,
        .mode = 3,
        .spics_io_num = SCREEN_CS,
        .pre_cb = spi_pre_cb,
        .post_cb = NULL,
        .queue_size = 5,
    };
    spi_bus_add_device(SPI2_HOST, &dev_config, spi);

       gpio_config_t lcd_gpioConfig = {
        // .pin_bit_mask = (1ULL << SCREEN_RST) | (1ULL << SCREEN_DC) | (1ULL << SCREEN_BK_LIGHT),
        .pin_bit_mask = (1ULL << SCREEN_RST) | (1ULL << SCREEN_DC),
        .mode = GPIO_MODE_OUTPUT,
        .pull_up_en = GPIO_PULLUP_DISABLE,
        .pull_down_en = GPIO_PULLDOWN_DISABLE,
        .intr_type = GPIO_INTR_DISABLE,
    };
    gpio_config(&lcd_gpioConfig);

    gpio_set_level(SCREEN_RST, 0);
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    gpio_set_level(SCREEN_RST, 1);

    vTaskDelay(20);
    int cmd = 0;
    while (lcd_init_cmds[cmd].databytes != 0xff)
    {
        lcd_cmd(spi, lcd_init_cmds[cmd].cmd, false);
        if (lcd_init_cmds[cmd].databytes & 0x80)
        {
            vTaskDelay(5);
        }
        cmd++;
    }
    .
    .
    .
    .
    .
    .
    .
    xTaskCreate(esp_send_task, "esp_send_task", 1024 * 8, NULL, 5, &zbSendTaskHandle);
}

xieqinan, 2025-04-11:
Could you please reorganize your code? The logic appears confusing—possibly due to formatting issues during pasting—and it's difficult to understand your exact intention. Are you trying to implement the following sequence?

- Initialize and enable the LCD  
- Wait for a button event to trigger the Zigbee stack  
- Enable a sleep end device using the install code commissioning method  
- Encounter an issue where `esp_zb_init()` blocks and the watchdog is triggered

If so, please try separating the above steps and ensure that each step runs correctly in isolation.

Jayliu0712, 2025-04-14:
> Hi [@Jayliu0712](https://github.com/Jayliu0712) ,
> 
> Please also share the log when the watchdog is triggered.

@xieqinan 
Sure.The log is as follow.

> I (23) boot: ESP-IDF v5.3.2 2nd stage bootloader
> I (24) boot: compile time Apr 10 2025 17:02:47
> I (25) boot: chip revision: v0.1
> I (26) boot: efuse block revision: v0.2
> I (31) boot.esp32h2: SPI Speed      : 64MHz
> I (36) boot.esp32h2: SPI Mode       : DIO
> I (41) boot.esp32h2: SPI Flash Size : 4MB
> I (45) boot: Enabling RNG early entropy source...
> I (51) boot: Partition Table:
> I (54) boot: ## Label            Usage          Type ST Offset   Length
> I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (76) boot:  2 factory          factory app      00 00 00010000 0015e000
> I (84) boot:  3 zb_storage       Unknown data     01 81 0016e000 00004000
> I (91) boot:  4 zb_fct           Unknown data     01 81 00172000 00000400
> I (99) boot:  5 rcp_fw           Unknown data     01 82 00173000 0007d000
> I (107) boot: End of partition table
> I (111) esp_image: segment 0: paddr=00010020 vaddr=42070020 size=2ae40h (175680) map
> I (169) esp_image: segment 1: paddr=0003ae68 vaddr=40800000 size=051b0h ( 20912) load
> I (177) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=649d0h (412112) map
> I (296) esp_image: segment 3: paddr=000a49f8 vaddr=408051b0 size=0dc74h ( 56436) load
> I (314) esp_image: segment 4: paddr=000b2674 vaddr=40812e30 size=04cc4h ( 19652) load
> I (322) esp_image: segment 5: paddr=000b7340 vaddr=50000000 size=0005ch (    92) load
> I (323) esp_image: segment 6: paddr=000b73a4 vaddr=5000005c size=00008h (     8) load
> I (335) boot: Loaded app from partition at offset 0x10000
> I (335) boot: Disabling RNG early entropy source...
> I (352) cpu_start: Unicore app
> W (877) clk: 32 kHz clock not found, switching to internal 150 kHz oscillator
> I (886) cpu_start: Pro cpu start user code
> I (887) cpu_start: cpu freq: 48000000 Hz
> I (887) app_init: Application information:
> I (889) app_init: Project name:     USB_senosor
> I (894) app_init: App version:      1
> I (899) app_init: Compile time:     Apr 14 2025 14:44:33
> I (905) app_init: ELF file SHA256:  5ab68308eb0e526d...
> I (910) app_init: ESP-IDF:          v5.3.2
> I (915) efuse_init: Min chip rev:     v0.0
> I (920) efuse_init: Max chip rev:     v0.99 
> I (925) efuse_init: Chip rev:         v0.1
> I (930) heap_init: Initializing. RAM available for dynamic allocation:
> I (937) heap_init: At 4081C130 len 00031250 (196 KiB): RAM
> I (943) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
> I (951) spi_flash: detected chip: generic
> I (954) spi_flash: flash io: dio
> I (959) sleep: Configure to isolate all GPIO pins in sleep state
> I (965) sleep: Enable automatic switching of GPIO sleep configuration
> I (984) main_task: Started on CPU0
> I (984) main_task: Calling app_main()
> I (984) pm: Frequency switching config: CPU_MAX: 48, APB_MAX: 48, APB_MIN: 48, Light sleep: ENABLED
> I (984) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
> I (994) gpio: GPIO[25]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
> I (1004) gpio: GPIO[8]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
> I (1014) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
> I (1024) gpio: GPIO[11]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
> I (1054) ADC_BATTARY: Calibration Success
> I (1054) sleep_clock: System Power, Clock and Reset sleep retention initialization
> I (1064) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
> I (1074) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
> I (2584) SPI_SCREEN: clear screen
> I (2584) ADC_BATTARY: ADC1 Channel[1] Raw Data: 4081
> I (2584) ADC_BATTARY: ADC1 Channel[1] VBAT_TEST: 1057 mV
> I (2594) ADC_BATTARY: ADC1 Channel[1] VBAT: 5602 mV
> I (2614) uart: queue free spaces: 10
[... 42 lines cut ...]
> T3      : 0x00fdbdbd  T4      : 0x00fcbcbc  T5      : 0x00000004  T6      : 0x00000001
> MSTATUS : 0x00001889  MTVEC   : 0x40800001  MCAUSE  : 0xdeadc0de  MTVAL   : 0xdeadc0de
> --- 0x40800001: _vector_table at C:/Users/Administrator/esp/v5.3.2/esp-idf/components/riscv/vectors_intc.S:54
> 
> MHARTID : 0x00000000
> E (17474) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
> E (17474) task_wdt:  - IDLE (CPU 0)
> E (17474) task_wdt: Tasks currently running:
> E (17474) task_wdt: CPU 0: esp_zb_task
> E (17474) task_wdt: Print CPU 0 (current core) backtrace
> esp_backtrace_print: Print CPU 0 (current core) registers
> Core  0 register dump:
> MEPC    : 0x4204eba0  RA      : 0x4204eb96  SP      : 0x40829700  GP      : 0x40813634
> --- 0x4204eba0: pwdet_tone_start at ??:?
> 0x4204eb96: pwdet_tone_start at ??:?
> 
> TP      : 0x40829840  T0      : 0x00303737  T1      : 0x00000063  T2      : 0x00000011
> S0/FP   : 0x00000000  S1      : 0x00000000  A0      : 0x00000060  A1      : 0x0001fffc
> A2      : 0x000111ce  A3      : 0x600a1000  A4      : 0x00000007  A5      : 0x00000000
> A6      : 0x0000001e  A7      : 0x00000000  S2      : 0x00000004  S3      : 0x42071000
> S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
> S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
> T3      : 0x00fdbdbd  T4      : 0x00fcbcbc  T5      : 0x00000004  T6      : 0x00000001
> MSTATUS : 0x00001889  MTVEC   : 0x40800001  MCAUSE  : 0xdeadc0de  MTVAL   : 0xdeadc0de
> --- 0x40800001: _vector_table at C:/Users/Administrator/esp/v5.3.2/esp-idf/components/riscv/vectors_intc.S:54