espressif/esp-zigbee-sdk #622: Zigbee network steering was not successful

Jayliu0712, 2025-04-11:
### Question

I delete esp_zb_task （This task created by another task.My code is as follow.Espressif soc platform config set and zigbee stack initialization by
 app_main()，just run once.)，when zigbee disconnect for saving power.But when i manually trigger the task creation event，i can’t connect zigbee network anymore.(Zigbee app signal status gets stuck at ESP_ZB_BDB_SIGNAL_STEERING，error status is ESP_FAIL）Could you please give some solution or advise？Thank you.

### Additional context.

void esp_zb_task(void *pvParameters)
{
    esp_zb_sleep_enable(true);
    ESP_ERROR_CHECK(zb_endpoint_cluster_attribute());
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    static uint8_t long_addr[] = {0xaa, 0xaa, 0xaa, 0xaa, 0xbb, 0xbb, 0xbb, 0xb5};
    static uint8_t install_code[] = {0x96, 0x6b, 0x9f, 0x3e, 0xf9, 0x8a, 0xe6, 0x05, 0x97, 0x08}; // IC + CRC
    esp_zb_set_long_address(long_addr);
    esp_zb_secur_ic_set(ESP_ZB_IC_TYPE_64, install_code);
    ESP_ERROR_CHECK(esp_zb_start(false));
    // esp_zb_main_loop_iteration();
    esp_zb_stack_main_loop();
}

xieqinan, 2025-04-11:
> I deleted `esp_zb_task` (This task was created by another task. My code is as follows: Espressif SoC platform config and Zigbee stack initialization are handled in `app_main()` and run only once), when Zigbee disconnects, to save power.

What is your intended purpose for deleting `esp_zb_task`?

> But when I manually trigger the task creation event, I can’t connect to the Zigbee network anymore.

I'm not entirely sure what operations are performed before you manually trigger the task creation again. However, please note that the Zigbee stack cannot be started multiple times.

Please share more details and context about your project so we can make sure we're on the same page and properly discuss the issue.

Jayliu0712, 2025-04-14:
> [@xieqinan](https://github.com/xieqinan)
> 
> > What is your intended purpose for deleting `esp_zb_task`?
> 
> Because zigbee terminal searching connect to coordinator will generate 30mA continuous electric current.So if i don’t need connect coordinator I delete ` esp_zb_task` for saving power.
> 
> > Please share more details and context about your project so we can make sure we're on the same page and properly discuss the issue.
> 
> The overall code logic is that terminal device power on.But ` esp_zb_task` not create at once.I set gpio interrupt to trigger `esp_zb_stack` create.When Zigbee_app_signal is ` ESP_ZB_ZDO_SIGNAL_LEAVE` ，other task trigger a logical program delete that task. Under expected circumstances, when I trigger the `esp_zb_stack` to be created again, it should try to connect to and succeed in connecting with the coordinator.But in fact，i can’t connect to coordinator.
> 
> And when terminal connecting coordinator，i configured zigbee sleep to reduce power consumption. But according to the phenomena on the power - consumption detection device, the power consumption didn't decrease. The real - time current remained at 30mA. And when monitoring `zigbee_app_signal` through serial port printing, it was found that the `ESP_ZB_COMMON_SIGNAL_CAN_SLEEP `signal was not detected at all.
> 
> These are my questions. Could you give me an answer or a direction for solution?Thank you.

@xieqinan Hi，I noticed you've read my message, and I understand you might be busy. But if possible, I'd really appreciate a reply.

xieqinan, 2025-04-14:
> Because zigbee terminal searching connect to coordinator will generate 30mA continuous electric current.So if i don’t need connect coordinator I delete  esp_zb_task for saving power.

I don't think deleting the Zigbee task is a good approach to resolve this issue. The operation of searching for a network is triggered by [this line](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_sleep/light_sleep/main/esp_zb_sleepy_end_device.c#L137). If you don't want the device to search for a network periodically, you can comment it out to allow the Zigbee task to enter an idle state.

If your intention is to search for a network only upon a specific event—such as a button press—you can manually call this function within the corresponding event handler.

Jayliu0712, 2025-04-14:
> I don't think deleting the Zigbee task is a good approach to resolve this issue. The operation of searching for a network is triggered by [this [...]

Ok，I see.Let me try your advice.Thanks for your help！

xieqinan, 2025-05-16:
@Jayliu0712 

Do you have any updates?

closed 2025-05-16 by Jayliu0712