espressif/esp-zigbee-sdk #623: Light sleep example doesn't work, it is not connecting to zigbee network

tomasz90, 2025-04-14:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1-1-gf117477c8

### esp-zigbee-lib version.

~1.6.0

### esp-zboss-lib version.

~1.6.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Working example of light sleep. Device is joining network, or if there is issues, it should be at least not consume 50mA.

### What is the actual behavior?

I get this from serial monitor: 

```
idf.py flash monitor                                          ✔  3.9.9   system   04:30:08  
Executing action: flash
Serial port /dev/cu.usbmodem1101
Connecting...
Detecting chip type... ESP32-C6
Running ninja in directory /Users/tomasz/espressif/esp-zigbee-sdk/examples/esp_zigbee_sleep/light_sleep/build
Executing "ninja flash"...
[1/5] cd /Users/tomasz/espressif/esp-zigbee-sdk/examples/esp_zigbee_sleep/light_sleep/...essif/esp-zigbee-sdk/examples/esp_zigbee_sleep/light_sleep/build/sleepy_end_device.bin
sleepy_end_device.bin binary size 0x7c9a0 bytes. Smallest app partition is 0xe1000 bytes. 0x64660 bytes (45%) free.
[1/1] cd /Users/tomasz/espressif/esp-zigbee-sdk/examples/esp_zigbee_sleep/light_sleep/...f/esp-zigbee-sdk/examples/esp_zigbee_sleep/light_sleep/build/bootloader/bootloader.bin
Bootloader binary size 0x55f0 bytes. 0x2a10 bytes (33%) free.
[4/5] cd /Users/tomasz/espressif/esp-idf-v5.4.1/components/esptool_py && /Users/tomasz... -P /Users/tomasz/espressif/esp-idf-v5.4.1/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32c6 -p /dev/cu.usbmodem1101 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 sleepy_end_device.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.8.1
Serial port /dev/cu.usbmodem1101
Connecting...
Chip is ESP32-C6FH4 (QFN32) (revision v0.1)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 7c:2c:67:ff:fe:64:b9:d8
BASE MAC: 7c:2c:67:64:b9:d8
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x0008cfff...
Flash will be erased from 0x00008000 to 0x00008fff...
SHA digest in image updated
Compressed 22000 bytes to 13616...
Writing at 0x00000000... (100 %)
Wrote 22000 bytes (13616 compressed) at 0x00000000 in 0.3 seconds (effective 532.5 kbit/s)...
Hash of data verified.
Compressed 510368 bytes to 296503...
Writing at 0x0008bb85... (100 %)
Wrote 510368 bytes (296503 compressed) at 0x00010000 in 4.0 seconds (effective 1013.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 136...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (136 compressed) at 0x00008000 in 0.0 seconds (effective 532.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /Users/tomasz/espressif/esp-zigbee-sdk/examples/esp_zigbee_sleep/light_sleep
Executing "/Users/tomasz/.espressif/python_env/idf5.4_py3.13_env/bin/python /Users/tomasz/espressif/esp-idf-v5.4.1/tools/idf_monitor.py -p /dev/cu.usbmodem1101 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c6 --revision 0 --decode-panic backtrace /Users/tomasz/espressif/esp-zigbee-sdk/examples/esp_zigbee_sleep/light_sleep/build/sleepy_end_device.elf -m '/Users/tomasz/.espressif/python_env/idf5.4_py3.13_env/bin/python' '/Users/tomasz/espressif/esp-idf-v5.4.1/tools/idf.py'"...
--- esp-idf-monitor 1.6.2 on /dev/cu.usbmodem1101 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
I (ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0x7f (SPI_FAST_FLASH_BOOT)
Saved PC:0x400294a6
--- 0x400294a6: uart_hal_get_txfifo_count in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe84
load:0x4086e610,len:0x3058
entry 0x4086c11a
I (23) boot: ESP-IDF v5.4.1-1-gf117477c8-dirty 2nd stage bootloader
I (23) boot: compile time Apr 14 2025 03:46:09
I (24) boot: chip revision: v0.1
I (24) boot: efuse block revision: v0.3
I (28) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 2MB
I (39) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (59) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (66) boot:  2 factory          factory app      00 00 00010000 000e1000
I (72) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (79) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (86) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=42060020 size=0fb90h ( 64400) map
I (108) esp_image: segment 1: paddr=0001fbb8 vaddr=40800000 size=00460h (  1120) load
I (109) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=5a108h (368904) map
I (180) esp_image: segment 3: paddr=0007a130 vaddr=40800460 size=10530h ( 66864) load
I (195) esp_image: segment 4: paddr=0008a668 vaddr=40810990 size=02288h (  8840) load
I (198) esp_image: segment 5: paddr=0008c8f8 vaddr=50000000 size=00050h (    80) load
I (199) esp_image: segment 6: paddr=0008c950 vaddr=50000050 size=00020h (    32) load
I (210) boot: Loaded app from partition at offset 0x10000
I (212) boot: Disabling RNG early entropy source...
I (228) cpu_start: Unicore app
--- Error: read failed: [Errno 6] Device not configured
--- Waiting for the device to reconnect
I (2706) ESP_ZB_SLEEP: Network steering was not successful (status: -1)
I (2707) ESP_ZB_SLEEP: Zigbee can sleep
--- Error: read failed: [Errno 6] Device not configured
--- Waiting for the device to reconnect.............................
```

### Steps to reproduce.

1. Open light sleep example
2. idf.py set-target esp32c6
3. idf.py erase-flash
4. idf.py build
5. idf.py flash monitor

xieqinan, 2025-04-14:
Hi @tomasz90,

Have you followed the [README](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/light_sleep) to run your test? If so, please double-check whether your Zigbee network is open.

> Device is not joining my network. It consumes 50mA, exactly like a radio is turned on constantly.

Since the application is designed to send radio signals every 1 second to search for an available network, the device doesn't enter light sleep mode during this process. Could you also please share your test method and result?

tomasz90, 2025-04-14:
https://github.com/user-attachments/assets/25456f3a-4dca-417e-9b94-1095b4a2d445

After program was flashed I clicked a reset button, because program will not start automatically.

In home assistant addon (right screen) I was able to add this board before, but using arduino on/off light zigbee example.

xieqinan, 2025-04-15:
> In home assistant addon (right screen) I was able to add this board before, but using arduino on/off light zigbee example.

I think the light sleep behavior is the same as the Arduino on/off light Zigbee example during the commissioning stage. I’ve just tested this example and confirmed that it can join the Home Assistant network, so something might be missing on your side.

Would you mind capturing a sniffer packet for us to analyze? You can refer to the [documentation](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark) for instructions on installing and using the sniffer with Wireshark.

> Since the application is designed to send radio signals every 1 second to search for an available network, the device doesn't enter light sleep mode during this process. Could you also please share your test method and result?

Sorry for the confusion—what I actually meant to ask is how the power consumption is being tested on your side.

tomasz90, 2025-04-16:
I also tried different sketch like HA_on_off_light, but the board is not connecting either. But for instance arduino example is working fine.

https://github.com/user-attachments/assets/77d2a61b-681f-4319-89b3-b73178895b42

oddlama, 2025-04-16:
I see you are using a XIAO esp32c6, have you made sure to enable the antenna switch chip? (set GPIO3 to low)
If that pin is now pulled low, you will have neither the internal nor the external antenna connected.

tomasz90, 2025-04-16:
@oddlama This is exactly Seeed Xiao ESP32C6, thanks for the tip! I didn't have this set, but I just tried it right now, without further success. I just checked their docs, not sure if it is exactly necessary.
I also tested above examples on the other board, waveshare ESP32-C6-Zero, same result, arduino sketch is working, 2 mentioned sdk examples just didn't.

[image]

tomasz90, 2025-04-17:
Now I tried to use official ESP-IDF extension in VSCode and start with examples from there. There are bunch of examples(and 2 zigbee, probably same as in sdk), I picked the first, helloworld, to test if I can simply flash it, yes it worked. Then I tried to run HA_on_off_light, but I just get errors while building project... Some dependencies are missing, I tried to run `git submodule update --init --recursive` on esp-idf directory, but nothing was pulled. Am I missing something? Is instruction faulty in examples? I just wanted to start with ready to use examples, then I will adjust code to my need, but now I am stuck at configuration :(

https://github.com/user-attachments/assets/7f149329-58f5-4180-ba2e-31df821821a9

xieqinan, 2025-04-17:
> I also tried different sketch like HA_on_off_light, but the board is not connecting either. But for instance arduino example is working fine.

I haven’t found any visible differences between the two examples either. Could you try calling `esp_zb_set_long_address()` to set a custom long address for the device, or use a new board to flash the `HA_on_off_light` example? I suspect there might be some cached information related to your board in the Home Assistant system.

It would also be helpful to install a sniffer—it can significantly speed up diagnosing this kind of issue.

By the way, the arduino-esp32 also supports the light sleep feature, why no submit an issue [here](https://github.com/espressif/arduino-esp32/issues) to ask help.

tomasz90, 2025-04-17:
Ok, that was my fault.. I didn't run task erase-flash, every time I wanted to flash project and wanted to newly join a network. 
Indeed light_sleep is working, at least it is joining the network. 

But I have to complain on the other thing, in light sleep power doesn't seem to reduce power consumption, I set longer ED_KEEP_ALIVE to 30s and then I tested with multimeter power consumption. Waveshare ESP32 C6 takes 23mA while in "light sleep", Xiao ESP32 C6 even more. That is two order of magnitude more that what is in README (avg 0.125mA)..

xieqinan, 2025-04-18:
Please share your results in a similar to this: https://github.com/espressif/arduino-esp32/issues/10746#issue-2745927672.  
Otherwise, no one will be able to understand the cause of the issue.

tomasz90, 2025-04-18:
@xieqinan I created a new report, since it is concerning a different issue -> https://github.com/espressif/esp-zigbee-sdk/issues/629

closed 2025-04-18 by tomasz90