espressif/esp-zigbee-sdk #624: Connections between Zigbee terminals and coordinators

Mushan-bin, 2025-04-14:
I want to enable zigbee terminal devices to quickly connect to each coordinator device. For example, when I use a terminal to connect to the first coordinator, I disconnect from the first coordinator and then connect to the second coordinator device. Why is this state unstable? Sometimes the device cannot be successfully connected without re-plugging? Sometimes the initialization stack fails. Can you provide an example of this?

xieqinan, 2025-04-14:
There is currently no official example for your specific requirement. However, when the device disconnects from the first coordinator, it will trigger the `ESP_ZB_ZDO_SIGNAL_LEAVE` signal. You only need to call:

```c
esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_INITIALIZATION, 0);
```

upon receiving this event. This will prompt the device to search for a new network.

Please note that in some cases, the joining process may fail—for example, if there is no open network available, the device misses the beacon request, or it considers the scanned network unsuitable due to weak link quality, etc.

By the way, the [esp-zigbee-sdk issue template](https://github.com/espressif/esp-zigbee-sdk/issues/new/choose) provides a structured format to describe your problem. Could you please use the template when submitting your issue? That would help us better understand your situation and respond more effectively.

xieqinan, 2025-05-16:
Hi @Mushan-bin ,

Do you have any updates?

xieqinan, 2025-06-13:
Hi,

The comment https://github.com/espressif/esp-zigbee-sdk/issues/624#issuecomment-2800614379 attempts to address this problem. Since there has been no feedback for some time, I will go ahead and close this issue.
Please feel free to reopen it if you’re still experiencing confusion or need further clarification.

closed 2025-06-13 by xieqinan