espressif/esp-zigbee-sdk #625: Window Closure position not automatically reported

ram-0000, 2025-04-14:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.0

### esp-zigbee-lib version.

1.6.4

### esp-zboss-lib version.

3.0

### Espressif SoC revision.

ESP32 C6

### What is the expected behavior?

I am developping a Window Closure Zigbee component on a ESP32 C6 super mini card.

I perfectly receive open, close or stop events when I click on corresponding buttons in the Home Assistant view.

But, when window closure position is modified (between 0 and 100%), I would like to automatically receive and view its updated value in Home assistant (for example in the expose view of the zigbee component, see attachement).

For now, I have a task that simulate the movment of the window closure.

[image]

### What is the actual behavior?

The position is not automatically updated. I need to push on reload button to get its updated value.

### Steps to reproduce.

Attached is the code source I use. I tried to make it as simple as possible to isolate my problem.

[Main.txt](https://github.com/user-attachments/files/19733688/Main.txt)

xieqinan, 2025-04-15:
I don’t think this is a bug—you need to bind the report to Home Assistant; otherwise, the device won’t know where to send the report.

Binding to HA when device is connected: 
```c
        uint16_t ha_addr = 0x0000;
        uint8_t ha_ep = 1;
        esp_zb_zdo_bind_req_param_t bind_req;
        esp_zb_ieee_address_by_short(ha_addr, bind_req.dst_address_u.addr_long);
        esp_zb_get_long_address(bind_req.src_address);
        bind_req.src_endp = ENDPOINT_NUMBER;
        bind_req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING;
        bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        bind_req.dst_endp = ha_ep;
        bind_req.req_dst_addr = esp_zb_get_short_address();
        esp_zb_zdo_device_bind_req(&bind_req, NULL, NULL);
```

xieqinan, 2025-04-15:
I'm not very familiar with Home Assistant, but if the above binding operation can be done through the Home Automation page, that would be ideal.

ram-0000, 2025-04-15:
It's a hit !

esp_zb_zdo_device_bind_req() was the soltution to my problem, thank you.

You (or somebody else) may modify the label from bug to question

closed 2025-04-15 by ram-0000