espressif/esp-zigbee-sdk #627: Linker errror with additional c file

marrog86, 2025-04-15:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### esp-zigbee-lib version.

~1.6.0

### esp-zboss-lib version.

~1.6.0

### Espressif SoC revision.

esp32-h2

### What is the expected behavior?

successful build. 

### What is the actual behavior?

undefined reference to `**esp_zcl_utility_add_ep_basic_manufacturer_info**'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

### Steps to reproduce.

Perhaps simply simply this line:  SRCS "esp_zb_light.c"  to the example causes the failure.  
 
How I reproduce: 
3. added new c and h file. 
4. adding in CMakeLists.txt  to idf_component_register: SRCS "esp_zb_light.c" "worker.c"3.build
[example.tar.gz](https://github.com/user-attachments/files/19750676/example.tar.gz)

### More Information.

I only add the file to build in the configuration, and linker error comes up. 
i removed all includes to worker h file, and error still exists. 
simply adding this line:  SRCS "esp_zb_light.c"  to the example causes the failure, once I remove it, it builds correctly.

xieqinan, 2025-04-15:
Sorry for the inconvenience, but this is not a bug—it's simply a compilation issue. Please refer to the implementation of `idf_component_register`:  https://github.com/espressif/esp-idf/blob/master/tools/cmake/component.cmake#L426

Fixs:
```
idf_component_register(
    SRCS "worker.c" "esp_zb_light.c" "../../../common/zcl_utility/src/zcl_utility.c"
    INCLUDE_DIRS "." "../../../common/zcl_utility/include"
)
```

marrog86, 2025-04-16:
Works, Thanks for this hint!

closed 2025-04-16 by marrog86