espressif/esp-zigbee-sdk #628: esp_zb_init() fails with "Failed to find zb_storage partition" on ESP32-C6 with IDF v5.4.1 / ZB Libs v1.6.4 (HA-Zigbee Router)

roboshs, 2025-04-15:
esp_zb_init() asserts at zb_esp_nvram.c:84 ("Failed to find zb_storage partition") when using ESP-IDF v5.4.1 with Zigbee libraries v1.6.4 on an ESP32-C6 target, even when a valid NVS partition (nvs or zb_storage) exists and is correctly specified.

**Project** - 8 Output Relay Control(Lights)/8 Inputs(Physical Switches)/ 1 i2c (SHT44 Humidity/Temperature Sensor) Zigbee Router running on ESP32C6-DEVKITC-1-N8(Power on behavior control, Previous State Save)

I have limited SDK knowledge but i kept trying to correct the code by sending the whole Latest SDK through Sonnet 3.7 as well as gemini 2.5 pro. Even then i kept getting the same error. Please help me out. Stuck on this for quite sometime. Let me know if you need anything else from me.

**Environment**

Target: ESP32-C6 (Revision: v0.0)
ESP-IDF Version: v5.4.1 (v5.4.1-dirty from logs)
esp-zigbee-lib Version: v1.6.4
esp-zboss-lib Version: v1.6.4
Operating System: Windows 11
Build System: idf.py (CMake)
Development Board - ESP32C6-DEVKITC-1-N8

When attempting to initialize the Zigbee stack using esp_zb_init() on an ESP32-C6 with ESP-IDF v5.4.1 and Zigbee libraries v1.6.4, the application crashes consistently during the NVRAM initialization phase within esp_zb_init(). This occurs even when the partition table is correctly configured and flashed with a suitable NVS partition.

I have tried these 3 **partitions.csv**:
```
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,      data, nvs,     ,        48K,
phy_init, data, phy,     ,        4K,
factory,  app,  factory, ,        1M,
 ```
 ```
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,      data, nvs,      ,        24K,
phy_init, data, phy,      ,        4K,
zb_storage, data, nvs,    ,        24K,
factory,  app,  factory,  ,        1M,
 ```
 ```
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,      data, nvs,     0x9000,  24K,
phy_init, data, phy,     0xf000,  4K,
zb_storage, data, nvs,   ,        24K,
factory,  app,  factory,,        1M,
 ```

**Error**
```
ZB_ESP_NVRAM: Failed to find zb_storage partition
Zigbee stack assertion failed /builds/thread_zigbee/esp-zboss/components/zboss_port/src/zb_esp_nvram.c:84
abort() was called at PC 0x4201af39 on core 0

I (1623) ZB_RELAY_CONTROLLER: Starting Zigbee Task
E (1623) ZB_ESP_NVRAM: Failed to find zb_storage partition
Zigbee stack assertion failed /builds/thread_zigbee/esp-zboss/components/zboss_port/src/zb_esp_nvram.c:84
```

**Code main.c**
```c
#include <stdio.h>
#include <string.h>
#include <inttypes.h>

// FreeRTOS includes
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h" // Might be needed for future signaling

// ESP includes
#include "esp_log.h"
#include "esp_err.h"
#include "nvs_flash.h"
#include "esp_timer.h"
[... 977 lines cut ...]
esptool.py v4.8.1
Creating esp32c6 image...
Merged 2 ELF sections
Successfully created esp32c6 image.
Generated C:/Users/Shreyas/zigbee_relay_controller/build/bootloader/bootloader.bin
[114/114] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Users\S...s/zigbee_relay_controller/build/bootloader/bootloader.bin"
Bootloader binary size 0x55f0 bytes. 0x2a10 bytes (33%) free.
[1035/1037] Generating binary image from built executable
esptool.py v4.8.1
Creating esp32c6 image...
Merged 1 ELF section
Successfully created esp32c6 image.
Generated C:/Users/Shreyas/zigbee_relay_controller/build/zigbee_relay_controller.bin
[1036/1037] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Users...zigbee_relay_controller/build/zigbee_relay_controller.bin"
zigbee_relay_controller.bin binary size 0x975c0 bytes. Smallest app partition is 0x100000 bytes. 0x68a40 bytes (41%) free.

Project build complete. To flash, run:
 idf.py flash
or
 idf.py -p PORT flash
or
 python -m esptool --chip esp32c6 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x20000 build\zigbee_relay_controller.bin
or from the "C:\Users\Shreyas\zigbee_relay_controller\build" directory
 python -m esptool --chip esp32c6 -b 460800 --before default_reset --after hard_reset write_flash "@flash_args"
```

xieqinan, 2025-04-16:
Fixs: 
```csv
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,      data, nvs,     0x9000,  24K,
phy_init, data, phy,     0xf000,  4K,
zb_storage, data, fat,   ,        24K,
zb_fct,     data, fat,      ,     1K,
factory,  app,  factory,,         1M,
```

The `zb_storage` and `zb_fct` partitions are essential for storing Zigbee-specific data and are designated as FAT (File Allocation Table) type partitions.

Chris-Mark-Wilson, 2025-04-24:
I have the same problem that ive been trying to fix for a while and this solution doesnt work, ive tried nvs subtype 
i get 
`SP-ROM:esp32c6-20220919`
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x18ec
load:0x4086c110,len:0xe5c
load:0x4086e610,len:0x2f4c
entry 0x4086c11a
I (23) boot: ESP-IDF v5.3.2 2nd stage bootloader
I (24) boot: compile time Apr 24 2025 20:33:37
I (24) boot: chip revision: v0.0
I (26) boot: efuse block revision: v0.2
I (31) boot.esp32c6: SPI Speed      : 80MHz
I (36) boot.esp32c6: SPI Mode       : DIO
I (40) boot.esp32c6: SPI Flash Size : 8MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:ELF file SHA256: 5df422fbe

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
--- 0x4001975a: software_reset_cpu in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x18ec
load:0x4086c110,len:0xe5c
load:0x4086e610,len:0x2f4c
entry 0x4086c11a
I (26) boot: ESP-IDF v5.3.2 2nd stage bootloader
I (27) boot: compile time Apr 24 2025 20:40:43
I (28) boot: chip revision: v0.0
I (30) boot: efuse block revision: v0.2
I (34) boot.esp32c6: SPI Speed      : 80MHz
I (39) boot.esp32c6: SPI Mode       : DIO
I (44) boot.esp32c6: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 zb_storage       WiFi data        01 02 00010000 00006000
I (87) boot:  3 factory          factory app      00 00 00020000 00100000
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00020020 vaddr=420b0020 size=3c2fch (246524) map
I (199) esp_image: segment 1: paddr=0005c324 vaddr=40800000 size=03cf4h ( 15604) load
I (207) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=a51b0h (676272) map
I (458) esp_image: segment 3: paddr=001051d8 vaddr=40803cf4 size=09f38h ( 40760) load
I (476) esp_image: segment 4: paddr=0010f118 vaddr=4080dc30 size=021f4h (  8692) load
I (488) boot: Loaded app from partition at offset 0x20000
I (488) boot: Disabling RNG early entropy source...
I (500) cpu_start: Unicore app
I (509) cpu_start: Pro cpu start user code
I (509) cpu_start: cpu freq: 160000000 Hz
I (510) app_init: Application information:
I (512) app_init: Project name:     zigbee-smart-stat
I (518) app_init: App version:      abe7950-dirty
I (523) app_init: Compile time:     Apr 24 2025 20:40:25
I (529) app_init: ELF file SHA256:  5df422fbe...
I (535) app_init: ESP-IDF:          v5.3.2
I (539) efuse_init: Min chip rev:     v0.0
I (544) efuse_init: Max chip rev:     v0.99 
I (549) efuse_init: Chip rev:         v0.0
I (554) heap_init: Initializing. RAM available for dynamic allocation:
I (561) heap_init: At 40814F90 len 00067680 (413 KiB): RAM
[... 146 lines cut ...]
E (820) ZB_ESP_NVRAM: Failed to find zb_storage partition
ZB_TRACE_LOG[0]: common/zb_debug.c:64    Assertion failed /builds/thread_zigbee/esp-zboss/components/zboss_port/src/zb_esp_nvram.c:84
Zigbee stack assertion failed /builds/thread_zigbee/esp-zboss/components/zboss_port/src/zb_esp_nvram.c:84

abort() was called at PC 0x4201e6cf on core 0
--- 0x4201e6cf: zb_assert at ??:?

--- Stack dump detected
Core  0 register dump:
MEPC    : 0x4080089e  RA      : 0x40807a6e  SP      : 0x40818000  GP      : 0x4080e434  
--- 0x4080089e: panic_abort at /home/chris/esp/esp-idf/components/esp_system/panic.c:463
--- 0x40807a6e: __ubsan_include at /home/chris/esp/esp-idf/components/esp_system/ubsan.c:311

TP      : 0x40818170  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x40818064  A0      : 0x4081802c  A1      : 0x40818062  
A2      : 0x00000000  A3      : 0x40818059  A4      : 0x00000001  A5      : 0x40815000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x420b1000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
--- 0x40800001: _vector_table at /home/chris/esp/esp-idf/components/riscv/vectors_intc.S:54

MHARTID : 0x00000000  `
# and the device continually reboots because it cant 'find' the partition table.. even though it has been defined and pulled in, something to do with nvram?

xieqinan, 2025-04-25:
@Chris-Mark-Wilson ，

> I have the same problem that ive been trying to fix for a while and this solution doesnt work, ive tried nvs subtype
i get

Please use the FAT subtype for the zb_storage and zb_fct.

xieqinan, 2025-04-27:
Hi @roboshs  @Chris-Mark-Wilson,

This solution https://github.com/espressif/esp-zigbee-sdk/issues/628#issuecomment-2808092426 should have already resolved this issue. If there are no further questions, please consider closing it.

Chris-Mark-Wilson, 2025-04-27:
> This solution #628 (comment)

Just making the subtype fat didn't work, no idea what the problem was so
started again from scratch

xieqinan, 2025-04-28:
@Chris-Mark-Wilson ,

Please share your app partition and `main.c` code. I believe this issue is quite clear based on the crashing log above, and it should not be difficult to resolve.

pdEmAk, 2025-05-08:
hi, I have implemented the suggested partitions table like so:

`# Name,       Type, SubType, Offset,   Size, Flags
nvs,          data, nvs,     0x9000,   24K,
phy_init,     data, phy,     0xf000,   4K,
zb_storage,   data, fat,     0x10000,  128K,
zb_fct,       data, fat,    0x30000,  1K,
factory,      app,  factory, 0x40000,  1M,`

However, when I run `idf.py flash monitor`, the partition table in the bootloader shows this:

`I (59) boot: Partition Table:
I (62) boot: ## Label Usage Type ST Offset Length
I (68) boot: 0 nvs WiFi data 01 02 00009000 00006000
I (75) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (81) boot: 2 zb_storage Unknown data 01 81 00010000 00020000
I (88) boot: 3 zb_fct Unknown data 01 81 00030000 00000400
I (94) boot: 4 factory factory app 00 00 00040000 00100000
I (101) boot: End of partition table`

I am trying to use Zigbee with an ESP32 C6 (coordinator) and Tuya CT (end device), but the Tuya CT does not join the network. I have tried everything I can think of. My research points to the partitions.csv not being correct since the partitions table in the bootloader log says "unknown data". 

Does anyone have a fix?

xieqinan, 2025-05-08:
@pdEmAk ,

> I am trying to use Zigbee with an ESP32 C6 (coordinator) and Tuya CT (end device), but the Tuya CT does not join the network. I have tried everything I can think of. My research points to the partitions.csv not being correct since the partitions table in the bootloader log says "unknown data".

This issue is not related to the partition table. Please submit a separate ticket to describe your problem.

Chris-Mark-Wilson, 2025-05-08:
my partition tabke currently looks like this
```
Name,      Type, SubType, Offset,   Size,  Flags
nvs,         data, nvs,     0x9000,   0x6000,
phy_init,    data, phy,     0xf000,   0x1000,
factory,     app,  factory, 0x10000,  2M,
zb_storage,  data, fat,     0x210000, 16K,
zb_fct,      data, fat,     0x214000, 1K,
```
and its working.. finds my sonoff trv and adds it... I have quite a large factory partition because my app is quite large and Im using lvgl on a esp32-c6-lcd 1.47

Make sure all your zigbee settings are correct? in menuconfig->components->zigbee as in set up as a corordinator, uses custom partition table etc

One thing I did waste a good few hours on was esp_zp_app_signal_handler(esp_app_signal_t *signal_struct) absolutely MUST be in the main.c (although you can just call your own signal handler from within it ...

I feel your pain, and I'm basically using AI to muddle my way through it, and every time it gets it wrong, I have to then go and hunt through the libraries (esp-zigbee-lib and esp-zboss-lib) for the correct function and types etc... one small step at a time, testing thoroughly each step before commiting. I am using a weird setup flow whereas I want to add an unknown amount of devices of specific type and then press a button when finished to enter main app but ill post my handler so you can see the enum definitions for the signals (the xqueue stuff is just queuing events for displaying on the lcd)

There does seem to be massive differences between versions, function/enum etc change

If i was gonna guess, id say your using mismatched libraries, or out of date libraries?

If it was me in your position, i'd feed in into claude or copilot, gpt04 whatever and dont expect a correct response, but try it anyway, then feed it back in ad nauseum until it finds the correct settings/definitions/whatever

Probably not much help to you but you never know...

my managed components from idf.component.yml
```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp-zboss-lib: ~1.6.0
  espressif/esp-zigbee-lib: ~1.6.0

  ## Required IDF version
  idf:
    version: '>=5.0.0
  lvgl/lvgl: "~8.3.0"
```

my signal handler in main.c
```
void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    zigbee_signal_handler(signal_struct);
}
```
my actual signal handler (in another file)
```
void zigbee_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p = signal_struct->p_app_signal;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    esp_zb_zdo_signal_device_annce_params_t *dev_annce_params = NULL;

    switch (sig_type) {
 

            case ESP_ZB_ZDO_SIGNAL_PRODUCTION_CONFIG_READY: {  // Signal 23 (0x17)
                ui_event_t event = {
                    .target_screen = SCREEN_BOOT,
                    .message = "Starting Zigbee stack..."
                };
                xQueueSend(ui_event_queue, &event, portMAX_DELAY);
                ESP_LOGI(TAG, "Production config ready, starting initialization");
                break;
            }

            case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP: {  // Signal 6 (0x06)
                ESP_LOGI(TAG, "Skip startup signal received with status: 0x%x", signal_struct->esp_err_status);
                ui_event_t event = {
                    .target_screen = SCREEN_BOOT,
[... 150 lines cut ...]
                (esp_zb_zdo_signal_device_authorized_params_t*)esp_zb_app_signal_get_params(p_sg_p);
            
            // Find the device in our stored list
            for (uint8_t i = 0; i < stored_device_count; i++) {
                if (stored_devices[i].short_addr == auth_params->short_addr) {
                    char message[64];
                    snprintf(message, sizeof(message), "Device %s authorized", stored_devices[i].name);
                    ui_event_t event = {
                        .target_screen = SCREEN_BOOT,
                        .message = ""
                    };
                    strncpy(event.message, message, sizeof(event.message) - 1);
                    xQueueSend(ui_event_queue, &event, 0);
                    ESP_LOGI(TAG, "%s", message);
                    break;
                }
            }
            break;
        }

        default:
            ESP_LOGW(TAG, "Unhandled Zigbee signal: %d (0x%x)", sig_type, sig_type);
            break;
    }
}```

pdEmAk, 2025-05-08:
@xieqinan opened a new issue here: https://github.com/espressif/esp-zigbee-sdk/issues/639. I would appreciate any help.

xieqinan, 2025-05-16:
I believe the solution provided in [this comment](https://github.com/espressif/esp-zigbee-sdk/issues/628#issuecomment-2808092426) can resolve the issue.  Please feel free to reopen the issue if it still persists.

closed 2025-05-16 by xieqinan