espressif/esp-zigbee-sdk #629: The device require too much energy in sleeping mode - 23mA

tomasz90, 2025-04-18:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1-1-gf117477c8

### esp-zigbee-lib version.

~1.6.0

### esp-zboss-lib version.

~1.6.0

### Espressif SoC revision.

Waveshare ESP32-C6 Zero

### What is the expected behavior?

### Hardware Configuration
Nothing connected

### Version
ESP-IDF: 5.4.1, esp-zigbee-sdk: main branch, latest.

### IDE Name
CLion + ESP-IDF extension

### Operating System
MacOS

### Flash frequency
80MHz

### PSRAM enabled
No

### Upload speed
460800

### Description
The device require too much energy in sleeping mode. When sleeping it consumes 23-24mA with shorts spikes up to 250mA every 4 seconds.

### Debug Message
No errors in debug

Have used PowerZ to measure current. Measuring starts just right before I clicked reset button on board. Board was able to establish connection with zigbee network.

[image]

[image]

### What is the actual behavior?

Device consumes 23-24mA while sleeping.

### Steps to reproduce.

Just flash light sleep zigbee example.
1. ESP-IDF extension, task clean
2. ESP-IDF extension, task erase
3. ESP-IDF extension, task flash
4. ESP-IDF extension, task monitor

tomasz90, 2025-04-18:
Here is the same issue on Xiao ESP32C6, 17-19mA in light sleep.

[image]

xieqinan, 2025-04-23:
Hi,

Please double-check the test method. The ESP32-C6 board provides specific pins to measure the chip's power consumption, as shown in the figure below. Are you performing the test in the same manner? Otherwise, the cunsumption is from the broad which including LEDs.

[image]

tomasz90, 2025-04-23:
I have Xiao ESP32C6. It doesn't contain any "special pins", and doesn't have to. I know how current is measured, you just place shunt resistor in series and measure voltage drop. That is how nearly every current meter is designed. 

Any leds were not lit. When I flash program just with deepsleep command, I get 15-17uA (what is expected). 
So there is no issue with my measurements.

tomasz90, 2025-04-24:
Here is a chart from flashed ESP32H2, this example seems to be working correctly, current is so low - below 1mA, where my multimeter is struggling with picking up correct values.

[image]

xieqinan, 2025-04-25:
Hi，

We have reproduced this issue. It is triggered by the GPIO driver of the button. Adding `rtc_gpio_init(CONFIG_GPIO_INPUT_IO_WAKEUP);` can help fix it.

``` c
static esp_err_t deferred_driver_init(void)
{
    static bool is_inited = false;
    if (!is_inited) {
        ESP_RETURN_ON_FALSE(switch_driver_init(button_func_pair, PAIR_SIZE(button_func_pair), zb_buttons_handler),
                            ESP_FAIL, TAG, "Failed to initialize switch driver");
        /* Configure RTC IO wake up:
        The configuration mode depends on your hardware design.
        Since the BOOT button is connected to a pull-up resistor, the wake-up mode is configured as LOW.
        */
        ESP_ERROR_CHECK(esp_sleep_enable_ext1_wakeup(1ULL << CONFIG_GPIO_INPUT_IO_WAKEUP, ESP_EXT1_WAKEUP_ANY_LOW));

#if SOC_RTCIO_INPUT_OUTPUT_SUPPORTED
        rtc_gpio_init(CONFIG_GPIO_INPUT_IO_WAKEUP);
        rtc_gpio_pulldown_dis(CONFIG_GPIO_INPUT_IO_WAKEUP);
        rtc_gpio_pullup_en(CONFIG_GPIO_INPUT_IO_WAKEUP);
#else
        gpio_pulldown_dis(CONFIG_GPIO_INPUT_IO_WAKEUP);
        gpio_pullup_en(CONFIG_GPIO_INPUT_IO_WAKEUP);
#endif
        is_inited = true;
    }
    return is_inited ? ESP_OK : ESP_FAIL;
}
```

tomasz90, 2025-04-25:
@xieqinan I confirm, it fixed the issue.

xieqinan, 2025-04-27:
Thanks for your confirm, this fix has been added to the [light sleep example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_sleep/light_sleep/main/esp_zb_sleepy_end_device.c#L82) of esp-zigbee-sdk, please double check it, if it works fine. please consider closing this issue.

closed 2025-04-27 by tomasz90