espressif/esp-zigbee-sdk #630: Some exposed constants of the ZCL are wrong

oddlama, 2025-04-20:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

latest

### esp-zigbee-lib version.

master

### esp-zboss-lib version.

master

### Espressif SoC revision.

N/A

### What is the expected behavior?

The ZCL should be followed exactly, and exposed constants and values should conform to the ZCL.

### What is the actual behavior?

In noticed that several min/max constants of some cluster attributes are wrong. Apart from the usual naming inconsistencies, there also have been inconsistencies the the zcl definitions itself:

- `ESP_ZB_ZCL_ATTR_PRESSURE_MEASUREMENT_MIN_VALUE_MIN_VALUE` is defined as `((int16_t)0x954d)`. This translates to `-27315` which is quite obviously the temperature of absolute zero i.e. `-273.15 °C`. This is the correct min value for the temperature cluster and according to the ZCL spec there is no specific minimum for pressure.
- Same for `ESP_ZB_ZCL_TEMP_MEASUREMENT_MAX_MEASURED_VALUE_MINIMUM`
- Most attribute comments are prefixed with `/*!<`, but some are prefixed with `/*!` with a missing `<`, e.g. `ESP_ZB_ZCL_FAN_CONTROL_FAN_MODE_OFF`, others yet are missing the space afterwards e.g. `ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID` has `/*!<desc` instead of `/*!< desc`
 
Now that I found these, I am unsure whether I can trust the rest of the exposed values. It looks to me as if large parts of the esp_zb_zcl library were hand-written which allowed errors to creep in.  Excuse me if I'm being a bit blunt here, but has there been any kind of automatic testing before this was published? This is very important to me, because I need to be 100% sure that by devices are behaving according to spec.

### Steps to reproduce.

1. Read the header files and compare to ZCL

### More Information.

N/A

xieqinan, 2025-04-21:
Thanks for your suggestion.

There is indeed an incorrect value for `ESP_ZB_ZCL_TEMP_MEASUREMENT_MAX_MEASURED_VALUE_MINIMUM`, and we will fix it. However, the value of `ESP_ZB_ZCL_TEMP_MEASUREMENT_MAX_MEASURED_VALUE_MAXIMUM` is correct.

Regarding the comment prefix style, we’ll try to maintain consistent alignment, but we believe it shouldn’t block your development.

xieqinan, 2025-05-22:
Hi, the above errors and typos has been fixed in [esp-zigbee-sdk v1.6.5](https://github.com/espressif/esp-zigbee-sdk/commit/735b9a72a4174d5465e20f3cc2c5389a6633927f#diff-7dfa2d47909544332fc1325aca4433a948ff047ac8b3a8e8682e3370d393afa5R36), closing this issue. Please feel free to reopen this issue, if these problems still presist.