espressif/esp-zigbee-sdk #631: zboss api zcl "zb_bdb_finding_binding_target" included in esp zigbee lib bdb

renson-thijs-dhaenens, 2025-04-24:
### Is your feature request related to a problem?

I would like to be able to have find and bind functionality

### Describe the solution you'd like.

I would like to be able to have access to to finding_binding API in the zboss zcl from the esp zigbee lib.

renson-thijs-dhaenens, 2025-05-12:
@chshu , is possible to check if this is on the road map? 
This feature determines if we will be using Espressif module or Silicon Labs module for our Zigbee projects.
We would like to be able to start the find and bind procedure on a coordinator and a node, in order to bind all corresponding clusters from both devices.

xieqinan, 2025-05-12:
Hi @renson-thijs-dhaenens ，

We will support this feature in the next version, which is scheduled for release next week.

> "bdb_comm start bind" does not have target/initiater funcionality

By the way, do you intend to use the  `esp-zigbee-console` to develop your project?

renson-thijs-dhaenens, 2025-05-12:
@xieqinan, At the moment we are using the "All device types" example to explore the esp zigbee sdk functionality. 
Our project is a battery powered zigbee end device, for which we are comparing energy usage with Silicon labs at the moment. 
We already have a Zigbee project on Silicon labs which uses find and bind, but we are looking for the possebility to use ESP zigbee sdk for this, as we have other ESP32 projects.

Any suggestions for us?

renson-thijs-dhaenens, 2025-05-13:
@xieqinan, are there any improvements on the roadmap regarding light sleep or deep slight in the zigbee sdk?

xieqinan, 2025-05-14:
> are there any improvements on the roadmap regarding light sleep or deep slight in the zigbee sdk?

The implementation of light sleep and deep sleep can be found [here](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep). Have you encountered any difficulties?

renson-thijs-dhaenens, 2025-05-14:
Not at the moment, but we are comparing energy usage of esp zigbee and Silicon Labs zigbee module. 
And we are not yet able to get the energy consumption as low as Silicon labs. 
I was able to create an application that uses a combination of deep sleep and light sleep to send a sensor report.
Per reporting message send we have a power consumption of +- 14mC for esp device, for Silicon labs this is +- 7mC. 
So I was wondering if there are any improvements on the roadmap regarding energy consumption for esp zigbee modules.
This impacts our companies decision on wich module we are going to implement in our zigbee project. 

Here you see the power consumption graph, we have a window of 5s where we rejoin the network, send the report, ... 
This needs to be as short as possible, but we stil need to be sure the message was succesfully send. 

The deep sleep power consumption, can this be any lower by using any other configuration settings?

Thanks for the quick responses. 

[image]

xieqinan, 2025-05-14:
Hi,

> I was wondering if there are any improvements on the roadmap regarding energy consumption for esp zigbee modules.

The ESP32-H2 has slightly higher power consumption compared to Silicon Labs' solution, which is primarily due to hardware design. However, an IEEE 802.15.4 SoC with power consumption comparable to Silicon Labs’ devices will be released in the near future. If you're interested, you can reach out to our [technical support](https://www.espressif.com/en/contact-us/technical-inquiries) to reserve or receive updates.

renson-thijs-dhaenens, 2025-05-14:
@xieqinan , thanks you I will contact technical support. 
This issue can be closed.

xieqinan, 2025-05-22:
Hello, the BDB finding and binding feature has already been supported in [esp-zigbee-sdk v1.6.5](https://github.com/espressif/esp-zigbee-sdk/commit/735b9a72a4174d5465e20f3cc2c5389a6633927f#diff-e28f476c0ff300bbb853429c5df1a503185e6cc73c8de99b0b4d498c6b589e7fR67).

renson-thijs-dhaenens, 2025-05-26:
Is there an example available?

renson-thijs-dhaenens, 2025-05-28:
@xieqinan, thanks !

renson-thijs-dhaenens, 2025-06-02:
@xieqinan , can you create a patch file to use find and bind in the customized light/switch example?

xieqinan, 2025-06-03:
> can you create a patch file to use find and bind in the customized light/switch example?

I think it’s similar to this comment: [https://github.com/espressif/esp-zigbee-sdk/issues/631#issuecomment-2910941859](https://github.com/espressif/esp-zigbee-sdk/issues/631#issuecomment-2910941859). If you encounter any issues while adding the Find & Bind feature to the `customized light/switch` example, feel free to open a new issue for further discussion.