espressif/esp-zigbee-sdk #632: Issue with Receiving Periodic Status Messages from IAS Zone Device after Commissioning on ESP-Zigbee-Gateway Coordinator

rshane960, 2025-04-24:
I am developing a coordinator using the esp-zigbee-gateway devkit. I am trying to commission an IAS Zone device to the coordinator. I believe the commissioning is successful because I can receive status messages after commissioning, depending on status changes. However, I want to receive periodic status messages even when there is no status change.

So, I sent a Simple Descriptor Request, and when the response came, I wrote the following code:

1. From the response message of the Simple Descriptor Request,
2. I requested esp_zb_zdo_match_cluster,
3. And when finding the device, I wrote the following code:

if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS)
{
    ESP_LOGI(TAG, "Find device");

    main_device_params_t *new = (main_device_params_t *)malloc(sizeof(main_device_params_t));

    new->short_addr = addr;
    new->endpoint = endpoint;

    esp_zb_zdo_bind_req_param_t bind_req;
    esp_zb_ieee_address_by_short(new->short_addr, new->ieee_addr);
    memcpy(bind_req.src_address, new->ieee_addr, sizeof(esp_zb_ieee_addr_t));
    bind_req.src_endp = 1;
    bind_req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE;
    bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
    esp_zb_get_long_address(bind_req.dst_address_u.addr_long);
    bind_req.dst_endp = 1;
    bind_req.req_dst_addr = new->short_addr;
    ESP_LOGI(TAG, "req_dst_addr = 0x%.4X", bind_req.req_dst_addr);
    esp_zb_zdo_device_bind_req(&bind_req, bind_cb, (void *)new);
}

I couldn’t receive periodic status messages. Please check it for me.

Let me know if you need further assistance with the issue or additional translation!

xieqinan, 2025-04-25:
Please refer to the link: https://github.com/espressif/esp-zigbee-sdk/issues/462#issuecomment-2455337059

xieqinan, 2025-05-07:
@rshane960 ,

Was the link above helpful? If not, please provide more details about the issue. Additionally, sharing a sniffer capture would be helpful for further investigation.

closed 2025-05-07 by rshane960