espressif/esp-zigbee-sdk #633: ESP32_Thread_BR Zigbee Gateway: Smart Plug OTA Upgrade Reboot Issue

rshane960, 2025-04-30:
I am developing an esp-zigbee-gateway with esp32_Thread_BR.  
Currently, I am commissioning a smart plug, but I am encountering a slight issue.  
After commissioning, I receive an "OTA Upgrade: Query Next Image Request" message from the smart plug, and an exception occurs.  
The commissioning itself is successful, and I can receive status updates, but I am unable to identify the cause of the reboot.  
Below is the reboot message:  

```
Guru Meditation Error: Core 0 panic'ed (LoadProhibited). Exception was unhandled.

Core 0 register dump:
PC      : 0x4204ce31  PS      : 0x00060f30  A0      : 0x8204d3c5  A1      : 0x3fcaa0b0
--- 0x4204ce31: zb_zcl_ota_upgrade_send_query_next_image_response at ??:?

A2      : 0x3fcaa10b  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x4201eb78
--- 0x4201eb78: zcl_ota_upgrade_srv_query_image_handler at esp_zigbee_zcl_ota_upgrade.c.obj:?

A6      : 0x3fcb2204  A7      : 0x00000015  A8      : 0x00000000  A9      : 0x3fcaa090
A10     : 0x3fcab368  A11     : 0x00000038  A12     : 0x00000001  A13     : 0x00000001
A14     : 0x00000000  A15     : 0x00000015  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000c  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000
--- 0x400570e8: memset in ROM
0x400570f3: memset in ROM

Backtrace: 0x4204ce2e:0x3fcaa0b0 0x4204d3c2:0x3fcaa0e0 0x42049783:0x3fcaa190 0x420497d0:0x3fcaa1b0 0x42043960:0x3fcaa1d0 0x42043ffe:0x3fcaa210 0x42044059:0x3fcaa230 0x4202c9b0:0x3fcaa270 0x42053b13:0x3fcaa2b0 0x4201cbfc:0x3fcaa2d0 0x4201cc07:0x3fcaa2f0 0x4200e624:0x3fcaa310 0x4037c025:0x3fcaa340
--- 0x4204ce2e: zb_zcl_ota_upgrade_send_query_next_image_response at ??:?
0x4204d3c2: zb_zcl_process_ota_upgrade_specific_commands_srv at zcl_ota_upgrade_srv_commands.c.obj:?
0x42049783: zb_zcl_handle_specific_commands at ??:?
0x420497d0: zb_zcl_handle at ??:?
0x42043960: ep_process_zcl_cmd at zcl_common.c.obj:?
0x42043ffe: zb_zcl_process_parsed_zcl_cmd at ??:?
0x42044059: zb_zcl_process_device_command at ??:?
0x4202c9b0: zb_sched_loop_iteration at ??:?
0x42053b13: zboss_main_loop_iteration at ??:?
0x4201cbfc: esp_zb_stack_main_loop_iteration at ??:?
0x4201cc07: esp_zb_stack_main_loop at ??:?
0x4200e624: zigbee_main_task at C:/1_LocalDisk_D/001_Project/90_NoneNameProject/D_Esp/hcs_w1002_esp32/main/user/User_Zigbee.c:1066
0x4037c025: vPortTaskWrapper at C:/1_LocalDisk_D/001_Project/H_ESP32/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

xieqinan, 2025-04-30:
Hi,

Could you please share the sniffer log for the further investigation?

rshane960, 2025-04-30:
Thank you for your prompt response.  
Below is the sniffer query log for the device:  

[11:21:35.048519] OTA Upgrade: Query Next Image Request 61-88-CB-09-81-00-00-B5-D6-08-02-00-00-B5-D6-1E-E0-28-7F-20-B0-00-F8-07-19-16-00-6F-0D-00-00-6D-A2-26-5C-14-26-69-7B-11-5F-5A-43-0C-E6-E8-42-68-1C-2A-2A-AE-46-2A-C1-FF-FF  

**Frame Information:** (57 bytes)  
- Packet Number: 280  
- Protocol: ZigBee  
- Timestamp: 11:21:35.048519  
- Time Delta: 0.050792  
- Channel: 13  
- Length: 57  
- Link Quality: -12 dBm  
- Source: USB9881  
- Layer: ZCL  
- Status: Decrypted  

**MAC Header:** (9 bytes)  
- Frame Control: 0x8861  
  - Frame Type: [0x1] Data  
  - Security Enabled: [0x0] No  
  - Frame Pending: [0x0] No  
  - Acknowledgement Request: [0x1] Yes  
  - Intra-PAN: [0x1] Yes  
  - Reserved: 0x0  
  - Destination Addr Mode: [0x2] 16-bit Short Address  
  - Reserved: 0x0  
  - Source Addr Mode: [0x2] 16-bit Short Address  
- Sequence Number: 203  
- Destination PAN ID: 0x8109  
- Destination Address: 0x0000  
- Source Address: 0xD6B5  

**MAC Payload:** (46 bytes)  
- **NWK Header:** 0xE01ED6B500000208  
  - Frame Control: 0x0208  
    - Frame Type: [0x0] Data  
    - Protocol Version: 0x2  
    - Route Discovery: [0x0] Suppressed  
    - Multicast Flag: [0x0] Unicast or Broadcast  
    - Security Enabled: [0x1] Yes  
    - Source Route Included: [0x0] No  
    - Destination IEEE Address Included: [0x0] No  
    - Source IEEE Address Included: [0x0] No  
    - Reserved: 0x0  
  - Destination Address: 0x0000  
  - Source Address: 0xD6B5  
  - Radius: 0x1E  
  - Sequence Number: 224  
- **NWK Aux Header:** (14 bytes)  
  - Network Security Control: 0x28  
    - Network Security Level: [0x0] None  
    - Key NWK ID: [0x1] Network Key  
    - Extended Nonce: [0x1] Yes  
    - Reserved: 0x0  
  - NWK Frame Counter: 11542655  
  - Source Address: 00:0D:6F:00:16:19:07:F8  
  - NWK Key Sequence Number: 0  
- **NWK Payload:** (20 bytes)  
  - **APS Header:** 0x4801010400190140  
    - Frame Control: 0x40  
      - Frame Type: [0x0] Data  
      - Delivery Mode: [0x0] Normal Unicast Delivery  
      - Acknowledgement Format: [0x0] Data Frame  
      - Security Enabled: [0x0] No  
      - Acknowledgement Request: [0x1] Yes  
      - Extended Header Present: [0x0] No  
    - Destination Endpoint: 0x01  
    - Cluster ID: [0x0019] General: OTA Upgrade  
    - Profile ID: [0x0104] ZigBee Home Automation  
    - Source Endpoint: 0x01  
    - APS Counter: 72  
  - **APS Payload:** (12 bytes)  
    - **ZCL Header:** 0x012501  
      - Frame Control: 0x01  
        - Frame Type: [0x1] Command is Specific to a Cluster  
        - Manufacturer Specific: [0x0] Manufacturer Code Not Included in the ZCL Frame  
        - Direction: [0x0] From Client to Server  
        - Disable Default Response: [0x0] No  
        - Reserved: 0x0  
      - Transaction Sequence Number: 37  
      - Command ID: [0x01] Query Next Image Request  
    - **ZCL Payload:** (9 bytes)  
      - Field Control: 0x00  
        - Hardware Version Present: [0x00] No  
      - Manufacturer Code: 0x120B  
      - Image Type: [0x210B] Manufacturer Specific  
      - File Version: 0x00000011  
- **NWK MIC:** 0x74BA35BD  

**MAC Footer:** 0xFFFF  
- Frame Check Sequence: 0xFFFF

xieqinan, 2025-04-30:
Have you added the `OTA Upgrade` server cluster to your application? If so, I need to understand how you added and configured it. It seems that an invalid OTA image might have been registered, which could be the cause of this crash.

rshane960, 2025-05-07:
Sorry for replying late.  
Below is the list of clusters currently in use.  
There is an OTA upgrade cluster included here.  

```c
uint16_t cluster[] = 
{
    ESP_ZB_ZCL_CLUSTER_ID_BASIC,
    ESP_ZB_ZCL_CLUSTER_ID_POWER_CONFIG, 
    ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY,
    ESP_ZB_ZCL_CLUSTER_ID_GROUPS,
    ESP_ZB_ZCL_CLUSTER_ID_ON_OFF, 
    ESP_ZB_ZCL_CLUSTER_ID_ON_OFF_SWITCH_CONFIG, 
    ESP_ZB_ZCL_CLUSTER_ID_ALARMS,
    ESP_ZB_ZCL_CLUSTER_ID_METERING,  
    ESP_ZB_ZCL_CLUSTER_ID_OTA_UPGRADE,
    ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE,
    ESP_ZB_ZCL_CLUSTER_ID_DOOR_LOCK,
    ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT,
    ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT, 
    ESP_ZB_ZCL_CLUSTER_ID_ELECTRICAL_MEASUREMENT, 
};
```

The method for adding a cluster is as follows:

```c
esp_err_t zb_cluster_list_add_ota_cluster_fn(esp_zb_cluster_list_t *cluster_list, esp_zb_attribute_list_t *attr_list, uint8_t role_mask)
{
    esp_zb_zcl_ota_upgrade_server_variable_t variable =
    {
        .query_jitter = OTA_UPGRADE_QUERY_JITTER,
        .current_time = OTA_UPGRADE_CURRENT_TIME,
        .file_count = OTA_UPGRADE_IMAGE_COUNT,
    };

    esp_zb_ota_cluster_add_attr(attr_list, ESP_ZB_ZCL_ATTR_OTA_UPGRADE_SERVER_DATA_ID, (void *)&variable);
    return esp_zb_cluster_list_add_ota_cluster(cluster_list, attr_list, role_mask);
}
```

This is how the cluster is being added.  
I don’t need to perform OTA.  
However, I just need to ensure that no exceptions occur from the update query sent by the device.  
The development environment is IDF V5.4.  
Please provide a response.  
Thank you.

xieqinan, 2025-05-07:
> I don’t need to perform OTA.
However, I just need to ensure that no exceptions occur from the update query sent by the device.

If so, I think you only need to remove the OTA upgrade cluster from the endpoint.

rshane960, 2025-05-07:
Thank you for your response.  
It seems the issue has been resolved.  
I have an additional question.  
How can we determine when a device has completed the Join process during commissioning?  
Currently, we are internally determining that the join is complete when the `zb authorized` command is received.  
Is this correct? Please confirm.  
Thank you.

---

xieqinan, 2025-05-07:
> How can we determine when a device has completed the Join process during commissioning?

The `ESP_ZB_BDB_SIGNAL_STEERING` signal with an `ESP_OK` status indicates that a factory-reset device has successfully completed commissioning. Alternatively, you can call `esp_zb_bdb_dev_joined()` to check whether the device has joined a network.

xieqinan, 2025-05-16:
@rshane960 

Do you have any updates?

closed 2025-05-16 by rshane960