espressif/esp-zigbee-sdk #634: Issue sending using APS a buffer of data that is larger than 1613 bytes.

Issaaaaaaaaam, 2025-05-01:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### esp-zigbee-lib version.

1.6.0

### esp-zboss-lib version.

1.6.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

I expect to be able to send a 1616 bytes buffer and receive a APS Data Confirm Callback that is successful. 

### What is the actual behavior?

ZB_TRACE_LOG[0]: aps/aps_main.c:1872    APSF: ERROR - can not fit 1616b into partner buffer 1613b, send fail confirm
E (10107) APSDE CONFIRM: Failed to send APSDE-DATA request, error code: 96. 

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### More Information.

I tried increasing the I/O buffer on the receiver side using esp_zb_io_buffer_size_set, but that did nothing. Also, the max I could pass to this function was 243, it would fail otherwise. Is there any reason for that?
I could not debug further since we do not have access to the aps_main.c. 
Is there any way I could bypass this limit of 1613 bytes? 
I appreciate any help you can provide. 

Code that I use for sending the message buffer. It worked reliably until now for thousands of requests all under 1613 bytes. 
esp_zb_apsde_data_req_t req;
                memset(&req, 0, sizeof(req));
                req.dst_addr_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
                req.dst_addr.addr_short = data_ind.src_short_addr;
                req.dst_endpoint = data_ind.src_endpoint;
                req.profile_id = ESP_ZB_AF_HA_PROFILE_ID;
                req.cluster_id = 0xFFC0;
                req.src_endpoint = HA_ESP_LIGHT_ENDPOINT;
                req.asdu_length = message_buf.size;
                req.asdu = message_buf.data;
                req.radius = 10;
                req.tx_options = (ESP_ZB_APSDE_TX_OPT_ACK_TX | ESP_ZB_APSDE_TX_OPT_FRAG_PERMITTED);
                req.use_alias = false;
                esp_zb_lock_acquire(portMAX_DELAY);
                esp_zb_aps_data_request(&req);
                esp_zb_lock_release();
                handshake_state = noise_handshakestate_get_action(responder);
                log_handshake_state(responder, "Responder");

Issaaaaaaaaam, 2025-05-01:
Small correction: 
esp-zigbee-lib version.

1.6.4
esp-zboss-lib version.

1.6.4

xieqinan, 2025-05-06:
Hi @Issaaaaaaaaam,

The maximum number of bytes in an APS request is constrained by several factors: the transport limitations of the receiver (as indicated in the ZDO Node Descriptor response), the sender’s memory resources (such as I/O buffer size), and the maximum fragment window size. Therefore, the payload size of a single APS request cannot be increased indefinitely. We recommend limiting the maximum APS payload size to **1024 bytes**. Exceeding this limit may lead to unexpected behavior.  
If you need to transmit large amounts of data, consider splitting it across **multiple APS frames**.

Issaaaaaaaaam, 2025-05-06:
Hi @xieqinan ,
Thank you for the answer! 
It seems that I will indeed have to split my payload across multiple APS frames, but since for my use case having an increase of 3 bytes in the I/O buffer size is all I need, is there no way to change any settings to kind of brute force it? I don't mind unexpected behavior/instability I am just testing various things and 1616 bytes should be my maximum payload.

xieqinan, 2025-05-07:
> but since for my use case having an increase of 3 bytes in the I/O buffer size is all I need, is there no way to change any settings to kind of brute force it?

The `esp-zigbee-sdk` (v1.6.4) does not provide an API to set the `Max Incoming/Outgoing Transfer Size` in the ZDO Node Descriptor—mainly because the default value of 1613 bytes is already considered sufficiently large. Therefore, if the APS receiver is an ESP device, you'll need to split your payload across multiple APS frames to ensure reliable transmission.

Issaaaaaaaaam, 2025-05-07:
Ok, I will close the issue.
Thanks for your help, but I do wish that eventually the API lets us set the Max Incoming/Outgoing Transfer Size.