espressif/esp-zigbee-sdk #637: Zigbee NCP over USB: Fundamental Approach

4utomator, 2025-05-02:
### Question

Dear Espressif-Team,
I am trying to get the network co-processor to work. I am using the ESP32-C6-DevKitC which is directly wired to the computer via the USB Type-C to UART Port. The ESP is flashed with the latest `esp_zigbee_ncp` example.

I managed to create a network and connect devices to it using `NETWORK_INIT 0x0000`, `NETWORK_FORM 0x0004` `NETWORK_START 0x0001` and `ZCL_ENDPOINT_ADD 0x0100`.
But how to get the data? Binding an ON/OFF end device cluster to the coordinator using `ZDO_BIND_SET 0x0200` does not seem to work. Instead, the ESP crashes while parsing the received command.

Can you show me a way to receive data from an ON/OFF end device using the ESP32C6 as an NCP dongle?
Kind regards

lane-ae, 2025-05-03:
If I understand correctly you are using just one board?

[Read me](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_ncp/README.md#hardware-required) clearly states that you need another board, third bullet. That board, using host example, is where you are doing all the necessary Zigbee code. NCP you just build and flash.

4utomator, 2025-05-03:
Yes, that is correct. But the NCP is just an serial interface to parts of the ESP Zigbee stack, right?

I took inspiration from [zigpy-espzb](https://github.com/lhespress/zigpy-espzb) which implements a radio for zigpy using the NCP.
I eventually stumbled upon some serious issues regarding the NCP implementation, for example the `ZCL_ATTR_DISC` parameters as [documented](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/ncp.html#zcl-attr-disc) are very different from the [actual implementation](https://github.com/espressif/esp-zigbee-sdk/blob/57fa4199a1b80fe6048f7c818f2e3a939556d467/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h#L249).

Can you provide me with a set of commands to receive data from a ZCL ON/OFF end device using NCP, or is the NCP implementation not mature enough?

lane-ae, 2025-05-03:
Are you using USBtoTTL adapter? As I understand in NCP example you need to use pins that are defined in the code for UART communication.

4utomator, 2025-05-03:
> Are you using USBtoTTL adapter? As I understand in NCP example you need to use pins that are defined in the code for UART communication.

The ESP32-C6-DevKitC has an onboard serial to usb converter.
As mentioned above, the entire connection works. I can create a network and connect endpoints at the NWK layer, all done remotely over the serial connection. The question is how to proceed from there.

lhespress, 2025-05-07:
@4utomator 

>> Instead, the ESP crashes while parsing the received command.

Please check the receive data, it'll trigger assert if the data length is less than expect.

>> receive data from an ON/OFF end device

You can use *ZCL_ATTR_READ 0x0102*, it'll notify application by notification frame.