espressif/esp-zigbee-sdk #638: After flashing on_off_switch example device is not connecting to a network

tomasz90, 2025-05-04:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1-1-gf117477c8

### esp-zigbee-lib version.

~1.6.0

### esp-zboss-lib version.

~1.6.0

### Espressif SoC revision.

ESP32-C6, ESP32H2

### What is the expected behavior?

Answers checklist.

I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.

I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.

I have searched the issue tracker for a similar issue and not found a similar issue.

What is the expected behavior?
Device is connecting to a network.

What is the actual behavior?
Device is not joining my network. And it doesn't produce any logs except when I click btn, I get printed info of toggled switch.

### What is the actual behavior?

Device is connecting to a network.

### Steps to reproduce.

Device is connecting to a network.

### More Information.

Device is connecting to a network.

xieqinan, 2025-05-06:
The on_off_switch example of esp-zigbee-sdk acts as a Coordinator of Zigbee network, which will form a specific Zigbee network and will not connect to other networks. If you intend a Zigbee device to join the network created by your HA, please consider using the on_off_light example.

xieqinan, 2025-05-16:
@tomasz90 ,

Any updates?

xieqinan, 2025-06-13:
Hi @tomasz90 ,

Please refer to comment: https://github.com/espressif/esp-zigbee-sdk/issues/638#issuecomment-2853151230. I think it is not a bug.

I will go ahead and close this issue. Please feel free to reopen it if you’re still experiencing confusion or need further clarification.

closed 2025-06-13 by xieqinan

xieqinan, 2025-06-20:
@mastersf ,

Do you know the device type of your Z2M instance — is it a coordinator, router, or end device?