espressif/esp-zigbee-sdk #639: Bootloader showing different partition table from partitions.csv file

pdEmAk, 2025-05-08:
Hi,

I had previously run into the "failed to find zb_storage partition" error. which I found a solution for in one of this open issue here: https://github.com/espressif/esp-zigbee-sdk/issues/628 however, upon building my project, the bootloader logs show a different partition table from what I currently have configured.

`partitions.csv`
# Name,       Type,  SubType, Offset,   Size,     Flags
nvs,          data,  nvs,     0x9000,   0x6000,
phy_init,     data,  phy,     0xf000,   0x1000,
zb_storage,   data,  fat,     ,         0x20000,
zb_fct,       data,  fat,     ,         0x4000,
factory,      app,   factory, ,         0x100000,

xieqinan, 2025-05-09:
We need more information to investigate this issue:

- Please enable Zigbee debug mode in *menuconfig*, then call `esp_zb_set_trace_level_mask(ESP_ZB_TRACE_LEVEL_WARN, ESP_ZB_TRACE_SUBSYSTEM_ZDO);` on the ESP coordinator side. After that, share the log output here.
- Follow the instructions at https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark to set up a sniffer and capture the frames when the CT attempts to join the coordinator. Then, please share the capture here.

pdEmAk, 2025-05-12:
> We need more information to investigate this issue:
>
> * Please enable Zigbee debug mode in _menuconfig_, then call `esp_zb_set_trace_level_mask(ESP_ZB_TRACE_LEVEL_WARN, ESP_ZB_TRACE_SUBSYSTEM_ZDO);` [...]

Hi, @xieqinan, I am somewhat stuck with this pesky error (which I didn't have originally) since I installed the most recent esp-idf vs code extension update (might be unrelated but it happened right after). I am unable to get the required logs to help debug my initial issue because all I see now is the following error.

```
Error at line 1: Value 'Type' is not valid. Known keywords: bootloader, partition_table, app, data
Please check extra_partition_subtypes.inc file in build/config directory
```

This is partitions table.

```
Name,Type,SubType,Offset,Size,Flags
nvs,data,nvs,0x9000,0x6000,
phy_init,data,phy,0xf000,0x1000,
zb_storage,data,fat,0x10000,0x20000,
zb_fct,data,fat,0x30000,0x4000,
factory,app,factory,0x40000,0x140000,
```
I have the right format set to UTF-8
My current flash size is 8MB so I am not sure if the offsets have anything to do with it, I left the offset blank so the sdk handles it automatically but no luck still.

I currently cannot replicate the original issue in this thread because of this so any help is deeply appreciated.

pdEmAk, 2025-05-12:
I fixed the issue above, however I now get this error.

```
E (733) ZIGBEE_SENDER: Production Config Ready failed with status: ESP_FAIL (0xffffffff)
E (743) ZIGBEE_SENDER: This typically indicates a problem with the zb_storage partition
E (753) ZIGBEE_SENDER: Restarting device to attempt recovery...
```

Based on my research into the issue, it appears to be the bootloader not reading my partitions table correctly even as I have set it to read a custom partitions table inside menuconfig.

My current partitions table is as follows:

```
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,0x6000,
phy_init,data,phy,0xf000,0x1000,
zb_storage,data,0x81,0x10000,0x20000,
zb_fct,data,0x81,0x30000,0x4000,
factory,app,factory,0x40000,0x140000,
```

Bootloader logs show this: 

```
I (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (62) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (68) boot:  2 zb_storage       Unknown data     01 81 00010000 00020000
I (75) boot:  3 zb_fct           Unknown data     01 81 00030000 00004000
I (81) boot:  4 factory          factory app      00 00 00040000 00140000
```

It is shown as "unknown data" which seems to cause the error. and this error makes it difficult for me to replicate the original issue with the CT not joining the network.

pdEmAk, 2025-05-12:
> I fixed the issue above, however I now get this error.
>
> ```
> E (733) ZIGBEE_SENDER: Production Config Ready failed with status: ESP_FAIL (0xffffffff)
> E (743) ZIGBEE_SENDER: This typically indicates a [...]

Cc @xieqinan

xieqinan, 2025-05-13:
> Error at line 1: Value 'Type' is not valid. Known keywords: bootloader, partition_table, app, data
Please check extra_partition_subtypes.inc file in build/config directory

Can the example from the esp-idf repository run on your side? I believe this type of issue can be better addressed through the [[esp-idf issue tracker](https://github.com/espressif/esp-idf)](https://github.com/espressif/esp-idf).

> I (68) boot:  2 zb_storage       Unknown data     01 81 00010000 00020000"

By the way, the above log is normal and not related to your issue.

pdEmAk, 2025-05-13:
> Can the example from the esp-idf repository run on your side? I believe this type of issue can be better addressed through the [[esp-idf issue [...]
Thank you for your feedback. I managed to fix the "'Type' is not valid error". The sdk did not recognize the headers in my partition table because Line 1 didn't begin with #. 
> > I (68) boot:  2 zb_storage       Unknown data     01 81 00010000 00020000"
> 
> By the way, the above log is normal and not related to your issue.
This helps, I was a bit worried that the bootloader wasn't replicating my partitions table. However, I have now moved onto a new error which still makes it difficult for me to adequately monitor the CT's behaviour as it tries to join the network. I would really appreciate pointers as to how navigate a Zigbee stack assertion failure. My bootloader log looks like this:

```
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe84
load:0x4086e610,len:0x3054
entry 0x4086c11a
I (22) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (22) boot: compile time May 13 2025 10:51:16
I (23) boot: chip revision: v0.0
I (24) boot: efuse block revision: v0.1
I (25) boot.esp32c6: SPI Speed      : 80MHz
I (29) boot.esp32c6: SPI Mode       : DIO
I (33) boot.esp32c6: SPI Flash Size : 8MB
I (37) boot: Enabling RNG early entropy source...
I (41) boot: Partition Table:
I (44) boot: ## Label            Usage          Type ST Offset   Length
I (50) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (57) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (63) boot:  2 factory          factory app      00 00 00010000 00140000
I (70) boot:  3 zb_storage       Unknown data     01 81 00150000 00020000
I (76) boot:  4 zb_fct           Unknown data     01 81 00170000 00010000
I (83) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=420c0020 size=44490h (279696) map
I (195) esp_image: segment 1: paddr=000544b8 vaddr=40800000 size=0bb60h ( 47968) load
I (217) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=bbca8h (769192) map
I (498) esp_image: segment 3: paddr=0011bcd0 vaddr=4080bb60 size=0262ch (  9772) load
I (503) esp_image: segment 4: paddr=0011e304 vaddr=4080e190 size=02208h (  8712) load
I (515) boot: Loaded app from partition at offset 0x10000
I (516) boot: Disabling RNG early entropy source...
I (526) cpu_start: Unicore app
I (535) cpu_start: Pro cpu start user code
I (535) cpu_start: cpu freq: 160000000 Hz
I (535) app_init: Application information:
I (536) app_init: Project name:     de-zigbee-diverter
I (540) app_init: App version:      1
I (543) app_init: Compile time:     May 13 2025 10:50:55
I (548) app_init: ELF file SHA256:  025535213...
I (552) app_init: ESP-IDF:          v5.4.1
I (556) efuse_init: Min chip rev:     v0.0
I (560) efuse_init: Max chip rev:     v0.99
I (564) efuse_init: Chip rev:         v0.0
I (568) heap_init: Initializing. RAM available for dynamic allocation:
I (574) heap_init: At 408155E0 len 00067030 (412 KiB): RAM
I (579) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (585) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (591) spi_flash: detected chip: generic
I (594) spi_flash: flash io: dio
I (597) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (603) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (609) coexist: coex firmware version: e727207
I (614) coexist: coexist rom version 5b8dcfa
I (618) main_task: Started on CPU0
I (618) main_task: Calling app_main()
I (628) ZIGBEE_SENDER: Listing all partitions found:
I (628) ZIGBEE_SENDER:   - Found partition: label=nvs       , type=0x01, subtype=0x02, address=0x00009000, size=0x00006000 (24 KB)
I (638) ZIGBEE_SENDER:   - Found partition: label=phy_init  , type=0x01, subtype=0x01, address=0x0000f000, size=0x00001000 (4 KB)
I (648) ZIGBEE_SENDER:   - Found partition: label=factory   , type=0x00, subtype=0x00, address=0x00010000, size=0x00140000 (1280 KB)
I (658) ZIGBEE_SENDER:   - Found partition: label=zb_storage, type=0x01, subtype=0x81, address=0x00150000, size=0x00020000 (128 KB)
I (668) ZIGBEE_SENDER:   - Found partition: label=zb_fct    , type=0x01, subtype=0x81, address=0x00170000, size=0x00010000 (64 KB)
I (678) ZIGBEE_SENDER: Finished listing partitions.
I (688) ZIGBEE_SENDER: Found Zigbee partitions: storage=zb_storage (type=0x1, subtype=0x81, size=128 KB), fct=zb_fct (type=0x1, subtype=0x81, size=64 KB)
I (698) ZIGBEE_SENDER: 'force_rst' flag not found in NVS, assuming normal.
[... 1 lines cut ...]
I (718) ZIGBEE_SENDER: Configuring Zigbee stack with explicit parameters
I (718) ZIGBEE_SENDER: Initializing Zigbee stack...
I (728) ESP_ZIGBEE_CORE: SDK Version: v1.6.4-9d7f2d69-c6fae44d-fd3e087; esp32c6; 2025-02-21 02:29:45 UTC
I (738) phy_init: phy_version 330,4af1121,Feb  6 2025,14:39:13
I (798) phy: libbtbb version: 99f6adc, Feb  6 2025, 14:39:27
I (798) ZIGBEE_SENDER: Setting Zigbee to use channel 15 for better stability
I (798) ZIGBEE_SENDER: Added client clusters to list.
I (818) ZIGBEE_SENDER: Zigbee stack initialized
I (818) ZIGBEE_SENDER: Core action handler registered.
I (818) ZIGBEE_SENDER: Zigbee main task started
I (818) ZIGBEE_SENDER: zbossSignalHandler called with signal pointer: 0x408205c0
I (828) ZIGBEE_SENDER: App Signal Received: type=23 (0x17), status=ESP_FAIL
W (828) ZIGBEE_SENDER: Production Config Ready failed with status: ESP_FAIL (0xffffffff)
[... cut ...]
I (848) ZIGBEE_SENDER: Starting network formation
I (858) ZIGBEE_SENDER: zbossSignalHandler called with signal pointer: 0x408205c0
I (858) ZIGBEE_SENDER: App Signal Received: type=1 (0x1), status=ESP_OK
I (868) ZIGBEE_SENDER: Skip startup signal received
I (878) ZIGBEE_SENDER: Starting network formation
ZB_TRACE_LOG[0]: common/zb_debug.c:64    Assertion failed /builds/thread_zigbee/esp-zboss/components/zboss_stack/zboss/zdo/zdo_app.c:1125
Zigbee stack assertion failed zdo/zdo_app.c:1125
```

Thank you.

pdEmAk, 2025-05-13:
Some update after several hours trying to hack the issue. I now get a different error that still leans into the same scope. I would appreciate any help.

Bootloader log is below: Cc @xieqinan 

```
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe84
load:0x4086e610,len:0x3054
entry 0x4086c11a
I (22) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (22) boot: compile time May 13 2025 17:00:00
I (23) boot: chip revision: v0.0
I (24) boot: efuse block revision: v0.1
I (25) boot.esp32c6: SPI Speed      : 80MHz
I (29) boot.esp32c6: SPI Mode       : DIO
I (33) boot.esp32c6: SPI Flash Size : 8MB
I (37) boot: Enabling RNG early entropy source...
I (41) boot: Partition Table:
I (44) boot: ## Label            Usage          Type ST Offset   Length
I (50) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (57) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (63) boot:  2 factory          factory app      00 00 00010000 00140000
I (70) boot:  3 zb_storage       Unknown data     01 81 00150000 00020000
I (76) boot:  4 zb_fct           Unknown data     01 81 00170000 00010000
I (83) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=420c0020 size=44980h (280960) map
I (196) esp_image: segment 1: paddr=000549a8 vaddr=40800000 size=0b670h ( 46704) load
I (217) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=bc1b0h (770480) map
I (498) esp_image: segment 3: paddr=0011c1d8 vaddr=4080b670 size=02b1ch ( 11036) load
I (504) esp_image: segment 4: paddr=0011ecfc vaddr=4080e190 size=02208h (  8712) load
I (516) boot: Loaded app from partition at offset 0x10000
I (517) boot: Disabling RNG early entropy source...
I (527) cpu_start: Unicore app
I (536) cpu_start: Pro cpu start user code
I (536) cpu_start: cpu freq: 160000000 Hz
I (536) app_init: Application information:
I (536) app_init: Project name:     de-zigbee-diverter
I (541) app_init: App version:      1
I (544) app_init: Compile time:     May 13 2025 16:59:40
I (549) app_init: ELF file SHA256:  e505393d4...
I (553) app_init: ESP-IDF:          v5.4.1
I (557) efuse_init: Min chip rev:     v0.0
I (561) efuse_init: Max chip rev:     v0.99
I (565) efuse_init: Chip rev:         v0.0
I (569) heap_init: Initializing. RAM available for dynamic allocation:
I (575) heap_init: At 408155E0 len 00067030 (412 KiB): RAM
I (580) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (585) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (592) spi_flash: detected chip: generic
I (595) spi_flash: flash io: dio
I (598) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (604) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (610) coexist: coex firmware version: e727207
I (615) coexist: coexist rom version 5b8dcfa
I (619) main_task: Started on CPU0
I (619) main_task: Calling app_main()
I (629) ZIGBEE_SENDER: Listing all partitions found:
I (629) ZIGBEE_SENDER:   - Found partition: label=nvs       , type=0x01, subtype=0x02, address=0x00009000, size=0x00006000 (24 KB)
I (639) ZIGBEE_SENDER:   - Found partition: label=phy_init  , type=0x01, subtype=0x01, address=0x0000f000, size=0x00001000 (4 KB)
I (649) ZIGBEE_SENDER:   - Found partition: label=factory   , type=0x00, subtype=0x00, address=0x00010000, size=0x00140000 (1280 KB)
I (659) ZIGBEE_SENDER:   - Found partition: label=zb_storage, type=0x01, subtype=0x81, address=0x00150000, size=0x00020000 (128 KB)
I (669) ZIGBEE_SENDER:   - Found partition: label=zb_fct    , type=0x01, subtype=0x81, address=0x00170000, size=0x00010000 (64 KB)
I (679) ZIGBEE_SENDER: Finished listing partitions.
I (689) ZIGBEE_SENDER: zb_storage partition details: address=0x150000, size=0x20000
I (699) ZIGBEE_SENDER: zb_fct partition details: address=0x170000, size=0x10000
I (699) ZIGBEE_SENDER: Found Zigbee partitions: storage=zb_storage (type=0x1, subtype=0x81, size=128 KB), fct=zb_fct (type=0x1, subtype=0x81, size=64 KB)
I (719) ZIGBEE_SENDER: Checking partition content for virgin flash.
I (719) ZIGBEE_SENDER: Detected virgin flash or read error. Marking as first-time setup.
[... 12 lines cut ...]
I (919) ZIGBEE_SENDER: Zigbee stack start task created
I (929) ZIGBEE_SENDER: Zigbee main task started
I (929) ZIGBEE_SENDER: Zigbee main loop task created
I (939) CT_APP: Zigbee coordinator started
I (939) CT_APP: Waiting for network formation and device discovery...
I (3919) ZIGBEE_SENDER: Calling esp_zb_start...
I (4019) ZIGBEE_SENDER: Zigbee stack started successfully
I (4019) ZIGBEE_SENDER: zbossSignalHandler called with signal pointer: 0x408225a0
I (4019) ZIGBEE_SENDER: App Signal Received: type=23 (0x17), status=ESP_FAIL
W (4029) ZIGBEE_SENDER: Production Config Ready failed with status: ESP_FAIL (0xffffffff)
W (4029) ZIGBEE_SENDER: This typically indicates a problem with the zb_storage partition
I (4039) ZIGBEE_SENDER: Attempting recovery by erasing Zigbee partitions...
I (4049) ZIGBEE_SENDER: Recovery: zb_storage partition details: address=0x150000, size=0x20000
I (4059) ZIGBEE_SENDER: Recovery: zb_fct partition details: address=0x170000, size=0x10000
I (4339) ZIGBEE_SENDER: Recovery: Successfully erased zb_storage partition
I (4369) ZIGBEE_SENDER: Recovery: Successfully erased zb_fct partition
I (4369) ZIGBEE_SENDER: Recovery successful. Proceeding with default configuration...
I (4369) ZIGBEE_SENDER: Cleared force_rst flag to prevent restart loop
I (4369) ZIGBEE_SENDER: Recovery successful. Will restart the device to apply partition changes.
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x7c (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
--- 0x4001975a: software_reset_cpu in ROM
```

xieqinan, 2025-05-14:
Could you provide a simple code snippet to reproduce your problem? I believe this is not a difficult issue—possibly not even Zigbee-related—but rather a matter of some specific implementation details.

pdEmAk, 2025-05-14:
> Could you provide a simple code snippet to reproduce your problem? I believe this is not a difficult issue—possibly not even Zigbee-related—but rather a matter of some specific implementation details.

Of course! 

```
// Private constructor implementation
ZigbeeSender::ZigbeeSender()
    : _networkFormed(false),
      _sensorFound(false),
      _lastCurrentValue(-1.0f),
      _remoteSensorShortAddr(0xFFFF),
      _remoteSensorEndpoint(0),
      _read_failure_count(0)
{
    _instance = this;

    // Initialize NVS flash
    ESP_ERROR_CHECK(nvs_flash_init());

    // --- Debug: List all partitions ---
    ESP_LOGI(TAG, "Listing all partitions found:");
    esp_partition_iterator_t it = esp_partition_find(ESP_PARTITION_TYPE_ANY, ESP_PARTITION_SUBTYPE_ANY, NULL);
    while (it != NULL) {
        const esp_partition_t *part = esp_partition_get(it);
        if (part) {
             ESP_LOGI(TAG, "  - Found partition: label=%-10s, type=0x%02x, subtype=0x%02x, address=0x%08lx, size=0x%08lx (%lu KB)",
                      part->label ? part->label : "(null)", part->type, part->subtype, (unsigned long)part->address, (unsigned long)part->size, (unsigned long)(part->size / 1024));
        }
        it = esp_partition_next(it);
    }
    esp_partition_iterator_release(it);
    ESP_LOGI(TAG, "Finished listing partitions.");
    // --- End Debug ---

    // Find Zigbee storage partitions with exact subtype 0x81 and name matching
    // The ESP-IDF Zigbee stack specifically requires partitions with these exact parameters
    const esp_partition_t* zb_storage_part = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, 
                                                                      static_cast<esp_partition_subtype_t>(0x81), 
                                                                      "zb_storage");
    const esp_partition_t* zb_fct_part = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, 
                                                                  static_cast<esp_partition_subtype_t>(0x81), 
                                                                  "zb_fct");
    
    // Log detailed partition information to help with debugging
    if (zb_storage_part) {
        ESP_LOGI(TAG, "zb_storage partition details: address=0x%lx, size=0x%lx", zb_storage_part->address, zb_storage_part->size);
    }
    if (zb_fct_part) {
        ESP_LOGI(TAG, "zb_fct partition details: address=0x%lx, size=0x%lx", zb_fct_part->address, zb_fct_part->size);
    }

    if (!zb_storage_part || !zb_fct_part) {
        ESP_LOGE(TAG, "Failed to find Zigbee partitions");
        return; // Critical error, cannot continue
    }

    ESP_LOGI(TAG, "Found Zigbee partitions: storage=%s (type=0x%x, subtype=0x%x, size=%lu KB), fct=%s (type=0x%x, subtype=0x%x, size=%lu KB)", 
             zb_storage_part->label, zb_storage_part->type, zb_storage_part->subtype, (unsigned long)(zb_storage_part->size / 1024),
             zb_fct_part->label, zb_fct_part->type, zb_fct_part->subtype, (unsigned long)(zb_fct_part->size / 1024));

    // Check NVS for a reset flag
    nvs_handle_t nvs_handle_flags = 0;
    uint8_t force_zb_reset_flag = 0;
    bool first_time_setup = false;

    esp_err_t nvs_err = nvs_open("zb_flags", NVS_READWRITE, &nvs_handle_flags);
    if (nvs_err == ESP_OK) {
        esp_err_t get_err = nvs_get_u8(nvs_handle_flags, "force_rst", &force_zb_reset_flag);
        if (get_err == ESP_ERR_NVS_NOT_FOUND) {
            ESP_LOGI(TAG, "'force_rst' flag not found in NVS, assuming normal.");
[... 386 lines cut ...]
        // Log detailed device information
        ESP_LOGI(TAG, "NEW DEVICE JOINED THE NETWORK!");
        ESP_LOGI(TAG, "  - Short Address: 0x%04hx", dev_annce_params->device_short_addr);
        ESP_LOGI(TAG, "  - IEEE Address: %s", ieee_addr_str);
        ESP_LOGI(TAG, "  - Capability: 0x%02x", dev_annce_params->capability);

        // Decode capability bits
        ESP_LOGI(TAG, "  - Device Type: %s", (dev_annce_params->capability & 0x02) ? "Router" : "End Device");
        ESP_LOGI(TAG, "  - Power Source: %s", (dev_annce_params->capability & 0x04) ? "Mains" : "Battery");
        ESP_LOGI(TAG, "  - Receiver On When Idle: %s", (dev_annce_params->capability & 0x08) ? "Yes" : "No");

        // Start discovery process to find the device's endpoints and clusters
        ESP_LOGI(TAG, "Starting discovery process for the new device...");
        discoverDevices();
    }
    break;

    default:
        ESP_LOGI(TAG, "Received unhandled signal: %s (0x%x), status: %s",
                 esp_zb_zdo_signal_to_string(signal_type), signal_type,
                 esp_err_to_name(err_status));
        break;
    }
}
```

xieqinan, 2025-05-15:
The issue seems to be triggered in the `ESP_ZB_ZDO_SIGNAL_PRODUCTION_CONFIG_READY` case. Have you configured the `zb_fct` storage using the [mfg_tool](https://github.com/espressif/esp-zigbee-sdk/tree/main/tools/mfg_tool)?  

If not, you can safely ignore the `ESP_ZB_ZDO_SIGNAL_PRODUCTION_CONFIG_READY` signal when it fails. This signal only indicates the presence (or absence) of preconfigured network information and does **not** imply a failure related to the `partitions`.

pdEmAk, 2025-05-15:
> The issue seems to be triggered in the `ESP_ZB_ZDO_SIGNAL_PRODUCTION_CONFIG_READY` case. Have you configured the `zb_fct` storage using the [...]

Safely ignoring the warning that the production config fails appears to get me over the hump there, thanks. However, the logs I get show some problem with the NVRAM image causing an assertion error. I could be mistaken but this is my current conclusion.

```
I (22) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (22) boot: compile time May 15 2025 09:51:41
I (23) boot: chip revision: v0.0
I (24) boot: efuse block revision: v0.1
I (25) boot.esp32c6: SPI Speed      : 80MHz
I (29) boot.esp32c6: SPI Mode       : DIO
I (33) boot.esp32c6: SPI Flash Size : 8MB
I (37) boot: Enabling RNG early entropy source...
I (41) boot: Partition Table:
I (44) boot: ## Label            Usage          Type ST Offset   Length
I (50) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (57) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (63) boot:  2 factory          factory app      00 00 00010000 00140000
I (70) boot:  3 zb_storage       Unknown data     01 81 00150000 00020000
I (76) boot:  4 zb_fct           Unknown data     01 81 00170000 00010000
I (83) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=420c0020 size=457d8h (284632) map
I (197) esp_image: segment 1: paddr=00055800 vaddr=40800000 size=0a818h ( 43032) load
I (216) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=bcd30h (773424) map
I (499) esp_image: segment 3: paddr=0011cd58 vaddr=4080a818 size=03974h ( 14708) load
I (506) esp_image: segment 4: paddr=001206d4 vaddr=4080e190 size=02208h (  8712) load
I (518) boot: Loaded app from partition at offset 0x10000
I (519) boot: Disabling RNG early entropy source...
I (529) cpu_start: Unicore app
I (538) cpu_start: Pro cpu start user code
I (538) cpu_start: cpu freq: 160000000 Hz
I (538) app_init: Application information:
I (539) app_init: Project name:     de-zigbee-diverter
I (543) app_init: App version:      1
I (546) app_init: Compile time:     May 15 2025 09:51:22
I (551) app_init: ELF file SHA256:  21665e6bd...
I (556) app_init: ESP-IDF:          v5.4.1
I (559) efuse_init: Min chip rev:     v0.0
I (563) efuse_init: Max chip rev:     v0.99 
I (567) efuse_init: Chip rev:         v0.0
I (571) heap_init: Initializing. RAM available for dynamic allocation:
I (577) heap_init: At 408155E0 len 00067030 (412 KiB): RAM
I (582) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (588) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (594) spi_flash: detected chip: generic
I (597) spi_flash: flash io: dio
I (600) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (606) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (613) coexist: coex firmware version: e727207
I (617) coexist: coexist rom version 5b8dcfa
I (621) main_task: Started on CPU0
I (621) main_task: Calling app_main()
I (631) ZIGBEE_SENDER: Listing all partitions found:
I (631) ZIGBEE_SENDER:   - Found partition: label=nvs       , type=0x01, subtype=0x02, address=0x00009000, size=0x00006000 (24 KB)
I (641) ZIGBEE_SENDER:   - Found partition: label=phy_init  , type=0x01, subtype=0x01, address=0x0000f000, size=0x00001000 (4 KB)
I (651) ZIGBEE_SENDER:   - Found partition: label=factory   , type=0x00, subtype=0x00, address=0x00010000, size=0x00140000 (1280 KB)
I (661) ZIGBEE_SENDER:   - Found partition: label=zb_storage, type=0x01, subtype=0x81, address=0x00150000, size=0x00020000 (128 KB)
I (671) ZIGBEE_SENDER:   - Found partition: label=zb_fct    , type=0x01, subtype=0x81, address=0x00170000, size=0x00010000 (64 KB)
I (681) ZIGBEE_SENDER: Finished listing partitions.
I (691) ZIGBEE_SENDER: zb_storage partition details: address=0x150000, size=0x20000
I (701) ZIGBEE_SENDER: zb_fct partition details: address=0x170000, size=0x10000
I (701) ZIGBEE_SENDER: Found Zigbee partitions: storage=zb_storage (type=0x1, subtype=0x81, size=128 KB), fct=zb_fct (type=0x1, subtype=0x81, size=64 KB)
I (721) ZIGBEE_SENDER: 'force_rst' flag not found in NVS, assuming normal.
I (721) ZIGBEE_SENDER: Checking partition content for virgin flash.
I (731) ZIGBEE_SENDER: Partitions appear to have existing data (not virgin flash).
I (741) ZIGBEE_SENDER: Initializing Zigbee stack with optimized ESP-IDF approach
I (741) ZIGBEE_SENDER: Configuring Zigbee stack as coordinator
I (751) ZIGBEE_SENDER: Initializing Zigbee NVRAM storage...
I (751) ZIGBEE_SENDER: Found Zigbee partitions: zb_storage (0x150000, 131072 bytes), zb_fct (0x170000, 65536 bytes)
W (761) ZIGBEE_SENDER: Zigbee reset attempt #2
[... 38 lines cut ...]

I (4611) ZIGBEE_SENDER: zigbee_stack_start_task_wrapper: Network formation initiated. Check Zigbee event logs for status.
I (4621) ZIGBEE_SENDER: zbossSignalHandler called with signal pointer: 0x408225c0
I (4631) ZIGBEE_SENDER: App Signal Received: type=23 (0x17), status=ESP_FAIL
W (4631) ZIGBEE_SENDER: No production config found (status: ESP_FAIL) - this is normal if not using mfg_tool
I (5641) ZIGBEE_SENDER: Setting network parameters
I (5641) ZIGBEE_SENDER: Starting initialization
I (5641) ZIGBEE_SENDER: Initialization successful, waiting before network formation
I (6641) ZIGBEE_SENDER: Starting network formation
I (6641) ZIGBEE_SENDER: Network formation command successful - awaiting formation event
[... cut ...]
A2      : 0x00000000  A3      : 0x40822579  A4      : 0x00000001  A5      : 0x40815000
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x4206d998
S4      : 0x005c50c9  S5      : 0x00000008  S6      : 0x00000000  S7      : 0x0000000e  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
```
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000002  MTVAL   : 0x000

xieqinan, 2025-05-15:
Why are there so many calls to `esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_FORMATION)`?  
Could you please refer to the [simple example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c#L104) for forming a Zigbee network?

xieqinan, 2025-06-13:
@pdEmAk ,

Do you have any updates?

pdEmAk, 2025-06-13:
> [@pdEmAk](https://github.com/pdEmAk) ,
> 
> Do you have any updates?

hi @xieqinan , thanks for checking in. I got pulled into working on something else but coincidentally, I picked up this feature this week and have managed to get the CT to join the network but my current error's around reading the RMS from the CT. Would be great if you give me pointers here. 

My logs and code are below.

```
I (35746) ZB_COORD: Received Zigbee signal: 0x30, status: ESP_OK
I (35746) ZB_COORD: Device update signal received
I (35756) ZB_COORD: Received Zigbee signal: 0x2F, status: ESP_OK
I (35756) ZB_COORD: Device announcement signal 0x2F received
I (35766) ZB_COORD: New device joined: 0x9a25
I (35766) ZB_COORD: Device 0x9a25 already known, skipping
I (41496) ZB_COORD: Periodic read from device 0: 0x9a25
E (41496) ZB_COORD: Failed to send read attribute command: ERROR (0x1)
E (41496) ZB_COORD: Request details: device=0x9a25, dst_ep=1, src_ep=1, cluster=0x0b04, attr=0x0508
I (43306) ZB_COORD: Received Zigbee signal: 0x12, status: ESP_OK
I (43306) ZB_COORD: Unhandled signal: 0x12, status: ESP_OK
ZB_TRACE_LOG[0]: common/zb_address.c:1138    zb_address_delete ref 57
I (43316) ZB_COORD: Received Zigbee signal: 0x30, status: ESP_OK
I (43316) ZB_COORD: Device update signal received
I (43316) ZB_COORD: Received Zigbee signal: 0x2F, status: ESP_OK
I (43326) ZB_COORD: Device announcement signal 0x2F received
I (43326) ZB_COORD: New device joined: 0x9a25
I (43336) ZB_COORD: Device 0x9a25 already known, skipping
I (47876) ZB_COORD: Received Zigbee signal: 0x32, status: ESP_OK
I (47876) ZB_COORD: Unhandled signal: 0x32, status: ESP_OK
I (50866) ZB_COORD: Received Zigbee signal: 0x12, status: ESP_OK
I (50866) ZB_COORD: Unhandled signal: 0x12, status: ESP_OK
ZB_TRACE_LOG[0]: common/zb_address.c:1138    zb_address_delete ref 57
I (50876) ZB_COORD: Received Zigbee signal: 0x30, status: ESP_OK
I (50876) ZB_COORD: Device update signal received
I (50886) ZB_COORD: Received Zigbee signal: 0x2F, status: ESP_OK
I (50886) ZB_COORD: Device announcement signal 0x2F received
I (50896) ZB_COORD: New device joined: 0x9a25
I (50896) ZB_COORD: Device 0x9a25 already known, skipping
I (52496) ZB_COORD: Periodic read from device 0: 0x9a25
E (52496) ZB_COORD: Failed to send read attribute command: ERROR (0x2)
E (52496) ZB_COORD: Request details: device=0x9a25, dst_ep=1, src_ep=1, cluster=0x0b04, attr=0x0508
I (58436) ZB_COORD: Received Zigbee signal: 0x12, status: ESP_OK
I (58436) ZB_COORD: Unhandled signal: 0x12, status: ESP_OK
ZB_TRACE_LOG[0]: common/zb_address.c:1138    zb_address_delete ref 57
I (58446) ZB_COORD: Received Zigbee signal: 0x30, status: ESP_OK
I (58446) ZB_COORD: Device update signal received
I (58446) ZB_COORD: Received Zigbee signal: 0x2F, status: ESP_OK
I (58456) ZB_COORD: Device announcement signal 0x2F received
I (58456) ZB_COORD: New device joined: 0x9a25
I (58466) ZB_COORD: Device 0x9a25 already known, skipping
```

```
static void read_ct_clamp_data(uint16_t device_addr, uint8_t endpoint)
{
    // Create the attribute ID array
    static uint16_t attr_id = ESP_ZB_ZCL_ATTR_ELECTRICAL_MEASUREMENT_RMS_CURRENT_ID;

    // Create the read attribute command structure
    esp_zb_zcl_read_attr_cmd_t read_req;
    read_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    read_req.zcl_basic_cmd.dst_addr_u.addr_short = device_addr;
    read_req.zcl_basic_cmd.dst_endpoint = endpoint;
    read_req.zcl_basic_cmd.src_endpoint = HA_ESP_COORDINATOR_ENDPOINT;
    read_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ELECTRICAL_MEASUREMENT;
    read_req.manuf_specific = 0;
    read_req.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
    read_req.dis_defalut_resp = 0;
    read_req.manuf_code = 0;
    read_req.attr_number = 1;
[... 65 lines cut ...]
                        break;
                    case ESP_ZB_ZCL_ATTR_TYPE_U32:
                        ESP_LOGI(TAG, "Attribute value (U32): %lu", *(uint32_t *)(variable->attribute.data.value));
                        break;
                    case ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING:
                        ESP_LOGI(TAG, "Attribute value (String): %.*s",
                                 variable->attribute.data.size,
                                 (char *)variable->attribute.data.value);
                        break;
                    default:
                        ESP_LOGI(TAG, "Attribute type: 0x%02x",
                                 variable->attribute.data.type);
                        break;
                    }
                }
            }
            variable = variable->next;
        }
    }
    else
    {
        ESP_LOGW(TAG, "Failed to read attribute, status: 0x%02x", message->info.status);
    }
}
```

closed 2025-06-17 by pdEmAk