espressif/esp-zigbee-sdk #640: iperf over Zigbee

NMassimo, 2025-05-08:
### Question

Hello, I can't figure out how to get the iperf command on the Zigbee console to work.
The firmware is built from the esp_zigbee_all_device_types_app example.
To register the ping_iperf_cluster I ran on every node the command:
`dm add -e 3 -c 0xff01`
and received the following error:
```
E (101921) ESP_ZIGBEE_ENDPOINT: The requested endpoint ID (3) is not found
E (101921) ESP_ZIGBEE_CLUSTER: The requested cluster ID (0xff01) is not found
```
Running the command a second time doesn't give an error.
After this I create the network on the first node and connect the other nodes to it.
Then when i run the iperf command:
`iperf start -e 3 --dst-ep 3 -d 0xfc85 -t 3 -i 50 -l 100`
the Zigbee stack crashes giving the following error:
```
Zigbee stack assertion failed zcl/zcl_general_commands.c:612

abort() was called at PC 0x4202e30b on core 0
Core  0 register dump:
MEPC    : 0x408007c6  RA      : 0x40808908  SP      : 0x4081c300  GP      : 0x4080f8e4  
TP      : 0x4081c530  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x4081c364  A0      : 0x4081c32c  A1      : 0x4081c362  
A2      : 0x00000000  A3      : 0x4081c359  A4      : 0x00000001  A5      : 0x40816000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000003  S3      : 0x0000ff01  
S4      : 0x00000002  S5      : 0x00000001  S6      : 0x00000001  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
4081c300: 0x00000000 0x00000000 0x4081c360 0x4080de66 0x00000001 0x00000001 0x00000002 0x40810b38
4081c320: 0x4081c364 0x40810b54 0x4081c360 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
4081c340: 0x43502074 0x34783020 0x65323032 0x20623033 0x63206e6f 0x2065726f 0x00000030 0x42020000
4081c360: 0x40810030 0x32303234 0x62303365 0x00000000 0x0000002f 0x0000ffff 0x420a23a5 0x4202e30e
4081c380: 0x4081c444 0x0000000d 0x408193cc 0x00000264 0x40880000 0x4081930c 0x00000001 0x4204f73c
4081c3a0: 0x4081930c 0x408192d0 0x00000002 0x0000ff01 0x00000000 0x4081c41a 0x4081c438 0x4204d3e4
4081c3c0: 0x40880000 0x408192d0 0x40816ac0 0x00000000 0x00000000 0x4081b290 0x40818f44 0x0000000d
4081c3e0: 0x4209c000 0x40818f44 0x4081c438 0x420165aa 0x00000000 0x4081b290 0x40819270 0x4201137c
4081c400: 0x40818f44 0x408192d0 0x0000000d 0x42011a0e 0x00000000 0x4081b290 0x00038f44 0x00640032
4081c420: 0x4209c000 0x40818f44 0x0000000d 0x4200cca8 0x00000100 0x00001800 0x0064fc85 0x00030032
4081c440: 0x00000303 0x40818fc4 0x40819010 0x40819160 0x408191a4 0x408191e8 0x4081922c 0x40819270
4081c460: 0x4080f7ac 0x4080f7ac 0x00000000 0x4200aa24 0x408236dc 0x4081b28a 0x40818f40 0x0000000e
4081c480: 0x420981ac 0x4080f7ac 0x00000000 0x4200aa14 0x0000000e 0x00000000 0x40817000 0x0000000e
4081c4a0: 0x0000000e 0x40818f40 0x420981ac 0x4201088c 0x4081ac80 0x4081c4e0 0x40818f40 0x42012a6e
4081c4c0: 0x00000000 0x00000000 0x00000000 0x00000000 0xffffffff 0x4081aee4 0x408236dc 0x42012d98
4081c4e0: 0xffffffff 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x408096d8
4081c500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c520: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081c540: 0x00000150 0x4081c450 0x00002c5f 0x40811c50 0x40811c50 0x4081c544 0x40811c48 0x00000017
4081c560: 0x4081afcc 0x4081afcc 0x4081c544 0x00000000 0x00000002 0x4081b540 0x736e6f63 0x5f656c6f
4081c580: 0x6c706572 0x00000000 0x4081c530 0x00000002 0x00000001 0x00000000 0x00000000 0x00000000
4081c5a0: 0x408174f8 0x40817560 0x408175c8 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
4081c5c0: 0x00000000 0x42007430 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c5e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c600: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c620: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c640: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c660: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c680: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00001000 0xa5a5a5a5 0xa5a5a5a5
4081c6a0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081c6c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081c6e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 2e298a1b3

Rebooting...
```

What am I doing wrong?

### Additional context.

Devices: ESP32-C6
ESP-IDF v5.3.2
Latest esp-zigbee-sdk version

xieqinan, 2025-05-09:
Hi @NMassimo ,

Could you please share the complete command lines that trigger this `assert()`?

lpy4105, 2025-05-09:
Please also add the server cluster on the node: `dm add -e 3 0xff01`.
Don't forget to register the data model: `dm register` before you start commissioning.

xieqinan, 2025-05-09:
The available command lines may look like this:

- device 1, short address: 0xe4ad
```
channel 11
dm add -e 3 -c 0xff01
dm add -e 3 -c 0xff01 -r C
dm register
bdb_comm start form
network open -t 180
```
- device 2, short address: 0xb075
```
channel 11
dm add -e 3 -c 0xff01
dm add -e 3 -c 0xff01 -r C
dm register
bdb_comm start steer
iperf start -e 3 --dst-ep 3 -d 0xe4ad -t 3 -i 50 -l 100
```

NMassimo, 2025-05-09:
@xieqinan @lpy4105 
Thank you! I was missing the client cluster, now with `dm add -e 3 -c 0xff01 -r C` iperf works as intended.

closed 2025-05-09 by NMassimo